OpenIntelWireless/itlwm #870: [AX201] No config found for PCI dev a0f0/6074, itlwm is loaded, but attach fail.

lshbluesky, 2023-04-17:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
- Kext Version: itlwm v2.1.0 Stable
- WiFi Card Model: Intel Wi-Fi 6 AX201 160MHz
- PCI Product ID: pci8086,a0f0 (Subsystem ID: pci8086,6074)
- macOS Version: macOS Big Sur 11.7.5 (20G1225)

**Description**
Hello.
My laptop (NT750XDA-KF57U) is Intel Wi-Fi 6 AX201 160MHz and has macOS Big Sur 11.7.5 version installed.

By the way, I checked IOReg and found that the Kext is not attached to the Wi-Fi wireless LAN card device (CNVW).
Neither itlwm nor AirportItlwm is attached to CNVW.
Of course, I also checked the FAQ and Troubleshooting site, but it was not solved.

itlwm / AirportItlwm has been loaded successfully.
However, it is just attach failed to CNVW Device.

```
[    5.714505]: itlwm: No config found for PCI dev a0f0/6074, rev=0x354, rfid=0x10a100
[    5.714751]: itlwm: ieee80211_ifdetach
[    5.714753]: itlwm: ieee80211_proto_detach
[    5.714754]: itlwm: ieee80211_crypto_detach
[    5.714755]: itlwm: ieee80211_node_detach
[    5.714757]: itlwm: ifmedia_delete_instance
[    5.714759]: itlwm: attach fail
[    5.714763]: itlwm: free
```

I searched for the sentence `No config found for PCI` in Issues and found the following.
https://github.com/OpenIntelWireless/itlwm/issues/793

So, I try Cold Booting, Warm Booting, but it still doesn't work.

By any chance, even if Device ID is A0F0, AX201 with Sub ID pci8086,6074 is not supported by itlwm / AirportItlwm?

Here are the Heliport and IOReg screenshots.
I uploaded the Bug Report Archive file at the bottom.

[image]
[image]

**Bug Report Archive**
[bugreport_21747.zip](https://github.com/OpenIntelWireless/itlwm/files/11249139/bugreport_21747.zip)

**Kext Download Source**
Official OpenIntelWireless itlwm Releaes

zxystd, 2023-05-13:
@lshbluesky Hi, please use this one, I think it should be work.
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/11468822/itlwm.kext.zip)

lshbluesky, 2023-05-13:
> @lshbluesky Hi, please use this one, I think it should be work.
> [itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/11468822/itlwm.kext.zip)

Hello.
Now, both itlwm and AirportItlwm work fine. ✨
Thank you. 🙂

closed 2023-05-13 by lshbluesky