OpenIntelWireless/itlwm #876: Wifi and Bluetooth not working on Intel AC 7265

Human79, 2023-05-26:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Bug Report Ar
[bugreport_39922.zip](https://github.com/OpenIntelWireless/itlwm/files/11574367/bugreport_39922.zip)
[bugreport_49563.zip](https://github.com/OpenIntelWireless/itlwm/files/11574368/bugreport_49563.zip)
chive**

**Kext Download Source**

zxystd, 2023-05-26:
Please try to disable 40mhz channel width on the router and then try again, if the problem also happens, please check your wifi password.

Human79, 2023-05-26:
Same issue, I've limited the channel width but no dice :(

receperoglu, 2023-05-27:
yes bluetooth not working.but wifi working

Human79, 2023-05-28:
@receperoglu Can you explain how wifi works, it doesn't work for me!

receperoglu, 2023-05-28:
Can you take a look at [this link](https://osxinfo.net/eklentiler/config-plist.49340/). There is a config file that I created for my own laptop. only wifi works.

Human79, 2023-05-28:
@receperoglu  which Wi-Fi card model?

zxystd, 2023-05-29:
This card model is confirmed to work(both WiFi and BT), and the log indicated that the card and driver is working, I suggest to use another router or cellphone hotspot to test again.

closed 2023-05-29 by zxystd

Human79, 2023-05-29:
@zxystd I tested with my phone hotspot as well as another wifi network but doesn’t work. Also, to confirm, yes the driver does load but doesn’t connect. Can we reopen this issue?

zxystd, 2023-05-29:
Maybe you can check the network card connected antennas? the relative rssi value is too low.
[image]

Human79, 2023-05-29:
@zxystd The wifi card works perfectly in windows! And yes, the antennas are properly connected!

zxystd, 2023-05-29:
> @zxystd The wifi card works perfectly in windows! And yes, the antennas are properly connected!

What about the wifi signal on Windows when it is connected? can you also show me the router setting?

Human79, 2023-05-29:
@zxystd Wifi signal is fine in windows!
Here's the router settings :
[image]
[image]

Human79, 2023-05-31:
@zxystd Wifi connects on Intel-ac8265, this means that this not a router problem. Could be a model specific problem.

Human79, 2023-06-03:
@zxystd Is any patch/modification possible that can make this card work?
Thanks a ton!

receperoglu, 2023-06-03:
> @zxystd Is any patch/modification possible that can make this card work? Thanks a ton!

did you download the last kext update (3 days ago) it should work fine now

Human79, 2023-06-03:
I'm using the `2.2.0 stable version`. Doesnt seem to work for me. Can you provide a screenshot of it working? Also the .kext you are using and maybe a log to help @zxystd ?

zxystd, 2023-06-03:
@Human79 I don't know why your router not accept the 4-way handshake packets, but at least you can try this version which use the Linux way of sending packets with RS scaled rate. Related to https://github.com/OpenIntelWireless/itlwm/issues/764#issue-1188039195

[Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/11642281/Ventura.zip)

Human79, 2023-06-04:
@zxystd This seems to work for me! Thanks

Human79, 2023-07-05:
> @Human79 I don't know why your router not accept the 4-way handshake packets, but at least you can try this version which use the Linux way of sending packets with RS scaled rate. Related to [#764 (comment)](https://github.com/OpenIntelWireless/itlwm/issues/764#issue-1188039195)
> 
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/11642281/Ventura.zip)

@zxystd Can you share the macOS Sonoma version of this kext? Thanks!

zxystd, 2023-10-19:
@Human79 Now it is on main branch.

mnzaroo, 2024-09-19:
I need Sequoia verison of ac7265 kext