OpenIntelWireless/itlwm #877: Wifi dropdown and wifi menu at system settings do not relate to each other

johnpapad24, 2023-05-30:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes
**Environment**

- Kext Version: v2.20-alpha (812e37a)
- WiFi Card Model: Intel Dual Band Wireless-AC 8260
- macOS Version: 13.4

**Description**
Although the kext does work well until now there is a strange issue that the results on the wifi dropdown menu do not relate to the ones at the system settings wifi menu and generally these menus do not communicate to each other as they should especially if wifi is first disabled and then restarted.

here is a screenshot: 

[image]

In this screenshot the wifi dropdown menu shows correctly that the wifi is connected and other networks are also available and on the other hand the wifi menu on the system settings wrongly shows that the wifi is not connected and no networks are available.

**Kext Download Source**
Github official

zxystd, 2023-06-02:
please check the latest stable version.

johnpapad24, 2023-06-02:
Yes the latest stable version seemingly fixed this problem so the issue can be considered closed.

closed 2023-06-02 by johnpapad24