OpenIntelWireless/itlwm #881: Both itlwm and AirportItlwm don't work. Can only see Wi-Fi device in gfxutil.

dsanke, 2023-06-05:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: 2.2.0
- WiFi Card Model: Intel AC-3160
- PCI Product ID: 8086:08b3
- macOS Version: 12.6.6 Monterey

**Description**

I am using AMD Ryzen 3 5300U in a tiny machine without discrete GPU.
Both itlwm and AirportItlwm don't work. Can only see Wi-Fi device in gfxutil.
[image]
[image]
I tried SSDT-BRG0 to make Wi-Fi card shown as PXSX, but the same result. It don't work.

**Bug Report Archive**

I can't get HeliPort run.

**Kext Download Source**

[
](https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.2.0)

zxystd, 2023-06-05:
https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs
[image]

Follow guide to get runtime log.

dsanke, 2023-06-05:
[Log_2023-06-05_14-50-28.log](https://github.com/OpenIntelWireless/itlwm/files/11649494/Log_2023-06-05_14-50-28.log)

closed 2023-06-05 by zxystd