OpenIntelWireless/itlwm #883: 大佬，月亮谷有适配计划吗

ghsssss, 2023-06-06:


zxystd, 2023-06-07:
等。itlwm.kext先顶着。

qcTimb3rlandqc, 2023-06-08:
> Apple had deleted the whole legacy 80211 stack on Sonoma, I am going to turn to the other `IO80211Family` which is using by macintosh of late 2018(although it will be also deleted in 2~3 years), it is a big work, almost all of the airport interfaces and network interfaces changed, need to start reversed over again. Just wait, I don't have too much spare time to diverse.
> 
> [image]
> 
> You can use `itlwm.kext + Heliport.app` instead for the while.

if you need tester lets me no ;)thx for work perfect job :)

ameenjuz, 2023-06-11:
@zxystd 
any progress AirportItlwm.kext for Sonoma support?
I missed this kext for Sonoma and I m waiting

zxystd, 2023-06-12:
New IO80211Family is loaded. but it is just a beginning...
[image]

cheesycod, 2023-06-13:
is there anything newer than the 2018 80211?

zxystd, 2023-06-14:
[image]
Now network controller start success and begin to exchange informations.

The IOCTLs almost rewrite, hundreds of IOCTL structs should be reversed...

cheesycod, 2023-06-15:
any chance of making a ath10k driver too or smth, just asking since I know that's a pipe dream

laszlo-t, 2023-06-16:
> @zxystd any progress AirportItlwm.kext for Sonoma support? I missed this kext for Sonoma and I m waiting

the itlwm.kext works for me on Sonoma

hendrilmendes, 2023-06-16:
> > @zxystd any progress AirportItlwm.kext for Sonoma support? I missed this kext for Sonoma and I m waiting
> 
> the itlwm.kext (v2.2.0) works for me on Sonoma

what model of your wifi card?

laszlo-t, 2023-06-16:
AX201

Hendril Mendes ***@***.***> ezt írta (időpont: 2023.
jún. 16., P, 15:01):

TheDarkosGang, 2023-06-16:
> > @zxystd any progress AirportItlwm.kext for Sonoma support? I missed this kext for Sonoma and I m waiting
> 
> the itlwm.kext (v2.2.0) works for me on Sonoma (AX201)

Did you use any boot-args?

laszlo-t, 2023-06-17:
> > the itlwm.kext (v2.2.0) works for me on Sonoma (AX201)
> 
> Did you use any boot-args?

not for the wifi

Lorys89, 2023-06-17:
> [image] Now network controller start success and begin to exchange informations.
> 
> The IOCTLs almost rewrite, hundreds of IOCTL structs should be reversed...

Thank you for your commitment, how is the development progressing?

zxystd, 2023-06-20:
[image]
Now we get the wifi information.

ameenjuz, 2023-06-20:
> [image] Now we get the wifi information.
Great
I m waiting your test kext

RyzenDew, 2023-06-20:
Such fast work

Lorys89, 2023-06-20:
> [image] Now we get the wifi information.

Have you already made the scan of wifi networks work?

zxystd, 2023-06-21:
Now we get the wifi list.
[image]

Lorys89, 2023-06-21:
> Now we get the wifi list. [image]

I'm speechless, you're really the Top! 
(if you need test I have 4 hackintosh with wifi cards 9462, 9560, ax200, ax210)

Lorys89, 2023-06-21:
Today probably comes out beta 2 of sonoma, it would be nice to already start testing your kext 😝

aryajuanda, 2023-06-21:
> Now we get the wifi list.
> [image]

nice work... very good progress, 

BTW the PCI vendor is 0x15AD (Vmware) not 0x8086 (intel)

RyzenDew, 2023-06-21:
> 

they are working within vmware is my guess but they are still doing good work none the less and yes you can put wifi and BT in to vmware and have it work i just tried it

RyzenDew, 2023-06-22:
> Now we get the wifi list. [image]

any updates on when we can test the kexts?

ameenjuz, 2023-06-22:
@zxystd nice work good progress
 can you provide the kext for testing?

xjh169, 2023-06-23:
坐等各位 发kext ，我的是 Wireless-AC 8265

bilijp153, 2023-06-23:
坐等发kext，我的是 AC 9461

notAperson535, 2023-06-23:
> Now we get the wifi list. [image]

Wow, you're definitely making progress!

notAperson535, 2023-06-23:
And to all the people asking to test the kext, usually whenever they commit the new changes the kext appears in releases as a prerelease. So just wait until they push the changes.

zxystd, 2023-06-25:
Now I am stuck at how to submit network packets into the new `IO80211Family` network stack, different with `Apple Silicon WiFi drivers`, I don't want to dig deep into the `Skywalk` stack which will almost force me rewrite the whole driver including the HAL layer, so I am  faking the network packets going through the Ethernet interface instead of `IOSkywalkLegacyEthernet` and `IOSkywalkLegacyEthernetInterface`.

Yup, I am debugging on the Vmware with a Realtek USB wireless dongle, because Intel PCIE cards can not do passthrough on macOS. So the achieving results should be sync to AirportItlwm after the entire process running smoothly.
[image]

Lorys89, 2023-06-28:
How do you proceed with the problem of network packets? Did you fix this problem?

o3p6, 2023-06-29:
Cheer on

zxystd, 2023-06-29:
Good news! Now we can connect to the internet and have a smoothly network, also have the two problem to be resolved.
- Can not DHCP, can be used by manual setting IP address.
- Can not load by OpenCore, can be loaded by manual load.

[image]
[image]

zxystd, 2023-06-30:
For the loading problem, here is the error log, I don't know if it is the error of my kext's virtual table declaration or the OpenCore kext injection, anyone have ideas? Is the SkywalkFamily located in the prelinkedKC?

[image]
[opencore-2023-06-30-041658.txt](https://github.com/OpenIntelWireless/itlwm/files/11915440/opencore-2023-06-30-041658.txt)

hendrilmendes, 2023-06-30:
I'm using opencore 0.9.4, I can test to see if the same thing happens

hendrilmendes, 2023-06-30:
Sonoma not working🥹

khaled-muhammad, 2023-06-30:
I want to know the needed skills to do something like this

zxystd, 2023-06-30:
New Progress!
Loading issue resolved, thanks for @imcdd inspired me.
The last problem is the IPv4 DHCP, will see it after weekend.

Lorys89, 2023-06-30:
@zxystd you have succeeded in this great undertaking, but will you be able to make airdrop work on intel cards?

khaled-muhammad, 2023-06-30:
@zxystd Can you please tell me what to learn to make such a driver or thing, so that I can help.

zxystd, 2023-07-03:
DHCP works!
[image]
[image]

rchunping, 2023-07-03:
> DHCP works!
> [image]
> [image]

great！

Daasin, 2023-07-03:
@zxystd Forgive me if this counts as off-topic but regarding the changes you made to be on skywalk instead of legacy family. Would that impact stuff like handover & continuity? 
(I know you said you disabled handover by default as of ventura  but I'm hoping there's a boot-arg flag that we can set if we don't mind less speed and more instability to get closer to native features 🤞🏽) 

And in anycase, thank you for your efforts on this, it truly makes a huge impact on the community and we are very grateful for your work. 🙏🏽

o3p6, 2023-07-03:
> 

？

zxystd, 2023-07-03:
> @zxystd Forgive me if this counts as off-topic but regarding the changes you made to be on IO80211Family (or skywalk) instead of the legacy family. Would that impact stuff like handover & continuity?
> 

It won't have an impact, I spoof the skywalk transport layer, not the upper controller layer, so it is also a airport network controller.

> (I know you said you disabled handover by default as of ventura but I'm hoping there's a boot-arg flag that we can set if we don't mind less speed and more instability to get closer to native features 🤞🏽)
> 

The continuity feature may be reenable in Sonoma version later, and maybe Ventura also have the chance to use IO80211Family instead of IO80211FamilyLegacy.

> And in anycase, thank you for your efforts on this, it truly makes a huge impact on the community and we are very grateful for your work. 🙏🏽

Thank you, it definitely takes me much time...

ExtremeXT, 2023-07-03:
Why not port all versions to IO80211Family (or IO80211FamilyV2)?

qcTimb3rlandqc, 2023-07-03:
> Great job, when the release of this version to start with testing?

Never for asking personne

zxystd, 2023-07-04:
[AirportItlwm-Sonoma-Preview01.zip](https://github.com/OpenIntelWireless/itlwm/files/11944708/AirportItlwm-Sonoma-Preview01.zip)
The first preview version.

Lorys89, 2023-07-04:
> [AirportItlwm-Sonoma-Preview01.zip](https://github.com/OpenIntelWireless/itlwm/files/11944708/AirportItlwm-Sonoma-Preview01.zip) The first preview version.

Works great, you're the top💪

maxpicelli, 2023-07-04:
Thanks a lot my friend!!! Works fine with AX211 intel!!! It's possible that works Airdrop? I using Fenvi T-1919 together with WIFI6 native from MOBO Gigabyte Aorus Z790. If works Airdrop im retired Fenvi Pcie.

neresident, 2023-07-04:
> [AirportItlwm-Sonoma-Preview01.zip](https://github.com/OpenIntelWireless/itlwm/files/11944708/AirportItlwm-Sonoma-Preview01.zip) The first preview version.

Great work, @zxystd! I wanted to clarify, will there be a similar test update for Ventura / Monterey and other early systems? Ofc, if it's possible to release that for you. Thanks in advance!

sytao123, 2023-07-04:
Thank you for your hard work！I will try it as soon as possible.

alcheng3344, 2023-07-04:
@zxystd Thank you for your Great work on AirportItlwm，it works now 
[image]

nsby494, 2023-07-04:
非常不错，这个必须要汉字留言！！

906051999, 2023-07-04:
nice

lwx1901, 2023-07-04:
Z大，您好，首先感谢你作出的重大贡献，测试中发现一个问题：就是Hackintool检测到无线网卡没有内建？不知道是个案还是普遍问题。
[image]

szp88, 2023-07-04:
是的。军刀里显示没有内建√是没有打上的。

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2023年07月04日 23:24 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenIntelWireless/itlwm] macOS Sonoma Support (Issue #883) |

Z大，您好，首先感谢你作出的重大贡献，测试中发现一个问题：就是Hackintool检测到无线网卡没有内建？不知道是个案还是普遍问题。

maxpicelli, 2023-07-04:
> Z大，您好，首先感谢你作出的重大贡献，测试中发现一个问题：就是Hackintool检测到无线网卡没有内建？不知道是个案还是普遍问题。 [image]

It's possible edit this with plist. extracted from Hackintool? 

Mine show up this:
[image]

bilijp153, 2023-07-04:
[image]
一样，我的也没有内建

bilijp153, 2023-07-04:
[image]
系统属性里还是能瞧见的

al5710, 2023-07-04:
ax210 work

[17 more replies]

closed 2023-10-18 by zxystd