OpenIntelWireless/itlwm #888: Problem with itlwm

Dicheto2302, 2023-06-12:
Have You Read Our Docs
Yes

Are You Reporting A Bug
Maybe Yes

Environment

Kext Version: 0.8.2
WiFi Card Model: AC 3165
PCI Product ID: Idk
macOS Version: 13.4
Description
After I reseted my nvram the OC disappeared as a boot entry. I fixed it, but the Airportitlwm is not working. OC configurator says that it's imported in config.plist and all the steps in troubleshoot menu are tried.

Bug Report Archive

It says that it is not running
Kext Download Source
My EFI came with it, but I replaced it with the official one.
Link to my EFI - https://github.com/mtahanian/EFI-Lenovo

The bluetooth is also not working, but my audio and video are working.
[Log_2023-06-12_09-55-55.log](https://github.com/OpenIntelWireless/itlwm/files/11725932/Log_2023-06-12_09-55-55.log)

Dicheto2302, 2023-06-12:
This attached log is for itlwm. I will attach the airportitlwm log soon

zxystd, 2023-06-12:
No wireless card is found from the log, you should check your EFI, not itlwm. Thank you.

closed 2023-06-12 by zxystd

Dicheto2302, 2023-06-12:
What can cause this problem, because my card is dual band and I have bluetooth working?

Dicheto2302, 2023-06-12:
I have my Intel SGX Software controlled
TPM disabled
VT - Disabled
And mostly everything disabled