OpenIntelWireless/itlwm #894: Cosmetic suggestion only...  NOT A ISSUE......

hectorgrom, 2023-06-27:
Hello everybody
First of all, I would like to congratulate the team for the wonderful work done, you are all to be congratulated...
I don't come to report a bug, just a suggestion, which maybe went unnoticed, just as a complement to the wonderful work you guys are doing...
Due to the support offered by the project, I purchased a compatible AX210 wifi card from our Chinese brothers "https://tinyurl.com/2xyeucz8"...
Works perfectly, both WIFI and Bluetooth...
However, just as a cosmetic, the card has an activity "led", which works on windows and linux, however, on Mac, using your driver, it does not work, maybe it is something simple to implement, or not, I would just like to leave my suggestion for the future implementation of this feature...
Not that it's fundamental, it's working perfectly the way it is, but, if possible, it would be perfect...

Grateful

Congratulations.....

zxystd, 2023-06-30:
Maybe I can try to implement this feature, would you mind help to testing?

hectorgrom, 2023-06-30:
Of course not, I would be happy to help .... you can count on me .... :)

zxystd, 2023-06-30:
@hectorgrom What's the macOS version you are using? I have made some changes, led may be work.

hectorgrom, 2023-06-30:
hello .... I use Monterey .... waiting for the final version of Sonoma to migrate to the latest system ....

zxystd, 2023-06-30:
[Monterey.zip](https://github.com/OpenIntelWireless/itlwm/files/11917881/Monterey.zip)
You can test this one, may have a LED working, although it is not behaves the same with Windows/Linux.

hectorgrom, 2023-06-30:
I tested the file (clover/kexts/12), it works like the normal version, the wifi works normally, however, without an activity "led"....

zxystd, 2023-06-30:
[Monterey.zip](https://github.com/OpenIntelWireless/itlwm/files/11918116/Monterey.zip)
I forgot something, try this one.

hectorgrom, 2023-06-30:
nope not yet....  :(

hectorgrom, 2023-06-30:
Just to show you how it works on windows, I'm turning the WIFI on and off

https://github.com/OpenIntelWireless/itlwm/assets/61702551/19c858b4-c3c8-4f6d-ac6b-7ff8600aa2ba

zxystd, 2023-07-14:
> Just to show you how it works on windows, I'm turning the WIFI on and off, I just can't say for sure if the led corresponds to "activity" or just "status"...
> 

Does the LED working on Linux?

hectorgrom, 2023-07-14:
Hi....
Yes, it works the same on Windows 11 and Linux (Ubuntu 22.04.2 LTS)