OpenIntelWireless/itlwm #896: AirportItlwm for Sonoma version feedback.

zxystd, 2023-07-04:


ThePenguinUser, 2023-07-04:
[image]
So far, so good on my KBL laptop and Intel 8265 adapter. No delay while connecting and everything works nice and fast.

maxpicelli, 2023-07-04:
Everything works fine with AX211 intel Wireless

[image]
[image]
[image]
[image]

qcTimb3rlandqc, 2023-07-04:
location,imessage not working for now!!!

ameenjuz, 2023-07-04:
@zxystd  great hard work thanks
anyony have working Airdrop on Sonoma?

Shaneee, 2023-07-04:
AX200 working well. Great work.

[image]

bliss95, 2023-07-04:
It works properly. Thank you for your support.

[image]

xav1erwang, 2023-07-04:
**Preview的内核扩展在macOS sonoma beta2（23a5276g）上正常工作！感谢！**

我的测试平台：Intel i9-13900，MSI B760M，BCM94360CD PCIEx1接口和USB通道的蓝牙，运行iOS17 beta2的iPhone12 Pro Max和iPadOS 17 beta2的iPad Pro 2021 Apple M1，只使用了AirportItlwm.kext，蓝牙没有驱动，我使用BCM无线网卡的蓝牙。

我测试了一些内容
测试通过：
`· 基于蓝牙的handoff功能使用正常⬇️`

[image]

`· 连接正常，WiFi 6的160Mhz频宽运行正常，这让握手速度达到了1732Mbps ⬇️`

[image]

`· 搜索SSID，识别iPhone个人热点正常⬇️`

[image]

`· 隔空投送可以发现其他设备⬇️`

[image]

                 发现一些问题：
`· 隔空投送运行不正常，无法发送和接收内容。有时iPhone不能搜索到设备，需要打开一次隔空投送开关，然后iPhone可以搜索到它了，但iPad始终不能搜索到Mac（我记得之前我买过其他需要打驱动的博通卡，它的问题是iPad可以正常隔空投送但是iPhone不行，不知道这条信息能不能帮到你，我可以提供测试）`

`· 隔空播放内容接收器（向Mac流式传输音频和视频画面）无法搜索到这台电脑（不排除是Sonoma的BUG）`

`· 无法通过Mac唤醒iPhone的个人热点，iPhone似乎已经切换到蜂窝网络，但个人热点始终不能连接⬇️`
[image]

```
· （我不确定这是不是一个软件问题）我的内网有一个服务器，我通过内网服务器测试Wi-Fi最大速度，我发现它比预期的慢，同样的距离，同样是WiFi6。
macOS： ⏫ 339Mbps ⏬ 419Mbps
iPhone：  ⏫641Mbps  ⏬ 556Mbps
```
macOS测试结果⬇️
[image]
iPhone测试结果⬇️
[image]

暂时没有更多问题，没有发现死机，内核恐慌等任何严重问题，没有对系统产生任何负面影响，非常稳定，最后非常感谢您最黑苹果这一社区的重要贡献，非常感谢！如果需要任何测试，请随时联系我，我的习惯让我始终会更新到最新的版本，即使是预览版和测试版，所以如果有需要我乐意提供协助，再次感谢您的付出和贡献❤️

mikejrh, 2023-07-04:
Will the  AX5400 Wireless WiFi 6E PCIe Card (has AX210 module) work for a desktop hackintosh running Sonoma?

oq-x, 2023-07-04:
Interesting discovery
I'm using both an Intel AC7265 and Fenvi T919
The Fenvi's bluetooth works but Wi-Fi doesn't
The Intel's Wi-Fi works but bluetooth doesn't (no kext)

I try to use airdrop
and I can see my iPhone
[image]

and I can also see my hack
[image]

However I can't transfer files

Daasin, 2023-07-04:
Got one-way airdrop but I think that was normal anyway, 
find my app works fine, WiFi is connected over 5ghz and 2.4ghz, although weirdly enough continuity isn't working but I can't rule out other causes behind it so I'll wait and see if others are facing similar issues with using phone as webcam/mic

Other than that, everything is as expected. WiFi works in the default menu 👍

maxpicelli, 2023-07-04:
> @zxystd great hard work thanks anyony have working Airdrop on Sonoma?

Very hard this work with intel but not impossible. Maybe is necessary the kexts missing on Sonoma, like Fenvi. Need that at same as Broadcom needed, I think.

igarashikenshin, 2023-07-04:
Airportitlwm.kext won't make Wi-Fi work without pre-creating the ethernet interface with itlwm.kext
[image]

ameenjuz, 2023-07-04:
Imessage is very slow 
I think that this issue is occured due to builtin false in ioreg

laszlo-t, 2023-07-04:
I cannot connect to Android hotspot (wpa2). Should it work?
but anyway it can connect to my router

Kwokping, 2023-07-05:
wpa2企业级的wifi无法连接

ndszz, 2023-07-05:
iMessage not working though it works with itwlm + heliport on same machine with same efi just switched kexts. also one 5G network I use it won't connect to 

[image]

Allanrfox, 2023-07-06:
as I'm currently updating to beta 3 more updates are soon to follow but in the short time I've had the airportitlwm preview installed I've noticed a couple things, first it actually fixed my annoying grievances with itlwm (refused to connect to 5ghz networks)

[image]

I see you got airport detection working, hope it doesn't disturb anything on the stability side but it's nice you got that working

[image]

Download speeds are impeccable, full performance on both upload and download
[image]
[image]

BUT it did break iMessage, not FaceTime surprisingly 

[image]

Also seems OTA updates have an issue swell, though I'm not sure if its just the device being weird or airportitlwm's fault, will report back

[image]

otherwise, good job mate, keep at it

mikailreialt, 2023-07-06:
[image]

Working as intended on 2.4G and 5G weird cases are when changing between SSIDs when Connected it just randomly disconnect and reconnect for a few moment (note that I'm using Intel Wireless AC 8260

[image]

Allanrfox, 2023-07-06:
intel AC 8265 here location seems to be bugged on find my, works on any other app though

maxpicelli, 2023-07-06:
To prevent erros on update, insert this boot args on NVRam Opencore:  revpatch=sbvmm 

And after reboot all SMBIOS will show Update.

Run terminal force update:

sudo softwareupdate -i -a -R

Good Luck

Airportitlwm working fine on Sonoma Beta 3 
[image]

bilijp153, 2023-07-06:
[image]
hackintool显示网卡没有内建这是咋回事

qasim799, 2023-07-06:
> To prevent errors on update, insert this boot args on NVRam Opencore: **revpatch=sbvmm**
> 
> And after reboot all SMBIOS will show Update.
> 
> Run terminal force update:
> 
> sudo softwareupdate -i -a -R
> 
> Good Luck
> 
> Airportitlwm working fine on Sonoma Beta 3 [image]

Tried but it's not showing beta 3 update, I'm on beta 2 btw

mikailreialt, 2023-07-06:
> Tried but it's not showing beta 3 update, I'm on beta 2 btw

try to download the package manually

qasim799, 2023-07-06:
> try to download the package manually

I can but wondering why the system update not working since beta 1 and also those .pkgs are huge in size

mikailreialt, 2023-07-06:
> > try to download the package manually
> 
> I can but wondering why the system update not working since beta 1 and also those .pkgs are huge in size

weird, im using the OTA updates is working...

qasim799, 2023-07-06:
> > > try to download the package manually
> > 
> > 
> > I can but wondering why the system update not working since beta 1 and also those .pkgs are huge in size
> 
> weird, im using the OTA updates is working...

I'm tried macbookpro 15,2 and 15,4 SMBIOS, are you using any specific kexts or what's your opencore version?

mikailreialt, 2023-07-06:
> I'm tried macbookpro 15,2 and 15,4 SMBIOS, are you using any specific kexts or what's your opencore version?

using 15,1 smbios and OC 0.9.3

mikailreialt, 2023-07-06:
> I'm tried macbookpro 15,2 and 15,4 SMBIOS, are you using any specific kexts or what's your opencore version?
[image]

Daasin, 2023-07-06:
Can confirm the issues stated by the others with iMessage (super slow even with ethernet set as main at over a gigabit download and >100Mbit Upload)

It started with the kext but I assumed it was because it was trying to send through my iPhone or something. Seems like it's defaulting to WiFi. Not sure about the whole built in thing though but if there's anyway I can test it then feel free to let me know.

qasim799, 2023-07-06:
Bluetooth stopped working after beta 2 update intel 8260 weird it shows NULL state OFF

melxman, 2023-07-07:
> Bluetooth stopped working after beta 2 update intel 8260 weird it shows NULL state OFF

https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/437#issuecomment-1579931908

Please try this one. This kext is also the author's self-compiled version. Comment out the two parameters related to bluetooth in nvram.
I tested it myself, 14beta3 is using bluetooth normally

Lorys89, 2023-07-07:
@zxystd Sorry to report that the builtin modification didn't work for imessage
 :(

alighafourian, 2023-07-07:
https://github.com/OpenIntelWireless/itlwm/issues/883#issuecomment-1625204187

in last release for sonoma Builtin fix but imessage dont send message :
[image]
[image]

qasim799, 2023-07-07:
> > Bluetooth stopped working after beta 2 update intel 8260 weird it shows NULL state OFF
> 
> [OpenIntelWireless/IntelBluetoothFirmware#437 (comment)](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/437#issuecomment-1579931908)
> 
> Please try this one. This kext is also the author's self-compiled version. Comment out the two parameters related to bluetooth in nvram. I tested it myself, 14beta3 is using bluetooth normally

Checked through kextstat the intelbluetoothfirmware.kext is not loading idk why it's in OC and listed properly config.plist I've tried many versions even compiled 2.3.0

dawalishi0821, 2023-07-07:
Killer1690i is working properly, thank you.
[image]

mikebeaton, 2023-07-07:
**Environment**

- Kext Version: AirportItlwm-Sonoma-Preview01.kext 
- WiFi Card Model: Intel(R) Wireless-AC 9560 160MHz
- PCI Product ID: 8086:A370, subsystem 8086:0030
- macOS Version: 12.6.8 (21G716)

**Description**
This alpha version is working fine for me on a slower network (~70Mbps up and down). On a much faster network (~200-400Mbps up and down, from the test PC location; up to ~700Mbps if very close to router (using different laptop)), it does not connect. The same network connects fine from Windows and Linux (Fedora), and also connects fine from macOS Monterey with your AirportItlwm (many thanks!), though with [slower than expected upload speeds](https://github.com/OpenIntelWireless/itlwm/issues/898).

The failure on Sonoma is that, if connecting from the WiFi icon, it tries for a bit then asks for the password again, even though it was correct (double and triple checked!). If connecting from Settings, it fails with the following message, without asking for the password again:

[image]

**Bug Report Archive**
[Log_2023-07-07_22-32-55.log](https://github.com/OpenIntelWireless/itlwm/files/11986397/Log_2023-07-07_22-32-55.log)

alders-lakes, 2023-07-08:
Is 802.x1 supposed to work?

906051999, 2023-07-08:
Intel AX201  + Sonoma preview 2
(Asus z490 + 10700k)

wifi can work
Bluetooth can't open

[image]

Daasin, 2023-07-08:
I don't seem to be able to boot with version 2, going to try updating the Bluetooth stuff and changing load order to see if that helps but currently stuck at apple logo with no loading bar 

(Sonoma Beta 3 with Airportitlwm Beta 2) the first one worked albeit not as built-in and with slow iMessage

wlb123git, 2023-07-09:
Mine is 9560 , too. But I can't even turn on WIFI

PCI Product ID: 8086:A370, subsystem 8086:0034

[mikebeaton]'s commits show that 0030 does work, and mine 0034 won't work.

Kwokping, 2023-07-10:
> [#883 (comment)](https://github.com/OpenIntelWireless/itlwm/issues/883#issuecomment-1625204187)
> 
> in last release for sonoma Builtin fix but imessage dont send message :
> 
> [image]
> 
> [image]
the same issue。

anonymous-writer, 2023-07-12:
For "INTEL NIC WI-FI 6 AX210 2230" new Sonama Kext works perfect. Many thanks for the work.

[image]

johnpapad24, 2023-07-12:
Intel 8260 seems to work fine with preview 2 right now

username11384, 2023-07-13:
> To prevent errors on update, insert this boot args on NVRam Opencore: **revpatch=sbvmm**
> 
> And after reboot all SMBIOS will show Update.
> 
> Run terminal force update:
> 
> sudo softwareupdate -i -a -R
> 
> Good Luck
> 
> Airportitlwm working fine on Sonoma Beta 3 [image]

How did you manage to show the old "About This Mac" and customise it? i'm kinda curious

maxpicelli, 2023-07-13:
https://github.com/0xCUB3/About-This-Hack 
Download release on beside

qasim799, 2023-07-13:
it's working great on sonoma beta 3 but wifi randomly disconnects sometime usually when under heavy load or downloading,
my system is a dell latitude 7400 with i5-8365U and intel AC 8265 wifi, logs attached
[Log_2023-07-13_21-08-32.log](https://github.com/OpenIntelWireless/itlwm/files/12041540/Log_2023-07-13_21-08-32.log)

xjh169, 2023-07-14:
> 我的小米笔记本Pro 15.6 2019，i78550U，KBL平台和Intel 8265，5G估计无法连接。目前2.4G 20Hz正常使用。 [image]

[image]
[image]
[image]
z
[Log_2023-07-14_10-53-33.log](https://github.com/OpenIntelWireless/itlwm/files/12045886/Log_2023-07-14_10-53-33.log)

之前的问题 附上log 和详细截图。

5T33Z0, 2023-07-14:
[image]

Working fine on my Lenovo T490 with an Intel AC-9650.

[Log_2023-07-14_07-31-13.log](https://github.com/OpenIntelWireless/itlwm/files/12046519/Log_2023-07-14_07-31-13.log)

ameenjuz, 2023-07-14:
anyone have iMessage working on Sonoma beta 3 using AirportItlwm-Sonoma-Preview02.kext?

IllSaft, 2023-07-14:
Working but without iMessage 
[Log_2023-07-14_10-21-01.log](https://github.com/OpenIntelWireless/itlwm/files/12051322/Log_2023-07-14_10-21-01.log)

maxpicelli, 2023-07-14:
Working fine and good after compiled all kexts with Xcode
Sonoma Beta 3 
Intel BT and Wifi 6E AX211

[image]

ameenjuz, 2023-07-14:
> Working iMessage and FaceTime fine and good after compiled all kexts with Xcode Sonoma Beta 3 Intel BT and Wifi 6E AX211
> 
> [image]

Please share your compiled kext  I will try because I have not working iMessage on Sonoma but on Ventura working fine iMessage facetime
And tell me how to do it

ndszz, 2023-07-14:
> > Working iMessage and FaceTime fine and good after compiled all kexts with Xcode Sonoma Beta 3 Intel BT and Wifi 6E AX211
> > [image]
> 
> Please share your compiled kext I will try because I have not working iMessage on Sonoma but on Ventura working fine iMessage facetime And tell me how to do it

Do you have a working Ethernet connection?

maxpicelli, 2023-07-14:
Hi All

im Max.1974 user from insanely Mac. I will post on feel minutes there on Sonoma Release forum. Here they considere like spam. Compiled kexts is necessary to do on your own Hardware. Need LILU debug SDK and Xcode. Not difficult.

maxpicelli, 2023-07-14:
[image]

IllSaft, 2023-07-14:
> Hi All
> 
> im Max.1974 user from insanely Mac. I will post on feel minutes there on Sonoma Release forum. Here they considere like spam. Compiled kexts is necessary to do on your own Hardware. Need LILU debug SDK and Xcode. Not difficult.

Perfect, I will give it a try now. I am still new at all of this
Maybe you can guide a little? 
I am setting up the Lilu SDK and try Compiling :)

Let me know, thanks!

maxpicelli, 2023-07-14:
I have a friend mine Henrique Nonato and he has a Youtube channel. 

"Hackintosh and Beyond"  - is in Portuguese but you will understand perfectly. 

He is a great Brazilian guy graduated in TI and know about all compilation 

[image]

maxpicelli, 2023-07-14:
[https://www.insanelymac.com/forum/topic/356881-pre-release-macos-sonoma/?do=findComment&comment=2808282](url)

maxpicelli, 2023-07-14:
Here is my EFIs

maxpicelli, 2023-07-14:
https://www.insanelymac.com/forum/topic/356881-pre-release-macos-sonoma/?do=findComment&comment=2808282

[127 more replies]

closed 2023-10-18 by zxystd