OpenIntelWireless/itlwm #897: how to enbale 802.11AX in intel AX200

jie-d-cheng, 2023-07-07:
as mentioned in FAQ,

Does it support 802.11AC and 802.11AX
Yes, 802.11AC and 802.11AX (currently disabled) support was implemented since v2.0.0.

now it is v2.2, so could 802.11AX be enabled when ax200 used, and how? 

thanks.

dawalishi0821, 2023-07-07:
> as mentioned in FAQ,
> 
> Does it support 802.11AC and 802.11AX Yes, 802.11AC and 802.11AX (currently disabled) support was implemented since v2.0.0.
> 
> now it is v2.2, so could 802.11AX be enabled when ax200 used, and how?
> 
> thanks.

This is my method, I hope it can help you
---
Firstly, you need to download the [source code](https://github.com/OpenIntelWireless/itlwm), [MacKernel SDK](https://github.com/acidanthera/MacKernelSDK)and[Xcode](https://developer.apple.com/xcode/);

Secondly, place the MacKernelSDK in the source code folder (the original code folder of the MacKernelSDK must be named 'MacKernelSDK', not 'MacKernelSDK master');
[image]

Thirdly, open 'itlwm. xcodeproj' with xcode;

Fourthly, use search tools to search for 'HEON';
[image]

Fifth, delete the preceding '//' in the code;
[image]

Sixth, choose to compile the project as airportitlwm and compile it；
[image]

Finally, you will find the editing result '/Users/XXX/Library/Developer/Xcode/DerivedData/itlwm XXX/Build/Products/Debug' in this path.
[image]

I hope it can help you. Good luck to you.

jie-d-cheng, 2023-07-08:
yep, I make it following your great guide. thanks, buddy.

[image]

Dragster27, 2023-07-08:
Hi there. Thanks to [dawalishi0821] I was able to enable AX support and compile the driver, however.. I cannot connect with it to my Wifi6 WPA2/WPA3 Personal network... I keep getting "invalid password"... the regular release works just fine with no changes in the network... Did it work for you [jie-d-cheng]?

Cheers,
D.

jie-d-cheng, 2023-07-09:
> Hi there. Thanks to [dawalishi0821] I was able to enable AX support and compile the driver, however.. I cannot connect with it to my Wifi6 WPA2/WPA3 Personal network... I keep getting "invalid [...]

yes, it works well in my network,but mine is wap2 auth only.

anonymous-writer, 2023-07-10:
Also works with the "INTEL NIC WI-FI 6 AX210 2230". Thanks, great job. 

[image]

closed 2023-07-11 by zxystd

dawalishi0821, 2023-07-23:
> 这个macos14可以用吗 [image] 我是ax200

等z大把14的airportitlwm源码放出来再说

Tec0D3v, 2024-05-15:
i have a dell latitude 5510

"I have an Intel / Wi-Fi 6 AX201 160MHz. What would be the case to compile it and make it recognize it?"

form hackintool
[image]

[image]

lowprofileusername, 2025-04-12:
so i have ax210. after  enbale 802.11AX my tx limited to 140mbps. before it was ok. anyone has any idea why?
 i have a be router with me . this bother me for a day.

[image]

so if i use kext without ax enable. the tx is showing normal.