OpenIntelWireless/itlwm #898: Intel(R) Wireless-AC 9560 160MHz Monterey slow upload speed

mikebeaton, 2023-07-07:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v2.2.0_stable_Monterey or Monterey-v2.3.0-DEBUG-alpha-bd47afe 
- WiFi Card Model: Intel(R) Wireless-AC 9560 160MHz
- PCI Product ID: 8086:A370, subsystem 8086:0030
- macOS Version: 12.6.8 (21G716)

**Description**
This card connects fine to a slower network ~70Mbps up and down. On a much faster network (~200-400Mbps up and down, from test test PC location; up to ~700 if very close to router (using different laptop)), then Windows and Linux (Fedora) achieve the stated speeds, but on macOS with the above drivers, download speed is as expected, but upload is much slower than in Windows and Linux (~20-70Mbps). This is repeatable. Also of note, the problem is worse in the 2.3.0 alpha driver, with an even slower upload  speed (~7-10Mbps). Again, this difference is repeatable, and always faster back in Windows and Linux.

**Bug Report Archive**
v2.2.0_stable_Monterey: [Log_2023-07-07_21-34-39.log](https://github.com/OpenIntelWireless/itlwm/files/11986340/Log_2023-07-07_21-34-39.log)
Monterey-v2.3.0-DEBUG-alpha-bd47afe: [Log_2023-07-07_21-52-09.log](https://github.com/OpenIntelWireless/itlwm/files/11986341/Log_2023-07-07_21-52-09.log)

**Kext Download Source**
/releases

mikebeaton, 2023-07-12:
> Try disabling IPV6.

Set 'Link-local only' for IPv6 on Wi-Fi. No effect on this issue.

Issue #805 about IPv6 is for slow download speed; I actually have full (and v. fast; and matching Windows and macOS) download speed, but slow upload speed with itlwm (as compared to much faster upload speed on Windows and macOS).

Should probably add:
 - The slow speed actually gets progressively slower the more speed tests you run (whereas download gets marginally faster, if anything)

zxystd, 2023-07-13:
@mikebeaton Have you tried this one? #764

mikebeaton, 2023-07-13:
No, I will happily test - thank you. Away from Mac for a couple days, but will test when back.

zxystd, 2023-10-19:
Now it is on the main branch, you can download it on alpha release page.

closed 2023-10-19 by zxystd