OpenIntelWireless/itlwm #899: a

KemonDream, 2023-07-10:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: 2.2.0 2.3.0
- WiFi Card Model: ax200 ax210
- PCI Product ID:  0x2725
- macOS Version: 13.4.1

**Description**
- 求大佬解惑。我搭建了一个ryzentosh，然后一开始用了一块bcm94360ng有人说是原生支持的，然后只有蓝牙能用，wifi打不开，尝试了各种bcm驱动也无济于事。只得换成了ax200，安装了这个wifi驱动还有蓝牙驱动，蓝牙的id都是写在支持列表里的，oc也参照配置好了的，但是开机就是没有。cpu5900hx 有以太网 usb igpu全部都能用，但是wifi蓝牙死活打不开。看了log也不知道为什么会驱动不了。后面又换了ax210，问题依旧。windows下所有网卡一点问题没有，同样的硬件和bios设置。重写了一遍oc配置也无效，尝试禁用所有非启动必要的驱动也无效。实在是因为隔空投送对于我的使用场景比较重要，还是想能解决这个问题最好。
- bios里快速启动肯定关了，CNVi Support没有找到。网络设置也清除过了。
- 现在状态就是在军刀里看驱动是加载了的kextstat | grep -i "itlwm"这个命令也是正常输出，但是打开heliport会提示itlwm没有运行。系统信息蓝牙页显示一个null博通的，wifi页没有硬件信息。
- 辛苦看一下日志

**Bug Report Archive**

[bugreport_42290.zip](https://github.com/OpenIntelWireless/itlwm/files/11998455/bugreport_42290.zip)

**Kext Download Source**
github项目页面

zxystd, 2023-07-10:
[image]
EFI的问题，自己找原因吧。

zxystd, 2023-07-10:
或者你找别人的跟你相似主板的EFI试试

closed 2023-07-10 by zxystd

KemonDream, 2023-07-10:
能问一下是哪个部分的问题，kext还是 acpi 还是那个device property，找疯了也搞不懂是什么地方不对头，换了几个人的引导文件了不知道是哪里没调对，搜索引擎也不知道该搜索哪方面内容，求大佬指个路