OpenIntelWireless/itlwm #901: Connectivity Issues with AC 9560 and Mac Recovery

JayTurnsek, 2023-07-22:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
I might be

**Environment**

- Kext Version: v2.2.0
- WiFi Card Model: Intel Wireless-AC 9560
- PCI Product ID: 8086:9DF0 Subsystem 00308086
- macOS Version: Monterey (12)

**Description**
Using a Lenovo ThinkPad T490s and Monterey, cannot connect to a Wi-Fi in mac recovery mode despite them being listed in the Wi-Fi menu. Eventually one of a few things happen if I try to connect to a network:

1. It tells me to move closer to the router
2. It repeatedly prompts me for the wifi password
3. It says "Your Mac successfully joined the Wi-Fi network, but cannot reach Internet"

Both AirportItlwm and IO80211Family are running as reflected in Kextstat command, securebootmodel is set to default and I am forcing IO80211Family to load in my config. Am I missing a BIOS setting for networks or some other Kext? First hackintosh, this is the only snag :( I also attached my config as a hail mary.
[config.plist.txt](https://github.com/OpenIntelWireless/itlwm/files/12137177/config.plist.txt)

 The SAME issues arising with Big Sur and Ventura, so I know it's not dependent on that.

**Bug Report Archive**
I can't do this, I'm stuck in recovery; If there is a way to get this from recovery please let me know.

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.2.0, not custom built

zxystd, 2023-07-24:
What's the Wi-Fi setting? Are you sure the password is correct?  By the way, you can also get the itlwm running log in recovery mode.

JayTurnsek, 2023-07-24:
What do you mean by whats the wifi setting? and yes I am sure, and have checked with 3 different networks so it must not be that. How do I get the log in recovery mode?

Thanks for the reply btw!

zxystd, 2023-07-24:
> What do you mean by whats the wifi setting?
>
wifi setting is like this:
[image]

> How do I get the log in recovery mode?

https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs

JayTurnsek, 2023-07-24:
I'm assuming that screenshot is from Linux, since I have windows I attached what I think you're looking for from my two networks; please let me know if I'm wrong.
[image]
[image]
Also, the terminal in recovery mode did not recognize the ```sudo``` command and said "location not found" when I inputted ```/System/Library/PrivateFrameworks/Apple80211.framework/Versions/A/Resources/airport debug +AllUserland +AllDriver +AllVendor```. As a workaround, I called ```dmesg | grep -E "itlwm|Airport|IO80211|EAPOL"``` and attached a picture of the logs below as well.
I understand that is unorthodox, so please let me know if you want me to do something different!
[image]
[image]

JayTurnsek, 2023-07-24:
Update: Was able to "connect" to my hotspot network, but it didn't actually connect and had the exclamation mark on the wifi symbol. See below:
[image]

I was also able to finally get the log FILE from this occurrence, and that is also attached below.
[Log_itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/12149648/Log_itlwm.log)

Thank you!!

JayTurnsek, 2023-07-25:
An update to things I tried to rule them out:
- Ensured fast boot was off in windows and BIOS
- Flipped antennas on physical wifi card
- Downgrading to Catalina 
- Various version downgrades matching OS tested (Catalina, Big Sur, Monterey, Ventura)
- Force/don't force IO80211 to run
- SecureBootMode j137 and Enabled
- Deleted system network preferences 

Still just prompting for password over and over. Hoping there's a fix for this for others with same issue!

zxystd, 2023-07-26:
[image]
Tested with Ventura recovery and it works fine(and some people had confirmed that), I don't know what's wrong there, but it is most likely not the driver issue.

Does it works fine on macOS(not recovery)?

zxystd, 2023-07-26:
From the log it seems that scaning is not always working, sometimes it is stuck at somewhere, and never return.
Have you tried another's EFI for testing?

zxystd, 2023-07-26:
And can you try to connect to an Open Wi-Fi?

JayTurnsek, 2023-07-26:
This might not be related at all, but iPhone USB tethering works fine. Eating up my data fast though

zxystd, 2023-07-26:
You can downgrade to v2.1.0 or v2.0.0, since we had upgrade 9xxx firmware to -46 version, maybe there have something compatible issue. If it not help, maybe the active scanning affect that, will post you a mod version to test later.

zxystd, 2023-07-26:
hmmm, you are on Ventura now, maybe you can download a Monterey recovery OS?

JayTurnsek, 2023-07-26:
Thank you for the planned mod version! And would downgrading to Monterey help?

zxystd, 2023-07-26:
a Monterey recovery OS is enough, you can test connection on there.

JayTurnsek, 2023-07-26:
Ok great. Will set that up.

zxystd, 2023-07-26:
[Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/12175612/Ventura.zip)
This is a mod version which disabled the active scanning feature.

JayTurnsek, 2023-07-26:
[
[image]
](url)

Attempted with new mod version, same result. Constant pop up like this

andreszerocross, 2023-10-08:
@zxystd  i have same experienced. My notebook is Lenovo T490 not T490S. But using same wifi adapter, Intel AC 9560 (PCI Product ID: 8086:9DF0 Subsystem 00308086) / 100% same Wifi modul.

1. Use 2.1.0, 2.2.0 or 2.3.0 Airportitlwm.kext. I can't connect to any wifi. I remove password of wifi, but still can't connect. it showing popup about diagnostic because fail to connect. Ventura and sonoma is same
2. Same like OP, i can't connect in Recovery too
3. Using itlwm.kext from 2.1.0, 2.2.0 and 2.3.0 + heliport is still same. I can't connect. Try to turn off wifi and turn on again, then it connect but no internet connection, after few sec it will disconnect again from wifi.

If you need more information, i very happy to help

zxystd, 2023-10-08:
@andreszerocross Follow the guide to collect runtime log.
https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs

andreszerocross, 2023-10-08:
> @andreszerocross Follow the guide to collect runtime log.
> https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs

I got error message in terminal when running 1st command
[image]

[bugreport_22022.zip](https://github.com/OpenIntelWireless/itlwm/files/12839845/bugreport_22022.zip)

andreszerocross, 2023-10-08:
And for the information, this AC9560 can't detect SSID for channel 149 (5GHz, 80MHz).

andreszerocross, 2023-10-08:
Sorry, i use 'itlwm_cc=US -novht -noht40" as boot-arg before. This is bug report without those boot arg.

[bugreport_60717.zip](https://github.com/OpenIntelWireless/itlwm/files/12839859/bugreport_60717.zip)

zxystd, 2023-10-08:
@andreszerocross Seems that the scanning is not responding, don't know why, is it works on Linux?

andreszerocross, 2023-10-08:
> @andreszerocross Seems that the scanning is not responding, don't know why, is it works on Linux?

Yes, i just try with Ubuntu and it works fine. I check the firmware. The firmware is same like your repository (9000-46)

zxystd, 2023-10-08:
@andreszerocross I increased the log level, so I can see the executing code path, please try and collect the log again.
[Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/12840055/Ventura.zip)

andreszerocross, 2023-10-08:
> @andreszerocross I increased the log level, so I can see the executing code path, please try and collect the log again.
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/12840055/Ventura.zip)

Kext is not loaded.
[bugreport_18595.zip](https://github.com/OpenIntelWireless/itlwm/files/12840550/bugreport_18595.zip)

andreszerocross, 2023-10-08:
> @andreszerocross I increased the log level, so I can see the executing code path, please try and collect the log again.
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/12840055/Ventura.zip)

Sorry, i shutdown the notebook then boot again and now kext is loaded. Here is the bugreport
[bugreport_64353.zip](https://github.com/OpenIntelWireless/itlwm/files/12840555/bugreport_64353.zip)

zxystd, 2023-10-08:
[Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/12840984/Ventura.zip)
@andreszerocross Please try this one.

andreszerocross, 2023-10-08:
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/12840984/Ventura.zip)
> @andreszerocross Please try this one.

1. Kext is loaded but no SSID is detected, i wait for 1 minutes and still no SSID is detected
2. I turn off the wifi then turn on again, now SSID is detected and i try to connect to one of SSID. But problem is still same. I try to connect to SSID without password and there is pop up message about "THe wifi network "SSID Name" could not be joined. Bla bla bla"

Here is the bugreports files
[bugreport_21218.zip](https://github.com/OpenIntelWireless/itlwm/files/12841020/bugreport_21218.zip)

andreszerocross, 2023-10-10:
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/12840984/Ventura.zip)
> @andreszerocross Please try this one.

Any clue?

andreszerocross, 2023-10-14:
> @andreszerocross I don't know if OpenBSD have the same problem(itlwm is based on it), can you try OpenBSD live system to see if the card have the same problem?

I try to use FreeBSD (i am not familiar with this OS). I can connect to the SSD 5GHZ with channel 149. But i can't connect to internet and got no ip. But in my  mikrotik it's already connect.
[image]

[image]

zxystd, 2023-10-14:
@andreszerocross itlwm is based on OpenBSD, FreeBSD only support 11abg, not the same driver... what I want to see is whether the original OpenBSD code can support your card.

https://www.openbsd.org/
OpenBSD can be more simple to install compared with FreeBSD.

zxystd, 2024-03-16:
inactive and no more people report the same issue. closed.

closed 2024-03-16 by zxystd