OpenIntelWireless/itlwm #902: Slow wifi speed than LAN

darescloud, 2023-07-25:
**Have You Read Our Docs** yes

**Are You Reporting A Bug** yes

**Environment**

- Kext Version: 2.3.0（the lastest code build）
- WiFi Card Model: 
- PCI Product ID: 
- macOS Version: 13.4.1（c）

**Description**

The display speed is 1300mpbs, and the actual spee
[image]
d measurement is less than half, about 600. wifi6(ax) is also.Here is my 
**Bug Report Archive**

**Kext Download Source**

the lastest code and Xcode bulid.

darescloud, 2023-07-25:
[image]
the actually speed

darescloud, 2023-07-25:
sorry, the first time take issue, mine is ax211 witch on the b660m mortar

zxystd, 2023-07-26:
It depends on your router, wireless can not be compared with LAN.

darescloud, 2023-07-26:
> It depends on your router, wireless can not be compared with LAN.

我上面给出了协商速率是1300mbps，但是测速只有一半，650mbps