OpenIntelWireless/itlwm #904: Issues with Nothing Phone 1 Hotspot and WPA/WEP Security when Connected via Xiaomi Mi Pad 6

BudDeuce, 2023-07-28:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version:  itlwm: 2.2.0 fw: 17.3216344376.0
- WiFi Card Model: Wireless 3165
- PCI Product ID: 0x8110
- macOS Version: 13.5 (22G74)

**Description**
When my friend is attempting to use the Nothing Phone 1's hotspot, it is necessary for him to first connect to his Xiaomi Mi Pad 6's hotspot. This leads to problems with WPA/WEP security. The system displays an error message stating that the phone was previously joined as WPA2/WPA3 Personal, not WEP. After some time, a prompt appears, asking him to enter a WEP password, but attempting to join the network using this password fails. However, when he manually selects the network option from the WiFi menu, the connection is established successfully.

**Bug Report Archive**
https://we.tl/t-8CqrgQbsTD

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.2.0/itlwm_v2.2.0_stable.kext.zip

zxystd, 2023-08-30:
Sorry, the log file is expired, can you upload here directly?

BudDeuce, 2023-08-31:
Sure, I will get the log files as soon as I can.

BudDeuce, 2023-09-02:
Here you go: https://drive.google.com/file/d/1cfoVMOvDfwBc9klRzesWtDCFpXPpxKwb/view?usp=sharing

The file was too big to upload here (264 MB), so I used Drive instead. The link shouldn't expire now.

zxystd, 2024-03-16:
Please follow this guide to get the full log.
https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs

BudDeuce, 2024-03-18:
What a ride of nostalgia, some commit fixed this issue btw.

closed 2024-03-18 by BudDeuce

BudDeuce, 2024-06-24:
Open your own issue

rkt21, 2024-06-24:
Buddeuce If you know how to solve this issue plz dm in discord.

Jkkushwaha49, 2024-06-25:
Budduece I jkkushwaha why you blocked me in discord please unblock me I can't send you any message

rkt21, 2024-06-25:
Plz unblock me also bud.