OpenIntelWireless/itlwm #905: Intel AX210 （M7633-006）Wireless network interface card cannot be driven

QianShanWoDu, 2023-08-06:
**Have You Read Our Docs**
Yes.

**Are You Reporting A Bug**
Yes.

**Environment**

- Kext Version: itlwm.kext(2.2.0 and 2.3.0 );airportitlwm.kext(2.2.0 and 2.3.0 )
- WiFi Card Model: Intel AX210 （M7633-006）
- PCI Product ID: PCI\VEN_8086&DEV_2725&SUBSYS_00248086&REV_1A
- macOS Version: Ventura13.5

**Description**
The CPU is amd5600H, and the wireless network interface card cannot be drived. itlwm prompts "未运行", and the airportitlwm end point detection background is not running either.

**Bug Report Archive**
[bugreport_56808.zip](https://github.com/OpenIntelWireless/itlwm/files/12269077/bugreport_56808.zip)
[Log_2023-08-06_17-18-22.log](https://github.com/OpenIntelWireless/itlwm/files/12269076/Log_2023-08-06_17-18-22.log)

**Kext Download Source**
(https://github.com/OpenIntelWireless/itlwm/releases)

diepeterpan, 2023-08-06:
I checked the code and the subsys you have '00248086' is in there so the card should be supported, then I also checked the kmutil command on my machine where my AX210 with a different subsys 40208086 model is working but the message "No variant specified, " seems irrelevant - kextstat | grep -i "itlwm"

[image]

This seems more closer to the error looking at the log file, do you have Windows on the same machine, check if the Wifi is working in Windows.

[    2.268482]: itlwm: iwx_set_hw_ready
[    2.273572]: itlwm: : L1 Disabled - LTR Disabled
[    2.476355]: itlwm: : acquiring device failed_

diepeterpan, 2023-08-06:
All see this - https://community.intel.com/t5/Wireless/Wifi-not-working-in-Linux-using-AX200-on-Beelink-mini-PC/m-p/1498596

My guess is the Wifi is switched off by BIOS, similar to Bluetooth issue I had, try taping the pins which will force the Wifi and Bluetooth to be on, could at least eliminate it as the source of the problem.

[image]

zxystd, 2023-08-06:
Fix your EFI please...
maybe BIOS/PCIE setting/VT-d etc.

closed 2023-08-06 by zxystd

QianShanWoDu, 2023-08-06:
谢谢，各位!

Wanchengshan, 2023-11-25:
@QianShanWoDu 请问解决了吗，我也是识别不到怎么办不知道

QianShanWoDu, 2023-11-25:
> @QianShanWoDu 请问解决了吗，我也是识别不到怎么办不知道

没有解决。我用USB WiFi了。