OpenIntelWireless/itlwm #906: Wi-Fi Disconnection Bug on Dell Latitude E6410 with Intel Centrino Advanced-N 6200 AGN

Zabloogas, 2023-08-07:
**Have You Read Our Docs**
Yes
**Are You Reporting A Bug**
Yes
**Environment**

- Kext Version: 2.2.0 Stable & 2.3.0 Alpha
- WiFi Card Model: Intel Centrino Advanced-N 6200 AGN
- PCI Product ID: 8086:422c
- macOS Version: 10.13.6 (Build 17G66)
- Device: Dell Latitude E6410

**Description**
The ping command works and some simple websites like Google load, but most sites cause the Wi-Fi to stop working then turn back on after a few seconds, and it happens instantly when I download anything. I tried downloading a file from a web server on my local network and it still does the same disconnection as it does when downloading files off the internet. The Wi-Fi light on the laptop also turns off when this happens. I have tested the Wi-Fi in Linux and it connects and downloads files on 2.4 GHz and 5 GHz, however on macOS it only connects to 2.4 GHz.

**Bug Report Archive**
[bugreport_46260.zip](https://github.com/OpenIntelWireless/itlwm/files/12271010/bugreport_46260.zip)

**Kext Download Source**
GitHub releases page

samranahmad09, 2023-09-19:
I also have the same issue on my HP Pavilion dv6t-2300 laptop

zxystd, 2024-03-16:
@Zabloogas @samranahmad09 Hi, I had add some updates on iwn related to agg frames issue, can you try v2.3.0-alpha version can resolved this problem?

samranahmad09, 2024-03-16:
Sure, I'll try in the next few days and report back