OpenIntelWireless/itlwm #909: Infinite reboots on sonoma beta5 and beta6 with AirportItlwm-Sonoma-Preview05 driver

WoChen5770, 2023-08-27:
**Have You Read Our Docs**

yes
**Are You Reporting A Bug**

yes
**Environment**

- Kext Version: AirportItlwm-Sonoma-Preview05
- WiFi Card Model: ax200
- PCI Product ID: what this?
- macOS Version: sonoma beta5 and beta6 

**Description**

Can't boot into desktop, reboot when printing logs
**Bug Report Archive**

-
**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/issues/883#issuecomment-1670749680

zxystd, 2023-08-27:
This version works fine on every user, please check EFI or system by yourself, thanks.

closed 2023-08-27 by zxystd

WoChen5770, 2023-09-03:
`
00:000 00:000 AAPL: #[EB.H.IS|!] Err(0xE) <- RT.GV boot-signature 7C436110-AB2A-4BBB-A880-FE41995C9F82
00:002 00:002 AAPL: #[EB.H.IS|!] Err(0xE) <- RT.GV boot-image-key 7C436110-AB2A-4BBB-A880-FE41995C9F82
00:003 00:001 AAPL: #[EB|H:IS] 0
00:005 00:001 AAPL: #[EB|LOG:INIT] 2023-09-03T07:15:22
00:007 00:001 AAPL: #[EB|VERSION] <"boot.efi 580~530 (Official), built 2023-08-22T03:18:41-0700">
00:008 00:001 AAPL: #[EB|BUILD] <"BUILD-INFO[392]:{"DisplayName":"boot.efi","DisplayVersion":"580~530","RecordUuid":"BD3BB405-1A2D-43F8-9AA3-FC5DD11AEC60","BuildTime":"2023-08-22T03:18:41-0700","ProjectName":"efiboot","ProductName":"boot.efi","SourceVersion":"580","BuildVersion":"530","BuildConfiguration":"Release","BuildType":"Official","Compiler":"clang-1500.0.37.1","SdkVersion":"14.0","SdkBuild":"23A331","TargetArchitectures":"x86_64"}">
00:010 00:001 AAPL: #[EB.CFG.DEV|!] Err(0xE) <- RT.GV booter-strict-xmlparser 7C436110-AB2A-4BBB-A880-FE41995C9F82
00:011 00:001 AAPL: #[EB|CFG:DEV] r5 0x0 0x0
00:013 00:001 AAPL: #[EB|H:IS] 0
00:014 00:001 AAPL: #[EB|WL:MODE] 0
00:016 00:001 AAPL: #[EB|CFG:ARG] boot-save-log 0x0000000000000002 (0x0000000000000002 < 0xFFFFFFFFFFFFFFFF) default
00:017 00:001 AAPL: #[EB|CFG:ARG] wake-save-log 0x0000000000000002 (0x0000000000000002 < 0x0000000000000002) default
00:019 00:001 AAPL: #[EB|CFG:ARG] console       0x0000000000000001 (0x0000000000000001 < 0x0000000000000001) default
00:020 00:001 AAPL: #[EB|CFG:ARG] serial        0x0000000000000001 (0x0000000000000001 < 0x0000000000000000) default
00:022 00:001 AAPL: #[EB|CFG:ARG] preoslog      0x0000000000000001 (0x0000000000000001 < 0xFFFFFFFFFFFFFFFF) default
00:023 00:001 AAPL: #[EB|CFG:ARG] timestamps    0x0000000000000000 (0x0000000000000000 < 0xFFFFFFFFFFFFFFFF) default
00:025 00:001 AAPL: #[EB|CFG:ARG] log-level     0x0000000000000001 (0x0000000000000001 & 0x0000000000000021) default
00:026 00:001 AAPL: #[EB|CFG:ARG] breakpoint    0x0000000000000000 (0x0000000000000000 & 0x0000000000000000) default
00:028 00:001 AAPL: #[EB|CFG:ARG] kc-read-size  0x0000000000100000 (0x0000000000100000 < 0xFFFFFFFFFFFFFFFF) default
00:029 00:001 AAPL: #[EB|CFG:ARG] force-error   0x0000000000000000 (0x0000000000000000 & 0x0000000000000000) default
00:031 00:001 AAPL: #[EB|H:IS] 0
00:032 00:001 AAPL: #[EB|WL] 0 0 0x01 0x01   0 0x00
00:034 00:001 AAPL: #[EB|BRD:NV] Mac-63001698E7A34814
00:036 00:001 AAPL: #[EB|B:VAw]
00:037 00:001 AAPL: #[EB|B:IAw]
00:039 00:001 AAPL: #[EB|WL] 0 0 0x01 0x01   2 0x00
00:040 00:001 AAPL: #[EB.BST.IDT|+]
00:042 00:001 AAPL: #[EB|BM] X86LEGACYAP
00:045 00:003 AAPL: #[EB.BST.IDT|-]
00:047 00:001 AAPL: #[EB|WL] 0 0 0x01 0x01   3 0x00
00:048 00:001 AAPL: #[EB|WL] 0 0 0x01 0x01   4 0x00
00:050 00:001 AAPL: #[EB|BRD:NV] Mac-63001698E7A34814
00:051 00:001 AAPL: #[EB|WL] 0 0 0x01 0x01   5 0x00
00:053 00:001 AAPL: #[EB.H.CHK|BM] 0x0000000000000000
00:055 00:001 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.GV boot-signature 7C436110-AB2A-4BBB-A880-FE41995C9F82
00:056 00:001 AAPL: #[EB|WL] 0 0 0x01 0x01  23 0x0E
00:058 00:001 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.GV boot-image-key 7C436110-AB2A-4BBB-A880-FE41995C9F82
00:059 00:001 AAPL: #[EB|WL] 0 0 0x01 0x01  24 0x0E
00:061 00:001 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.GV boot-image 7C436110-AB2A-4BBB-A880-FE41995C9F82
00:063 00:001 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.SV- boot-signature 7C436110-AB2A-4BBB-A880-FE41995C9F82
00:064 00:001 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.SV- boot-image-key 7C436110-AB2A-4BBB-A880-FE41995C9F82
00:066 00:001 AAPL: #[EB.H.LV|!] Err(0xE) <- RT.SV- boot-image 7C436110-AB2A-4BBB-A880-FE41995C9F82
00:068 00:001 AAPL: #[EB|H:NOT]
`

WoChen5770, 2023-09-03:
[opencore-2023-09-03-071516.txt](https://github.com/OpenIntelWireless/itlwm/files/12505428/opencore-2023-09-03-071516.txt)

kk1987, 2023-10-25:
I ran into the same and found out the issue was due to having a `IOSkywalkFamily.kext` downgrade in the OpenCore config (forgot to clean it up after switching from BRCM to Intel).