OpenIntelWireless/itlwm #910: Asus AX-3000 PCE Intel AX200 Monterey Bluetooth not working.

durgesh2020, 2023-08-28:
Asus AX-3000 PCE Intel AX200 Monterey Bluetooth not working but Wifi works fine.

My config:
1. Gigabyte Z360
2. Intel 9600K
3. Radeon RTX 590
4. RAM 32GB
5. Samsung 970 Evo Plus

Opencore: 0.94
Kexts used: IntelBluetoothFirmware and IntelBTPatcher 2.3.0, BlueToolFixup 2.6.8, AirportItlwm 2.2.0 

- Enabled AppleSecureBoot

What is tried till:
1. Reordering of kexts
2. Using latest version of all the kexts
3. Disabling XHCI port limit (Already have USBMap)

closed 2023-08-30 by zxystd

durgesh2020, 2023-09-01:
@zxystd Could you please point to a solution? Thank you.

diepeterpan, 2023-09-01:
Do some logging according to the documentation @ https://openintelwireless.github.io/IntelBluetoothFirmware/Troubleshooting.html#disclaimer ?

If you can check with Windows or Linux (Live CD) if the Bluetooth is working. Setting up the OpenCore USB ports correctly is very important for Bluetooth.

durgesh2020, 2023-09-01:
Thanks for reply. I am using 19.1 iMac.

diepeterpan, 2023-09-01:
> Thanks for reply. I am using 19.1 iMac.

19,1 seems to be supported by the USBInjectAll.KEXT - https://github.com/RehabMan/OS-X-USB-Inject-All/blob/master/USBInjectAll/USBInjectAll-Info.plist

Remove your USBMap and replace it with USBInjectAll.KEXT and disable XhciPortLimit in Kernel, see if that makes a difference.  If so then the USBMap is missing the Bluetooth USB mapping.

durgesh2020, 2023-09-01:
Oh my god! You are a genius. It is working fine. I replaced the USBMap with USBInjectAll. Only one issue I see is, how shall I create a working USBMap. Please give me some reference.

Thank you so much I was struggling since so many months for this.

diepeterpan, 2023-09-01:
> Oh my god! You are a genius. It is working fine. I replaced the USBMap with USBInjectAll. Only one issue I see is, how shall I create a working USBMap. Please give me some reference.
> 
> Thank you so much I was struggling since so many months for this.

I am running with USBInjectAll.KEXT for 2 years now, no real problems to date, USBMap will remain on my todo list too for now :-)

durgesh2020, 2023-09-01:
OK. Thank you.
I mapped my ports by following below video. 
https://www.youtube.com/watch?v=2hZPMHSfkS0

Thank you.