OpenIntelWireless/itlwm #911: A1225 Compatibility with Wireless-AC 7260/9260

moriel5, 2023-08-31:
Hello everyone, I am currently repairing an old Early-2008 iMac (A1225) which had it's power supply die, and I was wondering whether it would be worth it to upgrade it's WiFi card to an Intel Wireless-AC 7260 or 9260 (in the past, this iMac was brought to me to be reformatted, and I upgraded it to El Capitan).

The owner of the iMac,a family friend, needs support for both hidden wireless networks (because our wireless networks are hidden, and there are times where she works from our house on joint projects with my parents) as well as WPA2 and (eventually) WP3 support, both personal, not enterprise.
Handoff support is not needed, as she uses an Android phone, and is uninterested in iPhones, nor does she rely on Apple ID features in general.

Hotspot creation support is, of course, optional, but would be very nice to eventually have.

My concern, is that since the AC 7260 and the 9260 (like the rest of the standard Intel cards) only have two antennae connectors, so I am not sure whether Bluetooth could be used with WiFi off (will probably be mostly used for testing, so not the end of the world).

I already have an M.2 to mPCIe (this is the last iMac model to have a standard connector, with USB signals being passed as well as PCIe signals) adapter, that includes integrated antennae adapters, so I could theoretically plop the card in right now, and use Ethernet for Internet connectivity while getting the Kexts ready.

moriel5, 2023-08-31:
It appears that the coexistence issues that require an external cable are only present on the 2009+ iMacs, so only WPA2/3 and (optional) hotspot support remains a question mark.

zxystd, 2023-10-19:
Using itlwm.kext+Heliport can coex with your native card, don't need to replace. 
For the itlwm limitation, please read out document.

closed 2023-10-19 by zxystd

moriel5, 2023-10-19:
Thank you very much, although I had opened the issue since the documentation, while very concise, does not answer all of my questions.

It would sound like itlwm with Heliport would support WPA2/3 personal, although it is not completely clear, and I am aware that there is no hotspot support, however I am asking whether it is planned to be worked upon in the future.

Also, I wonder whether BT5+LE will work properly on El Capitan, as without a CPU upgrade (and possible firmware upgrade to the Early 2008 iMac firmware, something that I am trying to research), we cannot upgrade further, and in any case, since the family friend will need 32-bit app support to be able to use InDesign CS6 (company policy), we will be stuck with an older Mac OS version in any case, which means that some features may not be supported regardless, and this I am trying to figure out which features those are, and if they will impact her usage (she is not a techie, unlike me).