OpenIntelWireless/itlwm #916: Sonoma 14.0 VMware Bridging

hipco, 2023-10-01:
VMware Network Bridging doesn't work anymore on Sonoma 14.0 with AirportItlwm.kext (Preview 05). VMware Network Sharing works.

Off topic but a big improvement: Airplay does work again !!!!

Parachute0801, 2023-10-02:
Same problem here!

Pidbid, 2023-10-05:
Same problem here!

zxystd, 2023-10-08:
Works with itlwm.kext or LAN driver?

hipco, 2023-10-08:
With the integrated Ethernet NIC everything works as expected.

Parachute0801, 2023-10-08:
Yes, with the LAN driver everything works fine, so the problem must be with the kext.

thx in advance!! 😄

zinohome, 2023-10-09:
Same problem here! VMware Fusion 13.0.2. VMware Network Bridging doesn't work on Sonoma 14.0

ito, 2023-10-10:
It’s a general problem, nog itlwm specific.

zinohome, 2023-10-10:
But In Ventura， the same card ， vmware bridge is working fine.

zxystd, 2023-10-11:
Anyone had tried USB WIFI dongle? is it the same result?

Parachute0801, 2023-10-13:
> Anyone had tried USB WIFI dongle? is it the same result?

I am trying to test it with a tplink usb wifi but vmware does not detect it in the configuration so i cannot test it.

zxystd, 2023-10-13:
@Parachute0801 I suspect it's a the compatible problem on Sonoma with the VMWare, not sure.

Parachute0801, 2023-10-13:
> @Parachute0801 I suspect it's a the compatible problem on Sonoma with the VMWare, not sure.

Ok, in the next few days I will try to put a virtual machine in bridge mode in parallels or vbox. 
Thank you!

Parachute0801, 2023-10-16:
I just tried it with parallels 18 and it happens the same as with VMware.

zinohome, 2023-10-18:
I tested in virtualbox, the same problem, bridge via wifi can not work.

inforj, 2023-11-02:
same problem in parallels 19.1.10, bridge network not work.

yflwz, 2024-01-17:
The main reason is that the WIFI wireless network card cannot be found in the new bridge in the mac system.
[image]

FlyingThaCat, 2024-02-17:
Same problem here

Huang-JunDong, 2024-02-26:
pd19 have same problem here.

closed 2024-05-28 by hipco

FlyingThaCat, 2024-07-22:
Are this resolved or ?

hipco, 2024-07-22:
Nope. Let's wait for macOS Sequoia.