OpenIntelWireless/itlwm #917: WIFI is not connecting at all wether on ventura or sonoma while the card is supported

Abdurrahman98XX, 2023-10-05:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- itlwm Kext Version: 2.2.0 / 2.3.0 (alpha) / AirportItlwm-Sonoma-Preview05
- WiFi Card Model: intel ac 9560 160mhz (Cannon Point-LP CNVi Wireless-AC)
- PCI Product ID: i don't know but i think it is one of those
PciRoot(0x0)/Pci(0x14,0x3) -- IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/CNVW@14,3
- macOS Version: sonoma (14) (also the bug was there on ventura)

**Description**

simply it doesn't connect to ANY wifi wether it is open or wpa2 or wpa3 
i tried every thing that may solve this but with no fruit
this bug (or may be something else on my side) is driving me crazy
i wish if someone kindly helps me and i would be very apreciated and much grateful

**Bug Report Archive**

i'll edit and add it now

**Kext Download Source**

here

zxystd, 2023-10-08:
Not a full log file, please reproduce the issue again and then collect the running log.

Abdurrahman98XX, 2023-10-13:
[bugreport_4778.zip](https://github.com/OpenIntelWireless/itlwm/files/12898490/bugreport_4778.zip)

i hope this one helps as i dont know the procedures to produce a helpful bug report

Abdurrahman98XX, 2023-10-13:
another bug report

[bugreport_23507.zip](https://github.com/OpenIntelWireless/itlwm/files/12900184/bugreport_23507.zip)

zxystd, 2023-10-14:
@Abdurrahman98XX 
[image]
Please enable this option

zxystd, 2023-10-14:
The driver is working and connect successfully, maybe the router is not compatible.

Abdurrahman98XX, 2023-10-14:
@zxystd it connected for a few seconds and then disconnected again

zxystd, 2023-10-14:
@Abdurrahman98XX What is 'connected' means? can connect to the internet or just show 'connected'? Anyway, log is needed.

Abdurrahman98XX, 2023-10-14:
@zxystd it showed connected and actually has internet and then a few seconds later it shows connected but no internet connection and then it shows no connection at all

This happens once in a while not often but the general state is that there is no connection at all

zxystd, 2023-10-14:
@Abdurrahman98XX from the log the card is fully working, I suggest to try another router. Thanks.

Abdurrahman98XX, 2023-10-15:
@KA2107 
> 1. Set "Transmission Mode" to "802.11n only" or at least "802.11 g/n".

N only not found
There's:
- 802.11 B/G/N
- 802.11 B/G
- 802.11 G
- 802.11 B

zxystd, 2023-10-15:
> @KA2107
> 
> > 1. Set "Transmission Mode" to "802.11n only" or at least "802.11 g/n".
> 
> N only not found There's:
> 
> * 802.11 B/G/N
> * 802.11 B/G
> * 802.11 G
> * 802.11 B

Try 802.11 G.

zxystd, 2023-10-17:
@Abdurrahman98XX Here is a kext which combined the RS impleimplement(see #764 ), you can test it.
[Sonoma_RS.zip](https://github.com/OpenIntelWireless/itlwm/files/12923847/Sonoma_RS.zip)

vampirexx78, 2023-10-18:
Wifi works great on Sonoma with AX210, but bluetooth not working :-( how to fix?

xiaowenqiluo, 2024-02-12:
> Wifi 在装有 AX210 的索诺玛上运行良好，但蓝牙无法正常工作:-( 如何修复？

是的在
AX101上也是 WI-FI工作正常，蓝牙无法正常工作

zxystd, 2024-02-12:
ax101的蓝牙固件Intel在去年11月才放出，所以2.2.0的驱动是不支持，我前几天提交了最新的蓝牙固件，增加了ax101和be200/be202的支持，你可以在CI Actions里下载

bathIncense, 2024-02-25:
> ax101的蓝牙固件Intel在去年11月才放出，所以2.2.0的驱动是不支持，我前几天提交了最新的蓝牙固件，增加了ax101和be200/be202的支持，你可以在CI Actions里下载

好的大佬我就去更新一下最新的驱动包

bathIncense, 2024-02-25:
> ax101的蓝牙固件Intel在去年11月才放出，所以2.2.0的驱动是不支持，我前几天提交了最新的蓝牙固件，增加了ax101和be200/be202的支持，你可以在CI Actions里下载

大佬刚才去interBlueTooth 看到您提交的记录，但是看您并没有发版，我下了V2.3.0 这个发现还是 和以前一样，请问该怎么样将您提交的蓝牙支持应用到我的引导EFI中呢,求指教

zxystd, 2024-02-25:
@bathIncense https://github.com/OpenIntelWireless/IntelBluetoothFirmware/actions/runs/7823658371

bathIncense, 2024-02-26:
> @bathIncense https://github.com/OpenIntelWireless/IntelBluetoothFirmware/actions/runs/7823658371

收到大佬，看到您的提交记录了，期待您早日发版。

zxystd, 2024-02-26:
> > @bathIncense https://github.com/OpenIntelWireless/IntelBluetoothFirmware/actions/runs/7823658371
> 
> 收到大佬，看到您的提交记录了，期待您早日发版。

上面的链接进去底下就是编译好的，下面的链接直接下载也行
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/actions/runs/7823658371/artifacts/1228843782

bathIncense, 2024-02-26:
> > > @bathIncense https://github.com/OpenIntelWireless/IntelBluetoothFirmware/actions/runs/7823658371
> > 
> > 
> > 收到大佬，看到您的提交记录了，期待您早日发版。
> 
> 上面的链接进去底下就是编译好的，下面的链接直接下载也行 https://github.com/OpenIntelWireless/IntelBluetoothFirmware/actions/runs/7823658371/artifacts/1228843782

非常感谢大佬的指导，这个新的包已经可以完全驱动蓝牙，非常感谢大佬的耐心指导（自己太蠢了） ，感谢大佬，感谢大佬，感谢大佬！！！！！！！！！！

istadem2077, 2024-07-29:
After trying to use Sonoma 14.5 with AirportItlwm 2.3.0, i encountered similar issue. Turns out it is the router that's causes this behavior. As @zxystd mentioned, the card works correctly, yet it "shows" in the settings "Connected" and immediately drops the connection. After connecting another router the network works alright, tho it seems very slow as I'm trying currently to install Sonoma since like 12p, it is 7:55pm right now, and the installer still shows 9 hours left. But network connects. After installation I'll test the websites performance, and give the heads up. Could it be that Sonoma itself causes this kind of behavior?

istadem2077, 2024-07-30:
> After trying to use Sonoma 14.5 with AirportItlwm 2.3.0, i encountered similar issue. Turns out it is the router that's causes this behavior. As @zxystd mentioned, the card works correctly, yet it [...]

From what I've seen, the problem exists only on Sonoma. I'm running the 2.3.0 in Ventura, I don't see this kind of behavior. Probably the Sonoma itself.