OpenIntelWireless/itlwm #921: AirportItlwm-Sonoma-Preview05 not list Wifi networks to connect for Sonoma 14.0

elsaidyousif, 2023-10-15:
My laptop contains Intel(R) Dual Band Wireless-AC 3160
and install Mac os Sonoma 14.0 using open core
I used AirportItlwm-Sonoma-Preview05 kext file it allow me to enable and disable wifi only without listing wifi networks to connect
? What is the problem

zxystd, 2023-10-15:
log?

closed 2023-10-15 by zxystd

elsaidyousif, 2023-10-15:
Please guide me to get required log.
What is the command to run to get the log?

zxystd, 2023-10-15:
https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs
FYI

elsaidyousif, 2023-10-15:
[Log_2023-10-15_10-41-26.log](https://github.com/OpenIntelWireless/itlwm/files/12909917/Log_2023-10-15_10-41-26.log)

zxystd, 2023-10-15:
>  2.587865]: itlwm: iwm_check_rfkill RF_KILL hw: 1

RFKILL is 1,  which means your card is locked by BIOS or EFI, should be resolve by yourself,  or you can try someone else's EFI to see if it can work, thanks.

elsaidyousif, 2023-10-15:
thanks for your response.
is there an recommended ios configurtions or efi from your point of view.