OpenIntelWireless/itlwm #923: Airplay working laggy and iservices not work in sonoma

alighafourian, 2023-10-19:
**Have You Read Our Docs**
<Yes>

**Are You Reporting A Bug**
<Yes>

**Environment**

- Kext Version: https://github.com/OpenIntelWireless/itlwm/releases/download/v2.3.0-alpha/AirportItlwm-Sonoma-v2.3.0-DEBUG-alpha-ff1138b.zip
- WiFi Card Model: 7265N
- PCI Product ID: 0x8086, 0x5002
- macOS Version: 14.1RC

**Description**
after update to 2.3 alpha airplay work laggy and when use VPN dont show any picture before this update in senoma b5 preview work good 

**Bug Report Archive**

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.3.0-alpha/AirportItlwm-Sonoma-v2.3.0-DEBUG-alpha-ff1138b.zip
**LOGs**
[Log_2023-10-19_10-09-39.log](https://github.com/OpenIntelWireless/itlwm/files/13038878/Log_2023-10-19_10-09-39.log)

diepeterpan, 2023-10-19:
Oh golly, will I be slammed for this?  On my side iServices did not work on Sonoma B5 nor does it work on 2.3 Alpha. So no change, for that reason I am running with Sonoma and itlwm (ethernet) until resolved.

alighafourian, 2023-10-19:
[WiFiDiagnostics_2023-10-19_10.14.27.119_macOS_MacBookPro15,4_23B73.zip](https://github.com/OpenIntelWireless/itlwm/files/13039000/WiFiDiagnostics_2023-10-19_10.14.27.119_macOS_MacBookPro15.4_23B73.zip)

zxystd, 2023-10-19:
@diepeterpan Please upload preview05 runtime log, let me seee the nss differences. what's the max mcs value when airplaying did you see on each versions?

alighafourian, 2023-10-19:
@zxystd preview05 runtime log:
[Log_2023-10-19_12-01-36.log](https://github.com/OpenIntelWireless/itlwm/files/13039977/Log_2023-10-19_12-01-36.log)

Vorshim92, 2023-10-19:
Will it be possible to restore Apple services in the future with this kext? They no longer work from beta 5. :(
thanks for your work! <3

alighafourian, 2023-10-19:
@zxystd can u post preview05 code ?

diepeterpan, 2023-10-19:
> @diepeterpan Please upload preview05 runtime log, let me seee the nss differences. what's the max mcs value when airplaying did you see on each versions?

When I refer to iServices I imply e.g. **Messages (SMS link to iPhone)** and **Find My**.....  I am not using airplay.

zxystd, 2023-10-19:
@alighafourian  Your router advertise support 1 ss only, for the new RS rate scaling code, it will disable MIMO if single ss is supported(Linux do the same thing), on preview05 or previous version, MIMO is always enabled, so it may increase performance.
I suggest use another better router to test.

gujiangjiang, 2023-10-19:
same with me using beta5 in sonoma that iService don't work anymore.

iMessage and Facetime always show waiting circle but can't login successful but when I plug in LAN the iService works again.

I have test and reinstall sonoma over 5 times even change SN and MLB it still have the same problem. The iService don't works correctly with beta5 kext in sonoma.

Hope to fix this.

Here is a topic about this in Chinese pcbeta.com:
https://bbs.pcbeta.com/viewthread-1979717-1-5.html

alighafourian, 2023-10-19:
> @alighafourian Your router advertise support 1 ss only, for the new RS rate scaling code, it will disable MIMO if single ss is supported(Linux do the same thing), on preview05 or previous version, MIMO is always enabled, so it may increase performance. I suggest use another better router to test.

its mobile hotespot same problem with home network need home network log?

alighafourian, 2023-10-19:
preview05 log
[Log_2023-10-19_21-23-07.log](https://github.com/OpenIntelWireless/itlwm/files/13046057/Log_2023-10-19_21-23-07.log)

mahronid, 2023-10-25:
> any help I could provide to solve this issue ,I am also facing the same problem.

I'm using heliport + itwlm kext right now and iServices is working

diepeterpan, 2023-10-25:
> > any help I could provide to solve this issue ,I am also facing the same problem.
> 
> I'm using heliport + itwlm kext right now and iServices is working

Ditto. But we would like it to work with **AirportItlwm.kext**; without it, the use of **AirportItlwm.kext** is kinda useless in Sonoma.

Umenezumi, 2023-11-03:
+1，iService not work in Sonoma

lohcve, 2023-11-06:
What's the fix plan? Can it be fixed?

Showdown76py, 2023-11-20:
Yes, I'm also having this issue and it is quite annoying. I hate to do tethering with my iPhone to access Apple Music/Messages/FaceTime.

harmohitsandhu, 2023-11-29:
+1，iService not work in Sonoma

mosesmakaka, 2024-01-02:
Somebody please fix this issue man... What's the exact issue with this driver that makes it impossible for iServices to work yet ethernet makes things work.

zxystd, 2024-03-16:
This issue is separated the two issues:
- airplay laggy #964 
- iService not working on Sonoma. Please Track to #942

closed 2024-03-16 by zxystd