OpenIntelWireless/itlwm #924: Sleep won't work with kext loaded in Sonoma

JoeOIVOV, 2023-10-21:
**Have You Read Our Docs**
no

**Are You Reporting A Bug**
yes

**Environment**

- Kext Version: 2.3.0 alpha Sonoma
- WiFi Card Model: Intel 8265
- PCI Product ID: 
- macOS Version: latest Sonoma

**Description**
Sleep won't work as soon as Airportlwm kext is loaded... if you decide to not load it, wifi doesn't work... but sleep will. 

**Bug Report Archive**

**Kext Download Source**

Inmike02, 2023-10-21:
I have observed the same behavior in Sonoma. I captured the following error message in the system logs

`Sleep prevented by ARPT`

Kext Version: 2.3.0 alpha Sonoma
WiFi Card Model: Intel ax211
PCI Product ID:
macOS Version: latest Sonoma

alighafourian, 2023-10-21:
@Inmike02  use ssdt rename ARPT

closed 2023-10-21 by zxystd

Inmike02, 2023-10-21:
@alighafourian, @zxystd  -- I'm not sure why this is marked as "User Mistake" and the comment 

> @Inmike02 use ssdt rename ARPT 

I am using an SSDT to rename the device to ARPT along with all the recommend properties. I am happy to debug this further and follow your guidance, but as far as I am aware I have followed all of the requirements and yet as the original user who opened this issue reported--sleep does not even start when Airportlwm.kext is loaded. If I disable the kext sleep works fine.

Thank you!

andreszerocross, 2023-10-21:
> @alighafourian, @zxystd -- I'm not sure why this is marked as "User Mistake" and the comment
>
>
> I am using an SSDT to rename the device to ARPT along with all the recommend properties. I am happy to [...]

Just remove _PRW method from ARPT device in ACPI. Or you can rename _PRW of ARPT device to XPRW. Try it

JoeOIVOV, 2023-10-21:
"Just remove _PRW method from ARPT device in ACPI. Or you can rename _PRW of ARPT device to XPRW. Try it"

What does this even mean?  I checked in Opencore, under ACPI>ADD download/update acpi, or ACPI>PATCH list of patches and there is nothing with ARPT in it or XPRW... 

Do you have a screen shot of what to do to fix this to allow sleep through this bug?  

Thank you

manucuervor, 2024-05-16:
Same problem here with Sonoma 14.4