OpenIntelWireless/itlwm #925: Can't Connect to WiFi

FlicksDaModdle, 2023-10-22:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

I think so
**Environment**

- Kext Version: 2.20 stable
- WiFi Card Model: Intel® Wi-Fi 6E AX210
- PCI Product ID: 8086:2725
- macOS Version: Ventura

**Description**
WiFi not working in installer, using airportitlwm. it shows icon and can allow me to type password, but does not connect to the network.

**Kext Download Source**
GIthub releases

zxystd, 2023-10-22:
please upload the running log or the issue will be closed.

FlicksDaModdle, 2023-10-22:
How do I get a log? This is when I'm trying to install Ventura.

zxystd, 2023-10-23:
https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs

FlicksDaModdle, 2023-10-25:
i haven't even gotten past install: it just doesn't work for some reason in macos installer or in macos when i use a full image .raw to install, neither works, like i can't even get into macos itself

diepeterpan, 2023-10-25:
> i haven't even gotten past install: it just doesn't work for some reason in macos installer or in macos when i use a full image .raw to install, neither works, like i can't even get into macos itself

You should be able to complete the installation of MacOS without Wifi and remove/disable any KEXT not needed for the installation.  It is most likely something else that is failing the installation to complete, not something that will be resolved here for your setup/hardware. Please consult forums related to the installation of MacOS.

FlicksDaModdle, 2023-10-26:
[Log_2023-10-26_Oct-22-12.log](https://github.com/OpenIntelWireless/itlwm/files/13180812/Log_2023-10-26_Oct-22-12.log)

FlicksDaModdle, 2023-10-26:
this is montery btw

FlicksDaModdle, 2023-10-26:
@zxystd help pls

FlicksDaModdle, 2023-10-26:
I currently dont have access to the router settings, let me check it in a few hours and come  back to you

zxystd, 2023-10-27:
@FlicksDaModdle The log says it is connected, it you don't have internet access, that maybe something other problems, just like @KA2107 said, WPA3 is not supported on itlwm, you can check it in the System Settings -> Wi-Fi panel.

FlicksDaModdle, 2023-10-27:
do I activate that in the router settings? it shows the wifi is asking for a wpa2 password

FlicksDaModdle, 2023-10-27:
I have an eero router, and wpa3 is turned off. It only asks for a wpa2 password in macos anyways.

FlicksDaModdle, 2023-10-27:
the eero router supports 802.11w and there is no off switch

zxystd, 2023-10-28:
802.11w is not support yet.

closed 2023-11-16 by zxystd