OpenIntelWireless/itlwm #926: compile with xcode issus (show wifi but not connect)

alighafourian, 2023-10-24:
**Have You Read Our Docs**
Yes 

**Are You Reporting A Bug**
Yes
**Environment**

- Kext Version: 
- WiFi Card Model: 
- PCI Product ID: 
- macOS Version: 

**Description**
I download code file and compiler . compile done but show wifi and dosnt connect 

**Bug Report Archive**

[image]
[image]

kext log:
[Log_2023-10-24_12-41-37.log](https://github.com/OpenIntelWireless/itlwm/files/13112983/Log_2023-10-24_12-41-37.log)

**Kext Download Source**

ShroomKing, 2023-10-24:
Same when building from source with latest Xcode, release build works fine though.

alighafourian, 2023-10-24:
@ShroomKing 
only way fork on github and run xcode Workflow

alighafourian, 2023-10-24:
ok i close issus

alighafourian, 2023-10-24:
@diepeterpan 
ok i close issus

ShroomKing, 2023-10-24:
I did more tests.
This appears to be a compiler code optimisation problem.

When creating a debug build it uses optimisation level 0 and the kext works.

But when creating an optimised build (Release build type / Product->Archive) the kext can't connect to wifi.

The highest optimisation level that works right now is 1, so I assume some flag in level 2 is breaking the code (more info here: https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html)

You can disable optimisation flags by using `-fno-(flag-name)` so for example to disable `-falign-functions` add `-fno-align-functions` to CFLAGS, this way we can keep the other optimisations but I don't know if there's a better way to debug this than trying to disable all the flags one by one.

usr-sse2, 2024-03-12:
> The highest optimisation level that works right now is 1, so I assume some flag in level 2 is breaking the code (more info here: https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html)

It's better to look options in Clang documentation because Xcode compiles with Clang, not GCC.

zxystd, 2024-03-12:
> > The highest optimisation level that works right now is 1, so I assume some flag in level 2 is breaking the code (more info here: https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html)
> 
> It's better to look options in Clang documentation because Xcode compiles with Clang, not GCC.

@usr-sse2 You are right, but I don't have the time and environment(no Sonoma/new XCode) to investigate this, if you have discoveries please let me know or open an PR, thank you!

zxystd, 2024-03-12:
From the log I can see the root issue is that the optimization breaks function `ieee80211_get_assoc_req`, with invalid assoc request(RSN invalid), for what's the differences I can't see with the decompiled code.

vanyasem, 2024-03-12:
@zxystd I can confirm that the issue is resolved with the latest commits. Release builds are now able to connect to Wi-Fi successfully. Tested on Sonoma 14.4

alighafourian, 2024-03-12:
> @zxystd I can confirm that the issue is resolved with the latest commits. Release builds are now able to connect to Wi-Fi successfully. Tested on Sonoma 14.4

Compile with xcode app 
If use terminal for build wifi can not connect  but show 
If use xcode app work good