OpenIntelWireless/itlwm #928: Wi-Fi appears disabled

kaelthasmanu, 2023-10-26:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

I don't know if it's a bug

**Environment**

- Kext Version: v2.2.0Stable
- WiFi Card Model: Intel(R) Wi-Fi 6 AX210
- PCI Product ID: I dont know
- macOS Version: Ventura

**Description**

Wi-Fi appears disabled, I cannot activate it

**Bug Report Archive**

**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.2.0

zxystd, 2023-10-27:
Are you sure it is your log?

kaelthasmanu, 2023-10-27:
oh sorry, apparently I was wrong is this :

[Log_2023-10-26_22-52-49.log](https://github.com/OpenIntelWireless/itlwm/files/13184536/Log_2023-10-26_22-52-49.log)

kaelthasmanu, 2023-10-27:
@KA2107 Yes it is true and from what I have read I think that Hackintosh does not have support so let me close this issue, thank you very much

closed 2023-10-27 by kaelthasmanu