OpenIntelWireless/itlwm #929: Intel Wifi 7260AC no wifi network seen Sonoma 14.0

b1t0y, 2023-10-28:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.3.0 Alpha Sonoma 14.0
- WiFi Card Model: Intel 7260AC 
- PCI Product ID: ?
- macOS Version: Sonoma 14.0

**Description**
Wifi is ON but not showing any wifi network or wifi network blank

**Bug Report Archive**

[Log_2023-10-28_08-41-40.log](https://github.com/OpenIntelWireless/itlwm/files/13196227/Log_2023-10-28_08-41-40.log)

**Kext Download Source**
(https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.3.0-alpha)

zxystd, 2023-10-29:
The driver is working well from the log. Check your other settings.

closed 2023-10-29 by zxystd

b1t0y, 2023-10-29:
> The driver is working well from the log. Check your other settings.

@zxystd 
Please advise What other settings should i check? I tried the trouble shooting in github.  But non is working so far. It worked well on ventura airportitwlm without any issues only on somoma.

zxystd, 2023-10-29:
@b1t0y The same as the preview05 version? Here are some advices:
- AirportItlwm alpha version should be only installed on Somona beta 5 and later verison.
- Avoid using OCLP WiFi patches.
- Check the WiFi antennas.
- Anything related.