OpenIntelWireless/itlwm #930: AirportItlwm-Sonoma-v2.3.0 causes kernel panic on macOS 14.1

KawaiiGirl2001, 2023-10-29:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm-Sonoma-v2.3.0-alpha
- WiFi Card Model: intel Wireless-AC 8265
- PCI Product ID: 0x24fd
- macOS Version: Sonoma 14.1

**Description**
This kext doesn't work. After enabling AirportItlwm, the system does not boot. After the Apple logo appears, a kernel panic immediately occurs and the laptop reboots. When using itlwm-v2.3.0+heliport everything works fine.

**Bug Report Archive**
Kernel panic logs are attached.
[Kernel-2023-10-28-114459.panic.zip](https://github.com/OpenIntelWireless/itlwm/files/13197687/Kernel-2023-10-28-114459.panic.zip)
**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.3.0-alpha/AirportItlwm-Sonoma-v2.3.0-DEBUG-alpha-ff1138b.zip

zxystd, 2023-10-29:
please add `keepsyms=1` to `boot-arg` and trigger the panic again.

KawaiiGirl2001, 2023-11-09:
The problem was caused by conflicts between the kexts IOSkywalkFamily.kext and IO80211FamilyLegacy.kext with AirportItlwm. After removing IOSkywalkFamily.kext and IO80211FamilyLegacy.kext, the kernel panic does not occur. Now Wi-Fi is working great and sleep is also working. Problem solved.

closed 2023-11-09 by KawaiiGirl2001

clveranis, 2023-11-14:
@KawaiiGirl2001  How did you remove the `IOSkywalkFamily` and `IO80211FamilyLegacy` kexts? I have the same issue when using the AirportItlwm kext and Sonoma. Without it, my Sonoma install boots fine but I have no wifi, however, if I add the AirportItlwm kext then my system won't even boot to the login screen, just panics and restarts during load.

KawaiiGirl2001, 2023-11-14:
@clveranis Simply disable these kexts in your OpenCore config file. They are not needed if you are using intel wireless and conflict with AirportItlwm.

clveranis, 2023-11-14:
@KawaiiGirl2001 How does one disable these kexts when they are being loaded by macOS itself and not via my OC config?