OpenIntelWireless/itlwm #931: Can't sleep

denizckai, 2023-10-29:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

Environment
Kext Version: AirportItlwm-Sonoma-v2.3.0-DEBUG-alpha-ff1138b
    WiFi Card Model: Intel Wi-Fi 6E AX211 (MSI PRO Z790-A WIFI DDR4 - M.2 WiFi 6E Card)
    PCI Product ID: 0x8086 (Vendor)
    macOS Version: Sonoma 14.1 (23B74)

**Description**
The same hardware that does not have a sleep problem in the Ventura version only has a sleep problem after the update to Sonoma. I tried clean installation and many options but I was not successful. If I disable the AirportItlwm kext it sleeps as before. I don't know what the difference is between Ventura and Sonoma kexts, but the Ventura version does not have this problem. No error report has been generated.

**Bug Report Archive**

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.3.0-alpha/AirportItlwm-Sonoma-v2.3.0-DEBUG-alpha-ff1138b.zip

denizckai, 2023-11-13:
I think it's important because Sonoma can't sleep in any way. I generated both Heliport Diagnostics Report and also a Bug report in terminal.

[bugreport_47539.zip](https://github.com/OpenIntelWireless/itlwm/files/13340786/bugreport_47539.zip)
[Log_2023-11-13_23-14-40.log](https://github.com/OpenIntelWireless/itlwm/files/13340787/Log_2023-11-13_23-14-40.log)

zxystd, 2023-11-16:
try using `log show --last 1d | grep "Wake reason"` command or hackintosh community to debug and resolve sleep problems.

closed 2023-11-16 by zxystd

andreszerocross, 2023-11-16:
> try using `log show --last 1d | grep "Wake reason"` command or hackintosh community to debug and resolve sleep problems.

Actually it's true. With Airportitlwm in sonoma, the notebook can't idle sleep. But if you click Sleep from menu then it will sleep. The problem is with idle sleep. It will stay awake not enter sleep except manual sleep.

denizckai, 2023-11-16:
> Actually it's true. With Airportitlwm in sonoma, the notebook can't idle sleep. But if you click Sleep from menu then it will sleep. The problem is with idle sleep. It will stay awake not enter [...]

It's not about this simple Hackitontosh situation. Neither idle sleep nor manual sleep is valid in this problem. There is a problem directly related to AirportItlwm-Sonoma-v2.3.0.

denizckai, 2023-11-16:
> try using `log show --last 1d | grep "Wake reason"` command or hackintosh community to debug and resolve sleep problems.

When I type this command and enter into the terminal, I don't get any response, just blank. Interesting but it's as if the entire sleep-related feature has been deactivated. As I said before, when I do not use AirportItlwm-Sonoma-v2.3.0., I can control everything and the commands provide the necessary logs.