OpenIntelWireless/itlwm #932: Itwlm not Remembering Hidden WI-FI Networks

Abdullah-Albanna, 2023-10-31:
**Have You Read Our Docs**

Yes, I did

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: itlwm-v2.3.0-DEBUG-alpha-ff1138b
- WiFi Card Model: Intel Centrino Advanced-N 6205
- PCI Product ID: 0x8086 (Vendor), 0x0085 (Device), 0x8086 (Sub Vendor), 0x1311 (Sub Device)
- macOS Version: MacOS Big Sur (so as Catalina)

**Description**

When I connect to a hidden network, it connects normally, but when I restart, it's like it wasn't  there, so I'd need to enter it manually every time

**Bug Report Archive**

[bugreport_13898.zip](https://github.com/OpenIntelWireless/itlwm/files/13211689/bugreport_13898.zip)

**Kext Download Source**

I downloaded it from here, from the releases, I didn't build it

Umenezumi, 2023-11-03:
change wifi channel

jlu-lx, 2023-12-30:
I have the same problem.

matte-oss, 2025-02-11:
#976