OpenIntelWireless/itlwm #935: macOS Frequent Panics || Sanoma 14.1.1 || AX200

naveenkrdy, 2023-11-14:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: [v2.3.0-alpha](https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.3.0-alpha)
- WiFi Card Model: AX200
- PCI Product ID: 0x2723
- macOS Version: Sanoma 14.1.1

**Description**
After upgrading to Sonoma, macOS crashes frequently when booting up, waking from sleep, or opening network settings, etc. 

Panic Report:
```
panic(cpu 6 caller 0xffffff8004f88fb3): Kernel trap at 0xffffff800721d130, type 14=page fault, registers:
CR0: 0x0000000080010033, CR2: 0x0000000000000000, CR3: 0x000000046477304f, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0x0000000000000003, RCX: 0xffffff80099129f0, RDX: 0x0000000000000003
RSP: 0xfffffff3fec0fa30, RBP: 0xfffffff3fec0fa40, RSI: 0xffffff800724b5a0, RDI: 0xffffff8b9c7e2c00
R8:  0x0200000110000698, R9:  0x0200000110000000, R10: 0x0200000110000698, R11: 0x00ffffff00000000
R12: 0x0000000000000000, R13: 0x0000000000000000, R14: 0x000000000000000c, R15: 0x0000000000000000
RFL: 0x0000000000010246, RIP: 0xffffff800721d130, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x6, PL: 0, VF: 0

Panicked task 0xffffff86d1177638: 9 threads: pid 211: airportd
Backtrace (CPU 6), panicked thread: 0xffffff8b9a5aeb30, Frame : Return Address
0xfffffff3fec0f3b0 : 0xffffff8004e3aca1 mach_kernel : _handle_debugger_trap + 0x4b1
0xfffffff3fec0f400 : 0xffffff8004f993c0 mach_kernel : _kdp_i386_trap + 0x110
0xfffffff3fec0f440 : 0xffffff8004f88b0c mach_kernel : _kernel_trap + 0x55c
0xfffffff3fec0f4c0 : 0xffffff8004dd7971 mach_kernel : _return_from_trap + 0xc1
0xfffffff3fec0f4e0 : 0xffffff8004e3af8d mach_kernel : _DebuggerTrapWithState + 0x5d
0xfffffff3fec0f5d0 : 0xffffff8004e3a633 mach_kernel : _panic_trap_to_debugger + 0x1e3
0xfffffff3fec0f630 : 0xffffff80055d957b mach_kernel : _panic + 0x84
0xfffffff3fec0f720 : 0xffffff8004f88fb3 mach_kernel : _sync_iss_to_iks + 0x2c3
0xfffffff3fec0f8a0 : 0xffffff8004f88c97 mach_kernel : _kernel_trap + 0x6e7
0xfffffff3fec0f920 : 0xffffff8004dd7971 mach_kernel : _return_from_trap + 0xc1
0xfffffff3fec0f940 : 0xffffff800721d130 com.apple.iokit.IOSkywalkFamily : __ZN25IOSkywalkNetworkInterface7_bpfTapEP7__ifnetjj + 0x2c
0xfffffff3fec0fa40 : 0xffffff800506b6ed mach_kernel : _bpfioctl + 0x1bdd
0xfffffff3fec0fa70 : 0xffffff800506ae44 mach_kernel : _bpfioctl + 0x1334
0xfffffff3fec0fab0 : 0xffffff800506a1a7 mach_kernel : _bpfioctl + 0x697
0xfffffff3fec0fb60 : 0xffffff800505a688 mach_kernel : _spec_ioctl + 0x98
0xfffffff3fec0fb90 : 0xffffff800504ca04 mach_kernel : _VNOP_IOCTL + 0xc4
0xfffffff3fec0fbf0 : 0xffffff800503f3eb mach_kernel : _u32CharToUTF8Bytes + 0x91b
0xfffffff3fec0fe00 : 0xffffff80052f2f64 mach_kernel : _sys_fcntl_nocancel + 0x3954
0xfffffff3fec0fe30 : 0xffffff80053559a4 mach_kernel : _ioctl + 0x484
0xfffffff3fec0ff40 : 0xffffff80054541cd mach_kernel : _unix_syscall64 + 0x1dd
0xfffffff3fec0ffa0 : 0xffffff8004dd7db6 mach_kernel : _hndl_unix_scall64 + 0x16
      Kernel Extensions in backtrace:
         com.apple.iokit.IOSkywalkFamily(1.0)[8FAC5BA7-70BB-3117-9D79-6B5B57F41E7A]@0xffffff800720e000->0xffffff800723cfff
            dependency: com.apple.driver.mDNSOffloadUserClient(1.0.1b8)[14410048-29B3-3126-BE24-D2A0BD5DFE52]@0xffffff800711a000->0xffffff800711efff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[5D912FD8-C4CD-3CF6-B214-DF5BF7AB4FA0]@0xffffff8006ec0000->0xffffff8006ed7fff

Process name corresponding to current thread (0xffffff8b9a5aeb30): airportd
Boot args: keepsyms=1 -novht

Mac OS version:
23B81

Kernel version:
Darwin Kernel Version 23.1.0: Mon Oct  9 21:27:27 PDT 2023; root:xnu-10002.41.9~6/RELEASE_X86_64
Kernel UUID: 7C8FE931-57C6-38AD-9EBB-CC5572293AE6
roots installed: 0
KernelCache slide: 0x0000000004a00000
KernelCache base:  0xffffff8004c00000
Kernel slide:      0x0000000004ae0000
Kernel text base:  0xffffff8004ce0000
__HIB  text base: 0xffffff8004b00000
System model name: MacBookPro16,1 (Mac-E1008331FDC96864)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 726575691989
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000000a92b493e32
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000a45dd73a6 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff806855c000 - 0xffffffa06855c000
  . PGZ   : 0xffffff806855c000 - 0xffffff806a55d000
  . VM    : 0xffffff806a55d000 - 0xffffff8536d5c000
  . RO    : 0xffffff8536d5c000 - 0xffffff86d055c000
  . GEN0  : 0xffffff86d055c000 - 0xffffff8b9cd5c000
  . GEN1  : 0xffffff8b9cd5c000 - 0xffffff906955c000
  . GEN2  : 0xffffff906955c000 - 0xffffff9535d5c000
  . GEN3  : 0xffffff9535d5c000 - 0xffffff9a0255c000
  . DATA  : 0xffffff9a0255c000 - 0xffffffa06855c000
  Metadata: 0xffffff8024dc8000 - 0xffffff8044dc8000
  Bitmaps : 0xffffff8044dc8000 - 0xffffff8047dc8000
  Extra   : 0 - 0

```

Thanks you !

qasim799, 2023-11-14:
i was also having soo many random drops with latest 2.3.0 airport and itlwm, then i tried 2.2.0 itlwm + heliport latest beta it's very stable now, you should also try itlwm 2.2.o stable with heliport combo on sonoma

naveenkrdy, 2023-11-15:
@qasim799 do you use to also get panics as just connection dropouts ?

closed 2023-11-16 by zxystd

naveenkrdy, 2023-11-16:
@zxystd Hi !
i dont think my efi has any issue as i have been using the on ventura fine.
in the panic report i see IOSkywalk in the backtrace doesnt seem like it had anything to do with the EFI. 
Are you sure that this is not caused by the kext ?