OpenIntelWireless/itlwm #937: AirportItlwm-Sonoma-v2.3.0 never sleep

denizckai, 2023-11-16:
Have You Read Our Docs
Yes

Are You Reporting A Bug
Yes

Environment
Kext Version: AirportItlwm-Sonoma-v2.3.0-DEBUG-alpha-ff1138b
WiFi Card Model: Intel Wi-Fi 6E AX211 (MSI PRO Z790-A WIFI DDR4 - M.2 WiFi 6E Card)
PCI Product ID: 0x8086 (Vendor)
macOS Version: Sonoma 14.1.1 

**Description**
AirportItlwm-Sonoma-v2.3.0 must be breaking something because it's not going to sleep at all. This topic is not something I would search on the Hackintosh community because it is directly related to AirportItlwm-Sonoma-v2.3.0. There is no problem sleeping when I don't use this kext. Also, there are no problems with the AirportItlwm-Ventura-v2.3.0 in Ventura. But there is a problem with AirportItlwm-Sonoma-v2.3.0 in Sonoma.

I tried almost all commands and settings related to sleep control (everything was fine while using the Ventura with same hardware, there was no sleep problem at all). Even I tried with clean installation Sonoma, but the situation did not change. If you don't use AirportItlwm-Sonoma-v2.3.0. everything goes back to normal except wifi connection. By the way, I insist on AirportItlwm because I use WPA enterprise connection.

darthsian, 2023-11-19:
Same problem here.
Kext Version: AirportItlwm-Sonoma-v2.3.0-DEBUG-alpha-ff1138b
WiFi Card Model: Intel® Wireless-AC 9560
macOS Version: Sonoma 14.1.1

log show | grep -i "wake reason"
2023-11-19 19:32:22.301023+0100 0xb3c4     Default     0x0                  277    0    airportd: (IO80211) systemWokenByWiFi: System Wake Reason not found
2023-11-19 19:32:22.301028+0100 0xb3c4     Default     0x0                  277    0    airportd: (IO80211) [com.apple.WiFiManager:] <airport[277]> systemWokenByWiFi: System Wake Reason not found
2023-11-19 19:58:23.900752+0100 0xf311     Default     0x0                  277    0    airportd: (IO80211) systemWokenByWiFi: System Wake Reason not found
2023-11-19 19:58:23.900756+0100 0xf311     Default     0x0                  277    0    airportd: (IO80211) [com.apple.WiFiManager:] <airport[277]> systemWokenByWiFi: System Wake Reason not found
2023-11-19 19:58:24.013429+0100 0xf143     Default     0x0                  277    0    airportd: (IO80211) systemWokenByWiFi: System Wake Reason not found
2023-11-19 19:58:24.013434+0100 0xf143     Default     0x0                  277    0    airportd: (IO80211) [com.apple.WiFiManager:] <airport[277]> systemWokenByWiFi: System Wake Reason not found
2023-11-19 19:58:32.476015+0100 0xf31b     Default     0x0                  277    0    airportd: (IO80211) systemWokenByWiFi: System Wake Reason not found
2023-11-19 19:58:32.476025+0100 0xf31b     Default     0x0                  277    0    airportd: (IO80211) [com.apple.WiFiManager:] <airport[277]> systemWokenByWiFi: System Wake Reason not found
2023-11-19 20:17:23.469324+0100 0x10ca3    Default     0x0                  277    0    airportd: (IO80211) systemWokenByWiFi: System Wake Reason not found
2023-11-19 20:17:23.469329+0100 0x10ca3    Default     0x0                  277    0    airportd: (IO80211) [com.apple.WiFiManager:] <airport[277]> systemWokenByWiFi: System Wake Reason not found
2023-11-19 20:17:23.895534+0100 0x10ca3    Default     0x0                  277    0    airportd: (IO80211) systemWokenByWiFi: System Wake Reason not found
2023-11-19 20:17:23.895539+0100 0x10ca3    Default     0x0                  277    0    airportd: (IO80211) [com.apple.WiFiManager:] <airport[277]> systemWokenByWiFi: System Wake Reason not found
2023-11-19 20:17:23.996199+0100 0x16367    Default     0x0                  277    0    airportd: (IO80211) systemWokenByWiFi: System Wake Reason not found
2023-11-19 20:17:23.996203+0100 0x16367    Default     0x0                  277    0    airportd: (IO80211) [com.apple.WiFiManager:] <airport[277]> systemWokenByWiFi: System Wake Reason not found
2023-11-19 21:17:36.550158+0100 0x1d6e2    Default     0x0                  277    0    airportd: (IO80211) systemWokenByWiFi: System Wake Reason not found
2023-11-19 21:17:36.550167+0100 0x1d6e2    Default     0x0                  277    0    airportd: (IO80211) [com.apple.WiFiManager:] <airport[277]> systemWokenByWiFi: System Wake Reason not found
2023-11-19 22:36:16.905464+0100 0x297a6    Default     0x0                  277    0    airportd: (IO80211) systemWokenByWiFi: System Wake Reason not found
2023-11-19 22:36:16.905469+0100 0x297a6    Default     0x0                  277    0    airportd: (IO80211) [com.apple.WiFiManager:] <airport[277]> systemWokenByWiFi: System Wake Reason not found

radianttap, 2023-11-21:
Same problem here for me, as well:

Kext Version: AirportItlwm-Sonoma-v2.3.0-DEBUG
WiFi Card Model: Intel AX200 on Asus ROG Strix B550-I
macOS Version: Sonoma 14.1.1

I previously had BCM94360NG card instead of AX200, sleep worked perfectly. 

Is there anything I can do / provide for debugging this..?

radianttap, 2023-11-21:
Here's console log, where I explicitly attempted to use Sleep menu option in the Apple menu.
Screen turns off but computer stays on. Tapping the keyboard brings the login window back on.

[sonoma-sleep-attempt-log.txt](https://github.com/OpenIntelWireless/itlwm/files/13429608/sonoma-sleep-attempt-log.txt)

I am not sure this issue is exclusively due to Airportitlwm but maybe you can recognize something in here, if it is related.

radianttap, 2023-11-22:
I just tested and can cofirm: simply disabling `Airportitlwm.kext` in the `config.plist` and sleep works, as well as wake. 

Let me know if you need any kind of test / debug to do, I have the build ready.

perez987, 2023-12-02:
Same for me as well. But I get this issue with both AirportItlwm and itlwm. Other than this, wifi works fine. Sonoma 14.1.1 and 14.1.2. Intel Wi-Fi 6E AX211.

perez987, 2023-12-03:
Fixed apparently with changes to `pmset` configuration. I have disabled some parameters related to wake while sleep:

- `powernap 0`: Power Nap disabled
- `proximitywake 0`: Do not wake the system from nearby devices with the same iCloud
- `standby 0`: Do not put the system into hibernation after a default sleep time
- `tcpkeepalive 0`: Do not keep sending and receiving TCP packets during system sleep (disconnect from Internet or not)
- `ttyskeepawake 0`: Do not wake up the system from remote access.

sudo pmset powernap 0;sudo pmset proximitywake 0;sudo pmset standby 0;sudo pmset tcpkeepalive 0;sudo pmset ttyskeepawake 0;defaults write com.apple.loginwindow PowerButtonSleepsSystem -bool no

I have disabled also Handoff and Airplay in System Settings >> General.

I have tried sleep in all 3 ways:

- PC power button
- Menu About This Mac >> Sleep
- sleep after the time configured with pmset.

macOS has gone to sleep in all 3 ways. Turning off lights and fans.

It may not work for everyone.

denizckai, 2023-12-03:
My report was previously closed due to a hackintosh problem. That's why I think it needs to be explained. I've been a hackintosh user for about 15 years (since the Leopard times). There was a time when I went crazy with a lot of hardware. clover, opencore etc. Therefore, I think I have the basic knowledge to determine whether this problem is caused by the operating system or not.
I tried almost every combination in sleep parameters.
USB map is ok, it should be fine anyway, especially due to the bluetooth problem (bluetooth works fine by the way). I even did it with usbtoolbox, I also did it with usbmap, also Hackintool. I'm pretty sure that USB mapping is fine.
I even did a clean installation because I thought it was after the update, but it still didn't work.
Out of curiosity, I installed Ventura on another disk and everything is working properly there.

perez987, 2023-12-03:
@denizckai 
My machine is Z390 + i9-9900 + RX 6600. Yours is not the same. I hope there will a fix. Sleep is an important feature for me.

perez987, 2023-12-03:
@denizckai 
One more thing to try.
Sometimes power or reminder events wake the computer or prevent sleep.
If you do not have events intentionally scheduled by you, you can try canceling all power events:

```
sudo pmset schedule cancelall
sudo chflags schg /Library/Preferences/SystemConfiguration/com.apple.AutoWake.plist
```

This prevents the system from updating the file and the system no longer wakes up to reminder events.

Try and comment if macOS goes to sleep.

It's advisable to remove the flag before a system update:

`sudo chflags noschg /Library/Preferences/SystemConfiguration/com.apple.AutoWake.plist`

Source: _MarkFromEarth_ user in Apple Discussions.

denizckai, 2023-12-03:
@perez987 
I tried your suggestion even though it had nothing to do with the situation I mentioned. Result: can't sleep. 
But thanks.

perez987, 2023-12-03:
@denizckai 
Sorry.
It seems evident that when using these kexts with Intel wifi and BT devices something is broken in the sleep system.
Which leads to your original approach.

ricardo42864, 2024-01-15:
> Same problem here. Kext Version: AirportItlwm-Sonoma-v2.3.0-DEBUG-alpha-ff1138b WiFi Card Model: Intel® Wireless-AC 9560 macOS Version: Sonoma 14.1.1
>
> `log show | grep -i "wake reason"` 2023-11-19 [...]

II have the same problem as you. Have you solved it

denizckai, 2024-01-15:
Unfortunately I couldn't find a solution, I'm waiting for the next alpha update. :/

win1010525, 2024-02-20:
Airportitlwm works fine, I think it's not a bug from this kext.
[image]
Furthermore, if you want to investigate this issue, you can use 'pmset -g log' to get a full power log and analyze it.

win1010525, 2024-02-20:
> Airportitlwm works fine, I think it's not a bug from this kext. ![截屏2024-02-20 15 49 10](https://private-user-images.githubusercontent.com/50404193/306176370-ac5c73b7-ea96-4e44-89f9-cda89ce51932.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg0MTYyNzQsIm5iZiI6MTcwODQxNTk3NCwicGF0aCI6Ii81MDQwNDE5My8zMDYxNzYzNzAtYWM1YzczYjctZWE5Ni00ZTQ0LTg5ZjktY2RhODljZTUxOTMyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIwVDA3NTkzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThhOTBhZGFjMGZjMGNlNDFkOWUxMTM3MjkxNDNjMzAzOWJiYjY5MTYyZmViMDA0ZTExZDM3MWVhNGQ2MzkzNDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VLNNhgzW7Vh3fa-9zNEGOIhJjFBSFVI94I8i94REZIQ) Furthermore, if you want to investigate this issue, you can use 'pmset -g log' to get a full power log and analyze it. Here's mine ![截屏2024-02-20 15 57 17](https://private-user-images.githubusercontent.com/50404193/306178580-51e545b7-f4ef-43dd-a4c6-6918636becb5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg0MTYyNzQsIm5iZiI6MTcwODQxNTk3NCwicGF0aCI6Ii81MDQwNDE5My8zMDYxNzg1ODAtNTFlNTQ1YjctZjRlZi00M2RkLWE0YzYtNjkxODYzNmJlY2I1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIwVDA3NTkzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ0NWI3YjVjMzkwNzdkZThmNzUyY2U1Nzk4MzhhYzc4YTYzNTFiOWI0ZThlMjU4MTE2NmQwM2UwMjExZjEzN2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Mw9YI821uBH-WSxuW-vNTdiw9EbgzFbsjUhBUdcYywQ) By the way, have you tried to upgrade to macOS 14.4 Beta? I'm using the new system with the new developing kext, but this issue didn't happen on my device before upgrading, I'm not sure the relevance.

And, do not turn off RTC wake scheduling as there's a bug that causes more wake schedules.

zxystd, 2024-02-26:
[Sonoma.zip](https://github.com/OpenIntelWireless/itlwm/files/14401722/Sonoma.zip)
Is anyone can try this one? I had do little optimization, maybe it will have advantages on sleep/wake. kext is for 14.3 and below, not for 14.4.

denizckai, 2024-02-26:
> [Sonoma.zip](https://github.com/OpenIntelWireless/itlwm/files/14401722/Sonoma.zip) Is anyone can try this one? I had do little optimization, maybe it will have advantages on sleep/wake. kext is for 14.3 and below, not for 14.4.

it works! Thank you so much... :) Thanks for your efforts and thanks for all the help guys.

zxystd, 2024-02-26:
@denizckai does the wifi really stop on sleep? and can you upload the running log of itlwm after sleep and wakeup?

denizckai, 2024-02-26:
> @denizckai does the wifi really stop on sleep? and can you please upload the running log of itlwm after sleep and wakeup?

[bugreport_36202.zip](https://github.com/OpenIntelWireless/itlwm/files/14402132/bugreport_36202.zip)

radianttap, 2024-02-26:
> [Sonoma.zip](https://github.com/OpenIntelWireless/itlwm/files/14401722/Sonoma.zip) Is anyone can try this one? I had do little optimization, maybe it will have advantages on sleep/wake. kext is for 14.3 and below, not for 14.4.

This build resolved sleep, on 14.3.1
Machine went to sleep entirely, as far as I am concerned: display off, fans all off, USB keyboard lights went off. It woke up from sleep as expected.

Here are the logs
[Log_2024-02-26_10-18-21.log.zip](https://github.com/OpenIntelWireless/itlwm/files/14402350/Log_2024-02-26_10-18-21.log.zip)

zxystd, 2024-02-26:
@radianttap @denizckai Got it, thanks.

radianttap, 2024-02-26:
I can confirm the Power Nap works as expected with this build: macOS periodically wakes up, does its thing and goes back to sleep.

closed 2024-02-27 by denizckai

maxpicelli, 2024-03-09:
> [Sonoma.zip](https://github.com/OpenIntelWireless/itlwm/files/14401722/Sonoma.zip) Is anyone can try this one? I had do little optimization, maybe it will have advantages on sleep/wake. kext is for 14.3 and below, not for 14.4.

Thanks @zxystd  if was possible you make same change to 14.4 Sonoma build? Works my sleep on 14.3.1, and in 14.4 but without Airport Intel Kext

zxystd, 2024-03-09:
@maxpicelli Yes, it was merged already, now you can download them from alpha release page.

maxpicelli, 2024-03-09:
Thanks for your great Job, please show us how to contribute for you!!

derpuma, 2024-03-10:
Are there new kext for 14.4 regarding to Bluetooth? I tried with v2.3.0-alpha from link above. 
But Bluetooth does not work on my AX210s under Sonoma 14.4, also did not work under 14.4beta.
It works fine under Monterey with the same OpenCore (exept that I use there the kext Monterey needs).

In Sonoma I can see the Bluetooth as USB Hardware, but under Bluetooth nothing is loaded:

[image]

[image]

Kextstat says everything is loaded:
[image]

Any help?

denizckai, 2024-03-10:
> Are there new kext for 14.4 regarding to Bluetooth? I tried with v2.3.0-alpha from link above. But Bluetooth does not work on my AX210s under Sonoma 14.4, also did not work under 14.4beta. It works fine under Monterey with the same OpenCore (exept that I use there the kext Monterey needs).
> 
> In Sonoma I can see the Bluetooth as USB Hardware, but under Bluetooth nothing is loaded:
> <img alt="Bildschirmfoto 2024-03-10 um 15 35 23" width="562" src="https://private-user-images.githubusercontent.com/7608637/311517777-8af56485-ded6-4596-bff9-c34fa113abd4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwODIxOTYsIm5iZiI6MTcxMDA4MTg5NiwicGF0aCI6Ii83NjA4NjM3LzMxMTUxNzc3Ny04YWY1NjQ4NS1kZWQ2LTQ1OTYtYmZmOS1jMzRmYTExM2FiZDQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTBUMTQ0NDU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWRhZjY0ZmE5OTUxMjQxOGU5MWUwMzRlMDAxMTE4MmI2OWE1NmM3ZWI5N2JkMDc5OGVlNGNhMmZiYTNjZDQyMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Y12650MQ5L6b2VMM95Zw__80l8lTHDnrRdyJnfE5eyY"> <img alt="Bildschirmfoto 2024-03-10 um 15 35 37" width="429" src="https://private-user-images.githubusercontent.com/7608637/311517845-ed52869e-adc8-4f0a-a7b9-b185a3a4239c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwODIxOTYsIm5iZiI6MTcxMDA4MTg5NiwicGF0aCI6Ii83NjA4NjM3LzMxMTUxNzg0NS1lZDUyODY5ZS1hZGM4LTRmMGEtYTdiOS1iMTg1YTNhNDIzOWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTBUMTQ0NDU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzIwM2Q0NDI1Njg3NmZlNGE3MDlhMjNiNDU4OGNmNjkxMWEwN2I5OWNhMGJkMDdmMWM5NmE5YjcyZTI1MGYyOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Uj2LXAeMw_pcd_X0WKkhFW62dBJ5MNRCVEtdUIo5AmM">
> 
> Kextstat says everything is loaded: <img alt="Bildschirmfoto 2024-03-10 um 15 39 50" width="1250" src="https://private-user-images.githubusercontent.com/7608637/311517920-65004309-d395-4749-bba9-c76f246fb3a8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwODIxOTYsIm5iZiI6MTcxMDA4MTg5NiwicGF0aCI6Ii83NjA4NjM3LzMxMTUxNzkyMC02NTAwNDMwOS1kMzk1LTQ3NDktYmJhOS1jNzZmMjQ2ZmIzYTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTBUMTQ0NDU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDUxNWYxZmY4MWRjMWNhNzY3ODc2MzJmZjE3Njg3M2MzZjM4MTAyMDE1YjlkZThiODUzZmQ0NWM0M2IxYzdmOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.B9KjiLs51zIIMVBoqU62fHzxxQlZn7VoD5nSXs4Fzdc">
> 
> Any help?

I had to remap USB because Bluetooth problem after the 14.4 update (I remapped with USBtoolbox in Windows by the way). I suggest you do USB mapping again.

derpuma, 2024-03-10:
I already tried that, but was not succesfull so far. I mapped all ports with the USBtoolbox, made the kext file and integraded the USBToolBox.kext and UTBMap.kext in EFI/OC/Kext and config.plist.
How did you map this port? I see it in Windows under the onBoard AMD USB 3.1 controller at port 6:
[image]

denizckai, 2024-03-10:
> I already tried that, but was not succesfull so far. I mapped all ports with the USBtoolbox, made the kext file and integraded the USBToolBox.kext and UTBMap.kext in EFI/OC/Kext and config.plist. How did you map this port? I see it in Windows under the onBoard AMD USB 3.1 controller at port 6: ![image jpeg 215253f8f694fe96b3343bbe0b59c27b](https://private-user-images.githubusercontent.com/7608637/311519227-30054692-561e-443e-8e6f-18dc97458ec7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwODMyMDcsIm5iZiI6MTcxMDA4MjkwNywicGF0aCI6Ii83NjA4NjM3LzMxMTUxOTIyNy0zMDA1NDY5Mi01NjFlLTQ0M2UtOGU2Zi0xOGRjOTc0NThlYzcuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTBUMTUwMTQ3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTk4NDRhMzQ4ODMxNWVhZDEyNjZjNDgwNGQ4NzhjZjA0OTM1MGFiMmJkZDI4YzNkMzQ3MDdhNTUzZTgzZjA2OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.m9ncZdPgIU3G8eIGGXKegdrnJcZBaWx1herxTxPplgE)

Bluetooth port (port 6) must be internal so you should change USB type A to internal (enter port selection and write t:6:255 and press enter). You can search for how to remap USB on Google).

zxystd, 2024-03-10:
For Hackintosh related problems please turn to the forum for help. Thanks.

derpuma, 2024-03-10:
> Bluetooth port (port 6) must be internal so you should change USB type A to internal (enter port selection and write t:6:255 and press enter). You can search for how to remap USB on Google).

Do you mind sharing the kext you use for BT under 14.4? I think it is not a USB Port Problem, after mapping the port to internal nothing changed. BT is shown as USB Hardware, but nothing under Bluetooth.

maxpicelli, 2024-03-10:
My hardwares AX211 and AX210 works all fine with the same Bluetooth Kexts

Maybe you should try compile Bluetooth Kexts in your machine 

This is an alternative method from @chris1111 user

https://github.com/chris1111/Wifi-Intel-KextsBuilder/releases/tag/V1 

Only use Bluetooth kexts, not Wireless for Sonoma beta or 14.4