OpenIntelWireless/itlwm #938: macOS Sonoma - Unstable upload speeds with Intel 3168 Wifi

uyukat, 2023-11-23:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
- Kext Version: 2.2.0, 2.3.0 (current alpha)
- WiFi Card Model: Intel Dual Band Wireless-AC 3168
- PCI Product ID: Vendor: 8086, Device: 24FB, Sub Vendor: 21108086, REV_10
- macOS Version: Sonoma 14.1.1

**Description**
Using itlwm + HeliPort gets very good connection speeds overall but upload speeds will consistently dip down to 0.00Mbps for minutes at a time, seemingly at random. This causes streaming on Discord to take a hit in quality and some programs will fail to activate their licenses online if started while the upload speeds are at 0Mps.

**Bug Report Archive**
[bugreport_5350.zip](https://github.com/OpenIntelWireless/itlwm/files/13450782/bugreport_5350.zip)

**Kext Download Source**
Kexts tested were the precompiled itlwm.kext from the official Github project page, versions 2.3.0 and 2.2.0 just for good measure. Both versions showed this upload speed drop.

uyukat, 2023-11-23:
Screenshot of speed tests from Windows:
[image]
Screenshot of speed tests from Mac:
[image]
Mac would typically get similar results from the Windows screenshot, but tends to end back up at 0Mbps upload speed as shown in the second screenshot whereas Windows has not for me.

uyukat, 2023-11-28:
After some more testing between versions, it seems the problem only started occurring in v1.3.0 and higher. v1.2.0 is slow but the upload speed stays consistent and hasn't dipped to 0 for me yet, in case anyone else is having trouble with the current versions and needs a temporary solution.

zxystd, 2024-03-09:
[image]
stable with ac3168.

closed 2024-03-09 by zxystd