OpenIntelWireless/itlwm #942: AirportItlwm-2.3.0-Sonoma iServices not works

tuancuong92, 2023-12-05:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm 2.3.0-beta
- WiFi Card Model: Intel Wifi 6 AX201
- PCI Product ID: 0x06F0
- macOS Version: Sonoma 14.1.2

**Description**
I have clean install new Sonoma build on my PC, I used AirportItlwm 2.3.0-sonoma kext for compabile solution. I can login to iCloud and AppStore, but I cant log in to iMessage alsol Facetime. After that, I switched to itlwm kext 2.2.0 and HeliPort app, iServices working perfectly as before.

shoaibadel, 2023-12-06:
the same her 
found any solution?

abhishekthomasv, 2023-12-09:
> the same her 
> 
> found any solution?
> 
> 

Try disabling System integrity protection. Worked for me when I did that.

medinaandrez, 2023-12-09:
> > the same her
> > found any solution?
> 
> Try disabling System integrity protection. Worked for me when I did that.

How can i disable it?

radianttap, 2023-12-10:
> I have clean install new Sonoma build on my PC, I used AirportItlwm 2.3.0-sonoma kext for compatible solution. I can login to iCloud and AppStore, but I cant log in to iMessage alsol Facetime. 

I have this with AX200 on Asus B550I board while the same AX200 works great on Gigabyte B550I. Pretty confusing...what else in the system could influence this..?

beingVinesh, 2023-12-11:
> > the same her
> > found any solution?
> 
> Try disabling System integrity protection. Worked for me when I did that.

So by disabling SIP makes iServices like iMessage & FaceTime work. isn't it?

aikins01, 2023-12-13:
@abhishekthomasv what csr config value did you use to disable SIP

Sainahab, 2023-12-20:
@abhishekthomasv @beingVinesh @radianttap iservices activation is broken in Sonoma, fresh install ventura then activate iservices then update to sonoma !!!

medinaandrez, 2023-12-21:
> @abhishekthomasv @beingVinesh @radianttap iservices activation is broken in Sonoma, fresh install ventura then activate iservices then update to sonoma !!!

Does it work upgrading from Monterey?

beingVinesh, 2024-01-08:
> @abhishekthomasv @beingVinesh @radianttap iservices activation is broken in Sonoma, fresh install ventura then activate iservices then update to sonoma !!!

Tried installing BigSur then activated iServices and Updated to Sonoma. Still no Luck bro

lifeknife10A, 2024-01-17:
From Ventura "in which iservices were working fine" to Sonoma "in which iservices aren't working fine". I disabled the SIP by going to recovery mode then Terminal and then I wrote the command "csrutil disable" which disabled the SIP. then went to amazon website logged in and when it had asked for OTP I did not received it on the hackintosh. I received it on my iPhone as well as iPad. Keep in mind that SMS forwarding for both is enabled. 

My specs :-

CPU = Intel(R) i5-1035G1 [ICE LAKE]
Network Card = Intel Wireless AC 9560 160Mhz

mahronid, 2024-01-20:
> From Ventura "in which iservices were working fine" to Sonoma "in which iservices aren't working fine". I disabled the SIP by going to recovery mode then Terminal and then I wrote the command "csrutil disable" which disabled the SIP. then went to amazon website logged in and when it had asked for OTP I did not received it on the hackintosh. I received it on my iPhone as well as iPad. Keep in mind that SMS forwarding for both is enabled.
> 
> My specs :-
> 
> CPU = Intel(R) i5-1035G1 [ICE LAKE] Network Card = Intel Wireless AC 9560 160Mhz Kext = 2.3.0 Airportlwn Sonoma Alpha

So, is iService working well in Sonoma w/ 9560 wireless card?

ShroomKing, 2024-01-27:
> From Ventura "in which iservices were working fine" to Sonoma "in which iservices aren't working fine". I disabled the SIP by going to recovery mode then Terminal and then I wrote the command [...]

Same with Intel AX211, features that worked great in Ventura but stopped working with Sonoma:

- SMS Forwarding stopped working
- Receive Calls on Other Devices stopped working
- Share Focus Mode Across Devices stopped working
- iMessage doesn't sync with iCloud, cannot receive messages but sending works(sometimes...)
- FaceTime doesn't work at all
- Reminders, Calendar, iCloud Tabs all take a very long time to Sync with iCloud
- Can't receive Apple 2FA on Sonoma

rarreeees, 2024-02-15:
Same Intel AX201, Sonoma 14.2.1 and 14.3, iServices not working
Plugged in iPhone with hotspot and everything worked
Installed BigSur and Ventura, both working perfectly with v2.2.0 Airportitlwm
Hoping a fix for Sonoma comes soon

ezorith, 2024-02-17:
Hiya wanted to add the same on Intel Wireless-AC 9560, Sonoma 14.3.1 fresh install.
Exactly the same as above - issue resolved once iMessage and FaceTime are setup via USB hotspot.

fuzedxb1, 2024-02-23:
Commenting here as well have a AX210 upgraded to Sonoma and iservices don't work on 2.20. iMessage, FaceTime, focus mode device sharing from what I tested. Plugged in via Ethernet and boom all my services started syncing lol

LubosKolouch, 2024-02-24:
Problem persists with alpha-3999899

lifeknife10A, 2024-02-24:
> issue resolved once iMessage and FaceTime are setup via USB hotspot.

So basically your trying to say I erase my entire hackintosh and re install it using my iPhone's hotspot connected via a USB??

radianttap, 2024-02-24:
> > issue resolved once iMessage and FaceTime are setup via USB hotspot.
> 
> So basically your trying to say I erase my entire hackintosh and re install it using my iPhone's hotspot connected via a USB??

No, I think they are saying something else.
When connected to internet over iPhone/USB, iServices work OK. Connected over WiFi they do not work. Something in the driver is preventing the connection / sync.

On my Sonoma build, Messages work over iPhone/USB and over Ethernet (I225-V). Not when using WiFi through Airportitlwm.

lifeknife10A, 2024-02-24:
> No, I think they are saying something else. When connected to internet over iPhone/USB, iServices work OK. Connected over WiFi they do not work. Something in the driver is preventing the connection [...]

Hmmm...

lifeknife10A, 2024-02-24:
Let's see if there are any further changes in the kext file. hoping for the best...

anonymous-writer, 2024-02-27:
By using HeliPort + itlwm.kext iMessenger works perfekt. Something is different by using AirportItlwm.
May be it is possible to find someting in the system logs.

lifeknife10A, 2024-02-27:
> By using HeliPort + itlwm.kext iMessenger works perfekt. Something is different by using AirportItlwm. Maybe it is possible to find something helpful infos in the system logs.

Hmm. I have an idea about that heliport + itwln.kext works beautifully but the goal is that I services work with airportwln....

Umenezumi, 2024-03-08:
Pessimistic thinking : is it impossible to solve?

lifeknife10A, 2024-03-08:
> Pessimistic thinking : is it impossible to solve?

I'm not a developer so I can't say weather it's possible or not... I guess we just need to wait and hope that the developers find a way to solve this issue... 😅 Mostly it should be possible but let's see then.

adamilmanns, 2024-03-09:
@zxystd please notice.

radianttap, 2024-03-10:
Looking at the size of the issue, I think zxystd needs some help if this is to be solved / maintain furthered. It's a lot of work...

lifeknife10A, 2024-03-10:
> work...

If I were a developer I would have helped zxystd but I ain't 😅🥲

Kylar6, 2024-03-11:
AX211 cannot open WIFI on macOS+Sonoma14.4, using the latest 2.3.0alpha beta version

lifeknife10A, 2024-03-11:
> AX211 cannot open WIFI on macOS+Sonoma14.4, using the latest 2.3.0alpha beta version

Were you able to update from MacOS 14.3.1 to MacOS 14.4

Kylar6, 2024-03-11:
试试OC9.9才更新的，我是抹掉重新安装的

lifeknife10A, 2024-03-11:
> 试试OC9.9才更新的，我是抹掉重新安装的

You mean 0.9.9 ?

Umenezumi, 2024-03-12:
> 试试 OC9.9 才更新的，我是抹掉重新安装的

你的意思是说你用 OC 0.9.9 后 iService 正常？

Umenezumi, 2024-03-12:
@zxystd 

你好，我在更新到 macOS 14.4 、 OpenCore 0.9.9 之后 

iMessage 似乎恢复了一部分功能，目前账号是可以正常在「Message」上登录了（之前一直处于 loading 状态）

并且通过 iCloud 同步到了几天之前的所有短信（不清楚为啥没有最近几天的）

不过 iMessage 的收发仍有问题，劳烦知悉

Odroid13, 2024-03-14:
I also have the same problem, I think I found the problem. Now the developer needs to confirm or not.

https://github.com/OpenIntelWireless/itlwm/issues/966

lifeknife10A, 2024-03-14:
> @zxystd 
> 
> 
> 
> 你好，我在更新到 macOS 14.4 、 OpenCore 0.9.9 之后 
> 
> 
> 
> iMessage 似乎恢复了一部分功能，目前账号是可以正常在「Message」上登录了（之前一直处于 loading 状态）
> 
> 
> 
> 并且通过 iCloud 同步到了几天之前的所有短信（不清楚为啥没有最近几天的）
> 
> 
> 
> 不过 iMessage 的收发仍有问题，劳烦知悉
> 
> 
> 
> ------- 
> 
> （我的英语不太好，这是我用翻译软件生成的：My English is not very good. I generated it with translation software.）
> 
> Hello, after I updated to macOS 14.4, OpenCore 0.9.9, iMessage seems to have restored some functions. At present, the account can log in on "Message" normally (it has been loading before) and sync to all SMS from a few days ago through iCloud (I don't know why there are no recent days). However, there are still problems with the sending and receiving of iMessage. Please know.

Same here with Intel AC 9650 I'm able to sync in messages. But I am not able to login to FaceTime for some reason

anonymous-writer, 2024-03-16:
Could it be that this bug report describes the sending problem?

[sharingd.txt](https://github.com/OpenIntelWireless/itlwm/files/14624333/sharingd.txt)

radianttap, 2024-03-16:
No. It's irrelevant.

lisai9093, 2024-03-21:
Same problem here with latest alpha version..... Only itlwm works but not AirportItlwm

rbelusko, 2024-03-31:
Same issue here. iMessage only works with itlwm, but not AirportItlwm.
Want AirportItlwm because that enables airplay/airdrop.

I'm using a Dell Lattitude 5480 with Sonoma 14.3.1.

Reading an earlier post here, it sounds like an update to Sonoma 14.4 might resolve this?

SudoSuBash, 2024-04-05:
> @zxystd
>
> 你好，我在更新到 macOS 14.4 、 OpenCore 0.9.9 之后
>
> iMessage 似乎恢复了一部分功能，目前账号是可以正常在「Message」上登录了（之前一直处于 loading 状态）
>
> 并且通过 iCloud 同步到了几天之前的所有短信（不清楚为啥没有最近几天的）
>
> 不过 iMessage 的收发仍有问题，劳烦知悉
>
> ------- [...]

我在14.4下也出现了类似状况.这个问题现在怎么样了?
在ventura下它能工作正常吗？如果可以的话我就降回ventura了。
现在的情况是(14.4):
1.可以正常登陆iMessage，但是无法发送接收信息。当我尝试发送信息的时候，它会提示我接收失败。
2.Facetime无法正常拨打电话。
3.无法接收双重验证码

I have the same issue in macOS 14.4.How is the status of this issue now?
Can it work well on Ventura?If so,I will downgrade to it.
Now there are many problems on 14.4:
1.I CAN SIGN IN iMessage NORMALLY,but I cannot send and receive the message.When I send the message,it will tell me "send fail".
2.Facetime cannot call and phone normally.
3.Cannot receive 2FA Verification Code from my Mac.

anonymous-writer, 2024-04-10:
A pragmatic solutions can be found under this LINK to solve the problem. It might help to find other solutions. I wrote this in German. Sorry for that.

https://www.hackintosh-forum.de/forum/thread/59482-intel-wifi-unter-sonoma-l%C3%B6sungsfindungen-imessage-and-facetime-funktionieren-nic/?postID=789579#post789579

lifeknife10A, 2024-04-10:
> A pragmatic solutions can be found under this LINK to solve the problem. It might help to find other solutions. I wrote this in German. Sorry for that. [...]

Excellent work.... But we need a separate "server" to solve this issue. Correct if I'm wrong but would MacOS assume this as a "WI-FI connection" or an "Ethernet Connection" because if MacOS assumes this as an Ethernet connection then would it be same as using Itwlm.kext?   Please correct me if I'm wrong somewhere Thank you very much.

anonymous-writer, 2024-04-10:
MacOS assume this as "Ethernet Connection".
But it is not the same as Itwlm.kext. WI-FI connection is loading before login and localservice ar working.

This approach brings the kext back to the functionality of Ventura.

lifeknife10A, 2024-04-10:
> MacOS assume this as "Ethernet Connection". But it is not the same as Itwlm.kext. WI-FI connection is loading before login and location service are working.
> 
> This approach brings the kext back to the functionality of Ventura.

Ahh it makes sense.. Although it's very risky for me to do  and also I don't have any servers but great work  🙌🏻🙌🏻🙌🏻

SudoSuBash, 2024-04-13:
> MacOS assume this as "Ethernet Connection". But it is not the same as Itwlm.kext. WI-FI connection is loading before login and location service are working.
> 
> This approach brings the kext AirportItlwm.kext back to the functionality of Ventura.

According to your thread,the Airdrop seems works fine.
How do you achieved it? Could you give me some advice about how to do it?
My English does not well,plz forgive me :)

ResQre, 2024-04-23:
> A pragmatic solutions can be found under this LINK to solve the problem. It might help to find other solutions. I wrote this in German. Sorry for that. [...]

Great work ! Quite stable on Sonoma 14.4.1 . By the way, I use the free proxy method.

ResQre, 2024-04-23:
> According to your thread,the Airdrop seems works fine. How do you achieved it? Could you give me some advice about how to do it? My English does not well,plz forgive me :)

1. 下載["Null Ethernet Network Driver by RehabMan"](https://bitbucket.org/anonymous_writer/os-x-null-ethernet/src/master/)，安裝裡面的ssdt-rmne及NullEthernet.kext。ssdt-rmne裡內建網卡MAC為112233445566，你可以使用任何一個MAC生成器生成序列來取代。完成後把系統設定/網路裡的每一項網路服務設定都刪除，然後刪除Library/SystemConfiguration/NetworkInterfaces.plist，重開機後到系統設定/網路，新增一個Wi-Fi和一個乙太網路。打開系統資訊/網路，確定乙太網路為en0，Wi-Fi為en1。
[image]

2. 從[“這裡”](https://vpnoverview.com/privacy/anonymous-browsing/free-proxy-servers/)選一個你喜歡的免費proxy-server。
3. 讓Wi-Fi連上網路後，記錄下TCP/IP及DNS的設定。手動在乙太網路的設定裡填上TCP/IP（不可與Wi-Fi 位址完全相同）及DNS，並在proxy server選項內，填入你選好的免費proxy server設定。
[image]

4. 把Wi-Fi的服務順序置頂。這樣就成功了。
5. 注意，若連上不同的無線AP，會有不同的IP，乙太網路的IP，DNS就要跟著更改，這時候可以使用網路設定裡的“位置”，來切換不同的設定。比如“家”和“熱點”。
6. 若因為重開機或休眠導致iService失效，請在網路服務順序裡變換Wi-Fi和乙太網路的順序，幾分鐘後再回復Wi-Fi置頂，就可以恢復連線。

lifeknife10A, 2024-04-24:
> > > MacOS assume this as "Ethernet Connection". But it is not the same as Itwlm.kext. WI-FI connection is loading before login and location service are working.
> > > This approach brings the kext AirportItlwm.kext back to the functionality of Ventura.
> > 
> > 
> > According to your thread,the Airdrop seems works fine. How do you achieved it? Could you give me some advice about how to do it? My English does not well,plz forgive me :)
> 
> 1. 下載["Null Ethernet Network Driver by RehabMan"](https://bitbucket.org/anonymous_writer/os-x-null-ethernet/src/master/)，安裝裡面的ssdt-rmne及NullEthernet.kext。ssdt-rmne裡內建網卡MAC為112233445566，你可以使用任何一個MAC生成器生成序列來取代。完成後把系統設定/網路裡的每一項網路服務設定都刪除，然後刪除Library/SystemConfiguration/NetworkInterfaces.plist，重開機後到系統設定/網路，新增一個Wi-Fi和一個乙太網路。打開系統資訊/網路，確定乙太網路為en0，Wi-Fi為en1。
>    ![Info-Lan](https://private-user-images.githubusercontent.com/75792086/324854055-6e873c7b-4ca1-43b9-ac06-f1741f1a1921.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5Mzk4ODIsIm5iZiI6MTcxMzkzOTU4MiwicGF0aCI6Ii83NTc5MjA4Ni8zMjQ4NTQwNTUtNmU4NzNjN2ItNGNhMS00M2I5LWFjMDYtZjE3NDFmMWExOTIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA2MTk0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgzN2VhNzJjZjAzNDVmZjcyOWNjMDgxZDdlMGM2N2VmOWViMTlhMjFhMjhiODFiZjdlM2Q5ZjJmZTVjOWNkN2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.PYMh3xiLRTUqILQOczYveEZdP3_WI0U5RHqirpdF2XA)
> 2. 從[“這裡”](https://vpnoverview.com/privacy/anonymous-browsing/free-proxy-servers/)選一個你喜歡的免費proxy-server。
> 3. 讓Wi-Fi連上網路後，記錄下TCP/IP及DNS的設定。手動在乙太網路的設定裡填上TCP/IP（不可與Wi-Fi 位址完全相同）及DNS，並在proxy server選項內，填入你選好的免費proxy server設定。
>    ![Bildschirmfoto 2024-04-10 um 18 54 53](https://private-user-images.githubusercontent.com/75792086/324854169-52036589-3031-40b4-87b9-f0a78776e96b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5Mzk4ODIsIm5iZiI6MTcxMzkzOTU4MiwicGF0aCI6Ii83NTc5MjA4Ni8zMjQ4NTQxNjktNTIwMzY1ODktMzAzMS00MGI0LTg3YjktZjBhNzg3NzZlOTZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA2MTk0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg5M2M0NDBmMTM1NzZiNGIyOTY2ZmE1NDQwNDJmZGM3ZGRiODc5NjNjZmM3ZjU4NmY5YTllNWU5NTFmN2Q1YTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Vc6LFBTuJfwHXftrR16I9wXsvyefaRQFAMoBsnfvR0Y)
> 4. 把Wi-Fi的服務順序置頂。這樣就成功了。
> 5. 注意，若連上不同的無線AP，會有不同的IP，乙太網路的IP，DNS就要跟著更改，這時候可以使用網路設定裡的“位置”，來切換不同的設定。比如“家”和“熱點”。
> 6. 若因為重開機或休眠導致iService失效，請在網路服務順序裡變換Wi-Fi和乙太網路的順序，幾分鐘後再回復Wi-Fi置頂，就可以恢復連線。

In this workaround, I believe both "airportlwm.kext" and "itwlm.kext" should be disabled correct.. 

"在此解決方法中，我相信“airportlwm.kext”和“itwlm.kext”都應該正確停用。"

lifeknife10A, 2024-04-24:
> 1. 下載["Null Ethernet Network Driver by [...]

And How do you replace the MAC address ?? as in where do you replace the generated MAC address....
Also where do you recommend us to generate our MAC address.

ResQre, 2024-04-24:
> In this workaround, I believe both "airportlwm.kext" and "itwlm.kext" should be disabled correct..
>
> "在此解決方法中，我相信“airportlwm.kext”和“itwlm.kext”都應該正確停用。"

No. Once you disable “airportitlwm.kext “, you drop the intel wifi.
This workaround was created to solve the iService not working properly while using airportitlwm.kext.
So airportitlwm.kext is mandatory.

ResQre, 2024-04-24:
> And How do you replace the MAC address ?? as in where do you replace the generated MAC address.... Also where do you recommend us to generate our MAC address.

The default MAC is addressed in ssdt-rmne. You can use any available tool, like “OC auxiliary tool “ to generate a random MAC to replace it.

lifeknife10A, 2024-04-24:
> The default MAC is addressed in ssdt-rmne. You can use any available tool, like “OC auxiliary tool “ to generate a random MAC to replace it.

So I Have added "SSDT-RMNE.aml" onto my Plist file in OC Auxiliary Tool... now how to replace the MAC.

ResQre, 2024-04-24:
> So I Have added "SSDT-RMNE.aml" onto my Plist file in OC Auxiliary Tool... now how to replace the MAC.

You need to edit ssdt-rmne. 
[image]

lifeknife10A, 2024-04-24:
> > > > > > > MacOS assume this as "Ethernet Connection". But it is not the same as Itwlm.kext. WI-FI connection is loading before login and location service are working.
> > > > > > > This approach brings the kext AirportItlwm.kext back to the functionality of Ventura.
> > > > > > 
> > > > > > 
> > > > > > According to your thread,the Airdrop seems works fine. How do you achieved it? Could you give me some advice about how to do it? My English does not well,plz forgive me :)
> > > > > 
> > > > > 
> > > > > 
> > > > > 1. 下載["Null Ethernet Network Driver by RehabMan"](https://bitbucket.org/anonymous_writer/os-x-null-ethernet/src/master/)，安裝裡面的ssdt-rmne及NullEthernet.kext。ssdt-rmne裡內建網卡MAC為112233445566，你可以使用任何一個MAC生成器生成序列來取代。完成後把系統設定/網路裡的每一項網路服務設定都刪除，然後刪除Library/SystemConfiguration/NetworkInterfaces.plist，重開機後到系統設定/網路，新增一個Wi-Fi和一個乙太網路。打開系統資訊/網路，確定乙太網路為en0，Wi-Fi為en1。
> > > > >    ![Info-Lan](https://private-user-images.githubusercontent.com/75792086/324854055-6e873c7b-4ca1-43b9-ac06-f1741f1a1921.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5NDA4NjgsIm5iZiI6MTcxMzk0MDU2OCwicGF0aCI6Ii83NTc5MjA4Ni8zMjQ4NTQwNTUtNmU4NzNjN2ItNGNhMS00M2I5LWFjMDYtZjE3NDFmMWExOTIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA2MzYwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThlZDVmY2I5MDg1OWM0NzZmYjM5ZmFiOTgzMjUzYzNmNWQ4MDU1OTA2OTAzODgwNjQ4MjBmOGIzZmI1MTEzNjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tjkt0uuEq-EUrffc82s0sw0egstvkYTMF9HF4h-XXvs)
> > > > > 2. 從[“這裡”](https://vpnoverview.com/privacy/anonymous-browsing/free-proxy-servers/)選一個你喜歡的免費proxy-server。
> > > > > 3. 讓Wi-Fi連上網路後，記錄下TCP/IP及DNS的設定。手動在乙太網路的設定裡填上TCP/IP（不可與Wi-Fi 位址完全相同）及DNS，並在proxy server選項內，填入你選好的免費proxy server設定。
> > > > >    ![Bildschirmfoto 2024-04-10 um 18 54 53](https://private-user-images.githubusercontent.com/75792086/324854169-52036589-3031-40b4-87b9-f0a78776e96b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5NDA4NjgsIm5iZiI6MTcxMzk0MDU2OCwicGF0aCI6Ii83NTc5MjA4Ni8zMjQ4NTQxNjktNTIwMzY1ODktMzAzMS00MGI0LTg3YjktZjBhNzg3NzZlOTZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA2MzYwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg0NDk1MjM1NTA2Yzc0NDVmYWE0NjIwYTQ5MDYxY2ZmNTUzOGMxMGZhMDQ4MzU5NGNjYjZhNzZlNDhiYjIwZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.P96kWVZdfz-30XvHiLAR0TZWrv8Wk_TORzrHyalMSZA)
> > > > > 4. 把Wi-Fi的服務順序置頂。這樣就成功了。
> > > > > 5. 注意，若連上不同的無線AP，會有不同的IP，乙太網路的IP，DNS就要跟著更改，這時候可以使用網路設定裡的“位置”，來切換不同的設定。比如“家”和“熱點”。
> > > > > 6. 若因為重開機或休眠導致iService失效，請在網路服務順序裡變換Wi-Fi和乙太網路的順序，幾分鐘後再回復Wi-Fi置頂，就可以恢復連線。
> > > > 
> > > > 
> > > > And How do you replace the MAC address ?? as in where do you replace the generated MAC address.... Also where do you recommend us to generate our MAC address.
> > > 
> > > 
> > > The default MAC is addressed in ssdt-rmne. You can use any available tool, like “OC auxiliary tool “ to generate a random MAC to replace it.
> > 
> > 
> > So I Have added "SSDT-RMNE.aml" onto my Plist file in OC Auxiliary Tool... now how to replace the MAC.
> 
> You need to edit ssdt-rmne. <img alt="anonymous_writer os-x-null-ethernet SSDT-RMNE dsl — Bitbucket jpeg" width="744" src="https://private-user-images.githubusercontent.com/75792086/325111918-434750c1-12cb-46bc-bcb0-b4f92f1c435f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5NDIwNTUsIm5iZiI6MTcxMzk0MTc1NSwicGF0aCI6Ii83NTc5MjA4Ni8zMjUxMTE5MTgtNDM0NzUwYzEtMTJjYi00NmJjLWJjYjAtYjRmOTJmMWM0MzVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA2NTU1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZmNjEwMDhjYTM0OGFiMzdlYmY3ZmMxMWE0NWY5NWNlMzRkYjNjODVjMGUyNmVmNTQ5OGFkYzYwYzg0M2MxMDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Mb_yMz2lnOm4ys6FwaB7fG8uV-dQ0dg0ptASbuwKVYc">

So I need to remove everything from the curly brackets "0x11, 0x22, 0x33....." and then add my generated MAC address there ??...

ResQre, 2024-04-24:
> So I need to remove everything from the curly brackets "0x11, 0x22, 0x33....." and then add my generated MAC address there ??...

Yes. If your generated MAC is AABBCCDD… Then it should be 0xAA, 0xBB, 0xCC…

lifeknife10A, 2024-04-24:
> Yes. If your generated MAC is AABBCCDD… Then it should be 0xAA, 0xBB, 0xCC…

THANK YOU THANK YOU THANK YOU SO SO SO MUCH 🙏🏻🙏🏻

ResQre, 2024-04-24:
By the way, Airdrop works partially. The author recommends use “LocalSend “ instead. I tried and it works great.

lifeknife10A, 2024-04-24:
> By the way, Airdrop works partially. The author recommends use “LocalSend “ instead. I tried and it works great.

One last thing. So sorry to disturb you....

[image]

How to solve this now?

ResQre, 2024-04-24:
> > By the way, Airdrop works partially. The author recommends use “LocalSend “ instead. I tried and it works great.
> 
> One last thing. So sorry to disturb you....
> 
> ![Screenshot 2024-04-24 at 1 18 54 PM](https://private-user-images.githubusercontent.com/66748761/325129174-8724a582-4521-4e5f-827d-85352393fd64.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5NDc4NTUsIm5iZiI6MTcxMzk0NzU1NSwicGF0aCI6Ii82Njc0ODc2MS8zMjUxMjkxNzQtODcyNGE1ODItNDUyMS00ZTVmLTgyN2QtODUzNTIzOTNmZDY0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDA4MzIzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhNDMwZjFhYjRkOGYwNTE4YmIyZjJmYTNlOGMzZTllYmFmM2JhNGQ1ZjA0Mzk3NTg5NGFiM2EwMzk2OTAyODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RsXZTutv09sFsbwYn8bkjBAhnh3UnH-ecS0_SOn4ZvY)
> 
> How to solve this now?

The latest MaciASL can edit downloaded ssd-rmne.aml directly without any problem.

lifeknife10A, 2024-04-24:
> The latest MaciASL can edit downloaded ssd-rmne.aml directly without any problem.

Oh OK Thank you so much

[39 more replies]