OpenIntelWireless/itlwm #944: Time Machine over network doesn't work with AirportItlwm on Sonoma

usr-sse2, 2023-12-18:
**Have You Read Our Docs**
No

**Are You Reporting A Bug**
Yes

**Environment**
- Kext Version: v2.3.0-alpha
- WiFi Card Model: 8265
- PCI Product ID: 0x0150
- macOS Version: 14.2

**Description**
I have a Samba share on an OpenWRT router that was used as Time Machine backup destination. It used to work on Ventura, then I haven't used it for a long time because the HDD became full. Now I've upgraded to Sonoma and replaced the HDD on the router with a larger one, and I can no longer backup to it over Wi-Fi using AirportItlwm. I can setup the network backup destination successfully, but the initial backup doesn't start with a "network interruption" error, although the Wi-Fi does not disconnect. I've checked that backup to the same network destination works when using a wired Ethernet connection. If I do initial backup over Ethernet, then incremental backups over Wi-Fi also don't work.

I think that it may be related to the problem with iMessage and it may be easier to debug Time Machine than iMessage because you can control both sides of the connection.

Here is the error log of backupd:
```
TimeMachine	ошибка	14:18:56.617506+0300	backupd	Failed to enumerate instances of structure APFSBackup under path /Volumes/Резервные копии usrsse2's MacBook Pro 3
TimeMachine	ошибка	14:18:56.624702+0300	backupd	Failed to enumerate URLs under /Volumes/Резервные копии usrsse2's MacBook Pro 3 for SnapshotInProgressContainer reuse with error Error Domain=NSCocoaErrorDomain Code=260 "The file “Резервные копии usrsse2's MacBook Pro 3” couldn’t be opened because there is no such file." UserInfo={NSURL=file:///Volumes/%D0%A0%D0%B5%D0%B7%D0%B5%D1%80%D0%B2%D0%BD%D1%8B%D0%B5%20%D0%BA%D0%BE%D0%BF%D0%B8%D0%B8%20usrsse2's%20MacBook%20Pro%203/, NSFilePath=/Volumes/Резервные копии usrsse2's MacBook Pro 3, NSUnderlyingError=0x7fb33d526440 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}
TimeMachine	ошибка	14:18:56.625211+0300	backupd	Failed to open '/Volumes/Резервные копии usrsse2's MacBook Pro 3', error: No such file or directory
TimeMachine	ошибка	14:18:56.625426+0300	backupd	Failed to enumerate instances of structure ReadOnlyBackup under path /Volumes/Резервные копии usrsse2's MacBook Pro 3, error: Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory" UserInfo={NSFilePath=/Volumes/Резервные копии usrsse2's MacBook Pro 3, NSLocalizedDescription=/Volumes/Резервные копии usrsse2's MacBook Pro 3: No such file or directory (error 2)}
TimeMachine	ошибка	14:18:57.714359+0300	backupd	Error: (Error Domain=NSCocoaErrorDomain Code=4 "The file “2023-12-18-141856.inprogress” doesn’t exist." UserInfo={NSFilePath=/Volumes/Резервные копии usrsse2's MacBook Pro 3/2023-12-18-141856.inprogress, NSUnderlyingError=0x7fb33c797880 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}) Creating directory file:///Volumes/%D0%A0%D0%B5%D0%B7%D0%B5%D1%80%D0%B2%D0%BD%D1%8B%D0%B5%20%D0%BA%D0%BE%D0%BF%D0%B8%D0%B8%20usrsse2's%20MacBook%20Pro%203/2023-12-18-141856.inprogress
TimeMachine	ошибка	14:18:57.715003+0300	backupd	Failed to build new Backup with error Error Domain=NSCocoaErrorDomain Code=4 "The file “2023-12-18-141856.inprogress” doesn’t exist." UserInfo={NSFilePath=/Volumes/Резервные копии usrsse2's MacBook Pro 3/2023-12-18-141856.inprogress, NSUnderlyingError=0x7fb33c797880 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}
TimeMachine	ошибка	14:18:57.715770+0300	backupd	Failed to build new backup with error Error Domain=NSCocoaErrorDomain Code=4 "The file “2023-12-18-141856.inprogress” doesn’t exist." UserInfo={NSFilePath=/Volumes/Резервные копии usrsse2's MacBook Pro 3/2023-12-18-141856.inprogress, NSUnderlyingError=0x7fb33c797880 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}
TimeMachine	ошибка	14:18:57.716389+0300	backupd	Failed to prepare working set for new backup, error: Error Domain=NSCocoaErrorDomain Code=4 "The file “2023-12-18-141856.inprogress” doesn’t exist." UserInfo={NSFilePath=/Volumes/Резервные копии usrsse2's MacBook Pro 3/2023-12-18-141856.inprogress, NSUnderlyingError=0x7fb33c797880 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}
TimeMachine	ошибка	14:19:08.064285+0300	backupd	Backup failed because the destination disk was ejected or disconnected unexpectedly!
TimeMachine	ошибка	14:19:08.085268+0300	backupd	Backup failed: BACKUP_FAILED_DISK_INTERRUPTION (27)
```

And here is the cause of error:
```
apfs	стандартное	14:18:55.478142+0300	kernel	apfs_vfsop_sync:5127: disk4s1 disk4: failed to finish all transactions in sync() - Input/output error(5)
apfs	стандартное	14:18:55.478142+0300	kernel	apfs_vfsop_sync:5127: disk4s1 disk4: failed to finish all transactions in sync() - Input/output error(5)
diskimagesiod	стандартное	14:18:55.485939+0300	diskimagesiod	-[DIIODaemonDelegate unmountAll]: Forcing unmount of <private> (disk4s1)
apfs	стандартное	14:18:55.486876+0300	kernel	apfs_log_op_with_proc:3110: disk4s1 unmounting volume Резервные копии usrsse2's MacBook Pro, requested by: diskimagesiod (pid 65439); parent: launchd (pid 1)
apfs	стандартное	14:18:55.486891+0300	kernel	apfs_vfsop_unmount:3196: disk4s1 waiting for purgatory cleaner to finish
apfs	стандартное	14:18:55.486977+0300	kernel	apfs_vfsop_unmount:3370: disk4s1 failed to finish all transactions before unmount! (err 6)
diskarbitrationd	информационное	14:18:55.517729+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.518131+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.518294+0300	diskarbitrationd	  dispatched callback, id = 0000000000000001:0000000000000001, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.518464+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.518621+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.518841+0300	diskarbitrationd	  dispatched callback, id = 0000000000000001:0000000000000001, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.519263+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.519900+0300	diskarbitrationd	  dispatched callback, id = 000000000000000A:000000000000000A, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.520243+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.520577+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.520983+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.521342+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.521641+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.521878+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
diskarbitrationd	информационное	14:18:55.522108+0300	diskarbitrationd	  dispatched callback, id = 0000000000000002:0000000000000002, kind = disk description changed, disk = /dev/disk4s1, key = DAVolumePath.
[... cut ...] : No }>
storagekitd	стандартное	14:19:07.823371+0300	storagekitd	<SKAPFSDisk: [78] { BSD Name: disk4s1, Mounted: No, APFS UUID: FB7A9B58-EC5D-43B6-8C5E-A1BD58DDE33F, Encrypted: No }>
StorageKit	стандартное	14:19:07.828521+0300	wallpaperexportd	<SKAPFSDisk: [78] { BSD Name: disk4s1, Mounted: No, APFS UUID: FB7A9B58-EC5D-43B6-8C5E-A1BD58DDE33F, Encrypted: No }>
```

**Bug Report Archive**

**Kext Download Source**
[v2.3.0-alpha](https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.3.0-alpha)

f-1-0, 2024-01-29:
Can't confirm -
I just tried on Sonoma 23D56 and it works normally for me, with 2.3.0 alpha intel Wifi driver for AC9560 (and ethernet unplugged on my Z390 Designare). I use an original 3rd gen AirPort Time Capsule for storage though.

usr-sse2, 2024-02-28:
The problem is not specific to OpenWRT. I've tested in another network, and tried to back up to a Mac instead of SMB share on a router and got the same failure.
The next step is to check whether it is specific to 8265.

zxystd, 2024-03-01:
@usr-sse2 I think it is not specific to 8265. Maybe macOS some userspace applications/daemons don't treat that the network is UP.
Sonoma AirportItlwm implement is different from BRCMs, the network traffic is through to Ethernet interface which should be SkywalkNetworkInterface(although it is the subclass of Ethernet interface) on Sonoma. , the ioreg difference is here:
[image]
[image]

OpenNEONE, 2024-10-11:
I'm having the same problem with Sonoma 14.7, AX200