OpenIntelWireless/itlwm #945: AirportItlwm: KP after entering wifi setting

felicia-wen, 2023-12-19:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version:  v2.3.0-alpha
- WiFi Card Model: AX200
- PCI Product ID: 
- macOS Version: 14.2

**Description**

System reboot immediately after entering wifi/network settings when wifi is enabled.

```
panic(cpu 4 caller 0xffffff8012384f03): Kernel trap at 0xffffff8014650120, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000714e66000, CR4: 0x00000000003406e0
RAX: 0x0000000000000000, RBX: 0x0000000000000003, RCX: 0xffffff8016d9d9f0, RDX: 0x0000000000000003
RSP: 0xfffffffbe0963a30, RBP: 0xfffffffbe0963a40, RSI: 0xffffff801467e5a0, RDI: 0xffffff8b723ed800
R8:  0x0200000110000788, R9:  0x0200000110000000, R10: 0x0200000110000788, R11: 0x00ffffff00000000
R12: 0x0000000000000000, R13: 0x0000000000000000, R14: 0x000000000000000c, R15: 0x0000000000000000
RFL: 0x0000000000010282, RIP: 0xffffff8014650120, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 0

Panicked task 0xffffff9509c62638: 16 threads: pid 220: airportd
Backtrace (CPU 4), panicked thread: 0xffffff8b730780c8, Frame : Return Address
0xfffffffbe09633b0 : 0xffffff8012236c41 mach_kernel : _handle_debugger_trap + 0x4b1
0xfffffffbe0963400 : 0xffffff8012395310 mach_kernel : _kdp_i386_trap + 0x110
0xfffffffbe0963440 : 0xffffff8012384a5c mach_kernel : _kernel_trap + 0x55c
0xfffffffbe09634c0 : 0xffffff80121d3971 mach_kernel : _return_from_trap + 0xc1
0xfffffffbe09634e0 : 0xffffff8012236f2d mach_kernel : _DebuggerTrapWithState + 0x5d
0xfffffffbe09635d0 : 0xffffff80122365d3 mach_kernel : _panic_trap_to_debugger + 0x1e3
0xfffffffbe0963630 : 0xffffff80129d8bcb mach_kernel : _panic + 0x84
0xfffffffbe0963720 : 0xffffff8012384f03 mach_kernel : _sync_iss_to_iks + 0x2c3
0xfffffffbe09638a0 : 0xffffff8012384be7 mach_kernel : _kernel_trap + 0x6e7
0xfffffffbe0963920 : 0xffffff80121d3971 mach_kernel : _return_from_trap + 0xc1
0xfffffffbe0963940 : 0xffffff8014650120 com.apple.iokit.IOSkywalkFamily : __ZN25IOSkywalkNetworkInterface7_bpfTapEP7__ifnetjj + 0x2c
0xfffffffbe0963a40 : 0xffffff801246761d mach_kernel : _bpfioctl + 0x1bdd
0xfffffffbe0963a70 : 0xffffff8012466d74 mach_kernel : _bpfioctl + 0x1334
0xfffffffbe0963ab0 : 0xffffff80124660d7 mach_kernel : _bpfioctl + 0x697
0xfffffffbe0963b60 : 0xffffff80124565b8 mach_kernel : _spec_ioctl + 0x98
0xfffffffbe0963b90 : 0xffffff8012448974 mach_kernel : _VNOP_IOCTL + 0xc4
0xfffffffbe0963bf0 : 0xffffff801243b35b mach_kernel : _u32CharToUTF8Bytes + 0x91b
0xfffffffbe0963e00 : 0xffffff80126f1524 mach_kernel : _sys_fcntl_nocancel + 0x3954
0xfffffffbe0963e30 : 0xffffff8012753fc4 mach_kernel : _ioctl + 0x484
0xfffffffbe0963f40 : 0xffffff8012852c9d mach_kernel : _unix_syscall64 + 0x1dd
0xfffffffbe0963fa0 : 0xffffff80121d3db6 mach_kernel : _hndl_unix_scall64 + 0x16
      Kernel Extensions in backtrace:
         com.apple.iokit.IOSkywalkFamily(1.0)[1444E3C6-0B2D-300A-BE26-2D1AFDE9E65F]@0xffffff8014641000->0xffffff801466ffff
            dependency: com.apple.driver.mDNSOffloadUserClient(1.0.1b8)[FF5CF7B9-1643-3CA5-A871-DBAF3D1A167B]@0xffffff801454d000->0xffffff8014551fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[5D912FD8-C4CD-3CF6-B214-DF5BF7AB4FA0]@0xffffff80142f3000->0xffffff801430afff

Process name corresponding to current thread (0xffffff8b730780c8): airportd
Boot args: -v -nredvcn keepsyms=1 debug=0x100

Mac OS version:
23C64

Kernel version:
Darwin Kernel Version 23.2.0: Wed Nov 15 21:54:10 PST 2023; root:xnu-10002.61.3~2/RELEASE_X86_64
Kernel UUID: F73CFE5A-7EF6-3C21-AFA7-742645B733CC
roots installed: 0
KernelCache slide: 0x0000000011e00000
KernelCache base:  0xffffff8012000000
Kernel slide:      0x0000000011ee0000
Kernel text base:  0xffffff80120e0000
__HIB  text base: 0xffffff8011f00000
System model name: MacPro7,1 (Mac-27AD2F918AE68F61)
System shutdown begun: NO
Panic diags file available: NO (0xe00002bc)
Hibernation exit count: 0

System uptime in nanoseconds: 63279827211
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000000ebbc58c19
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000002fa7e8b302 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff803c8af000 - 0xffffffa03c8af000
  . PGZ   : 0xffffff803c8af000 - 0xffffff80408b0000
  . VM    : 0xffffff80408b0000 - 0xffffff850cbe3000
  . RO    : 0xffffff850cbe3000 - 0xffffff86a6249000
  . GEN0  : 0xffffff86a6249000 - 0xffffff8b7257c000
  . GEN1  : 0xffffff8b7257c000 - 0xffffff903e8af000
  . GEN2  : 0xffffff903e8af000 - 0xffffff950abe2000
  . GEN3  : 0xffffff950abe2000 - 0xffffff99d6f15000
  . DATA  : 0xffffff99d6f15000 - 0xffffffa03c8af000
  Metadata: 0xffffffa06de2d000 - 0xffffffa08de2d000
  Bitmaps : 0xffffffa08de2d000 - 0xffffffa093e2d000
  Extra   : 0 - 0
```

Aaryan-Anand, 2023-12-20:
Yep, having the same issue. Updated to Sonoma last night and switched to the alpha version, and now it crashes when I open the WiFi settings.

zxystd, 2023-12-25:
Are you sure you don't used the OCLP patches and not inject `IOSkywalkFamily.kext`?

felicia-wen, 2023-12-25:
> Are you sure you don't used the OCLP patches and not inject `IOSkywalkFamily.kext`?

Good afternoon,

No, I don't use any of these.

shank03, 2023-12-28:
Facing exact same issue (with same logs) here:
- Sonoma 14.2.1
- airportitlwm 2.3.0-alpha
- Intel AC-946x

```
panic(cpu 4 caller 0xffffff800ff84f03): Kernel trap at 0xffffff8012250120, type 14=page fault, registers:
CR0: 0x0000000080010033, CR2: 0x0000000000000000, CR3: 0x000000048f4eb036, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0x0000000000000003, RCX: 0xffffff8014f2e9f0, RDX: 0x0000000000000003
RSP: 0xfffffffe0019fa30, RBP: 0xfffffffe0019fa40, RSI: 0xffffff801227e5a0, RDI: 0xffffffa029071800
R8:  0x0200000110000c18, R9:  0x0200000110000000, R10: 0x0200000110000c18, R11: 0x00ffffff00000000
R12: 0x0000000000000000, R13: 0x0000000000000000, R14: 0x000000000000000c, R15: 0x0000000000000000
RFL: 0x0000000000010246, RIP: 0xffffff8012250120, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 0

Panicked task 0xffffffa029815638: 17 threads: pid 220: airportd
Backtrace (CPU 4), panicked thread: 0xffffffa02a0680c8, Frame : Return Address
0xfffffffe0019f3b0 : 0xffffff800fe36c41 
0xfffffffe0019f400 : 0xffffff800ff95310 
0xfffffffe0019f440 : 0xffffff800ff84a5c 
0xfffffffe0019f4c0 : 0xffffff800fdd3971 
0xfffffffe0019f4e0 : 0xffffff800fe36f2d 
0xfffffffe0019f5d0 : 0xffffff800fe365d3 
0xfffffffe0019f630 : 0xffffff80105d8bcb 
0xfffffffe0019f720 : 0xffffff800ff84f03 
0xfffffffe0019f8a0 : 0xffffff800ff84be7 
0xfffffffe0019f920 : 0xffffff800fdd3971 
0xfffffffe0019f940 : 0xffffff8012250120 
0xfffffffe0019fa40 : 0xffffff801006761d 
0xfffffffe0019fa70 : 0xffffff8010066d74 
0xfffffffe0019fab0 : 0xffffff80100660d7 
0xfffffffe0019fb60 : 0xffffff80100565b8 
0xfffffffe0019fb90 : 0xffffff8010048974 
0xfffffffe0019fbf0 : 0xffffff801003b35b 
0xfffffffe0019fe00 : 0xffffff80102f1524 
0xfffffffe0019fe30 : 0xffffff8010353fc4 
0xfffffffe0019ff40 : 0xffffff8010452c9d 
0xfffffffe0019ffa0 : 0xffffff800fdd3db6 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOSkywalkFamily(1.0)[1444E3C6-0B2D-300A-BE26-2D1AFDE9E65F]@0xffffff8012241000->0xffffff801226ffff
            dependency: com.apple.driver.mDNSOffloadUserClient(1.0.1b8)[FF5CF7B9-1643-3CA5-A871-DBAF3D1A167B]@0xffffff801214d000->0xffffff8012151fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[5D912FD8-C4CD-3CF6-B214-DF5BF7AB4FA0]@0xffffff8011ef3000->0xffffff8011f0afff

Process name corresponding to current thread (0xffffffa02a0680c8): airportd
Boot args: -v alcid=24 watchdog=0 igfxonln=1 agdpmod=vit9696 -vi2c-force-polling -igfxblr -noDC9 igfxfw=2 -igfxdbeo -wegnoegpu TSC_sync_margin=0 

Mac OS version:
23C71

Kernel version:
Darwin Kernel Version 23.2.0: Wed Nov 15 21:54:10 PST 2023; root:xnu-10002.61.3~2/RELEASE_X86_64
Kernel UUID: F73CFE5A-7EF6-3C21-AFA7-742645B733CC
roots installed: 0
KernelCache slide: 0x000000000fa00000
KernelCache base:  0xffffff800fc00000
Kernel slide:      0x000000000fae0000
Kernel text base:  0xffffff800fce0000
__HIB  text base: 0xffffff800fb00000
System model name: MacBookPro16,2 (Mac-5F9802EFE386AA28)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 1881228204309
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000001b601f05db5
  Sleep   : 0x000000d991c550f6 0x000000004a06f517 0x000000cdeeb780be
  Wake    : 0x000000d99706168f 0x0000000049698b66 0x000000d992582a13
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff9027cba000 - 0xffffffb027cba000
  . PGZ   : 0xffffff9027cba000 - 0xffffff9029cbb000
  . VM    : 0xffffff9029cbb000 - 0xffffff94f64ba000
  . RO    : 0xffffff94f64ba000 - 0xffffff968fcba000
  . GEN0  : 0xffffff968fcba000 - 0xffffff9b5c4ba000
  . GEN1  : 0xffffff9b5c4ba000 - 0xffffffa028cba000
  . GEN2  : 0xffffffa028cba000 - 0xffffffa4f54ba000
  . GEN3  : 0xffffffa4f54ba000 - 0xffffffa9c1cba000
  . DATA  : 0xffffffa9c1cba000 - 0xffffffb027cba000
  Metadata: 0xffffffb07c412000 - 0xffffffb09c412000
  Bitmaps : 0xffffffb09c412000 - 0xffffffb09f412000
  Extra   : 0 - 0
```

I switched to HeliPort for the time being.

zxystd, 2024-01-08:
What's the macOS version? exact version code like `13.4 (22F66)`. Will see when I am avaliable.

shank03, 2024-01-08:
@zxystd Mine is `23C71` and the author version is `23C64`

jmacie, 2024-01-08:
same with AX210 ON 14.3 23D5043d but thanks for all you are accomplishing. works well I I stay away from wifi settingsandthere is no airdrop but that might be bluetooth idk

Krazy-Killa, 2024-01-26:
This is still occurring in macOS Sonoma 14.3.  Opening both the WiFi and Network panes causes a panic after about 3 seconds of opening it.  Can connect to WiFi networks fine via the WiFi menubar, and Bluetooth functions as expected, but cannot make adjustments to network settings except through the terminal.

zxystd, 2024-01-29:
[Sonoma.zip](https://github.com/OpenIntelWireless/itlwm/files/14082111/Sonoma.zip)
Anyone can test this one? And please add `keep_syms=1` to `boot-args` so that we can see the backtrace when the KP occurred.

jmacie, 2024-01-29:
> [Sonoma.zip](https://github.com/OpenIntelWireless/itlwm/files/14082111/Sonoma.zip) Anyone can test this one? And please add `keep_syms=1` to `boot-args` so that we can see the backtrace when the KP occurred.

yes works! can access wifi settings now Thank you!

shank03, 2024-01-30:
> [Sonoma.zip](https://github.com/OpenIntelWireless/itlwm/files/14082111/Sonoma.zip) Anyone can test this one? And please add `keep_syms=1` to `boot-args` so that we can see the backtrace when the KP occurred.

Yup, this works a charm ! Thank you very much !

zxystd, 2024-01-30:
@jmacie @shank03 Thank you, will commit later.

felicia-wen, 2024-01-30:
I can confirm it has fixed the issue, thank you!

closed 2024-01-30 by felicia-wen