OpenIntelWireless/itlwm #949: Mac OS Mavericks and lower kexts request????

megatronskneecap, 2024-01-03:
Have You Read Our Docs:
Yes, I have read the documentation. 

Are You Reporting A Bug:
I mean, sort of...

Environment:
- Kext Version: None, not using a kext yet as can't 
- WiFi Card Model: Intel Centrino Advanced-N 6205S
- PCI Product ID: (Not sure)
- macOS Version: Mavericks

Description:
So, i'm trying to make a vintage hackintosh of sorts using Mavericks. It's on an old 2012 ThinkPad that has the WiFi card model as stated above. The problem is, none of the kexts on this repo support anything under Sierra and High Sierra. I was wondering if someone could compile a Mavericks text for me to use and test. I've also seen other people with older machines looking todo vintage hackintoshes like the Leopard & Lion family of OS X versions encountering the same issue and finding no readily available WiFi kexts up and about so I thought i'd just do everyone the favor and ask for a Sierra and lower version of the kexts and app please! Would be so grateful and hope this is possible. I know it's a strange request but there are people who want to use versions of MacOS that are El Capitan and lower. Thanks, MegatronsKneecap.

nsby494, 2024-01-08:
[itlwm.kext_10.7.zip](https://github.com/OpenIntelWireless/itlwm/files/13856644/itlwm.kext_10.7.zip)
Maybe you can try this driver and write the SSID and password to the plist in advance

usr-sse2, 2024-01-12:
On old laptops it's usually possible to replace the Wi-Fi module with Broadcom or Atheros.

nsby494, 2024-01-14:
> > [itlwm.kext_10.7.zip](https://github.com/OpenIntelWireless/itlwm/files/13856644/itlwm.kext_10.7.zip) Maybe you can try this driver and write the SSID and password to the plist in advance.This driver works very well under OSX 10.7.5
> 
> Ok, this sounds good but how would I put the network details in the config? Could you show me an example config.plist with what you're mentioning please :)

In the info.plist file contained in itlwm.kext, the SSID of the wireless network can be added
[image]

megatronskneecap, 2024-01-14:
Update: Thank you to @nsby494 for this text and the tips. I can confirm it's working beautifully on 10.9 Mavericks. It's actually better as you don't get the ugly WiFi icon on the menubar, it just quietly props itself up as an ethernet (en0) device in the network preferences tab. Thanks for all the help guys. Now closed.

closed 2024-01-14 by megatronskneecap