OpenIntelWireless/itlwm #953: Airportitlwm Broken on macOS 14.4 beta

Lorys89, 2024-01-29:
It broke like in the first betas

Lorys89, 2024-01-29:
API interfaces modified?

zxystd, 2024-01-30:
Close until complete information is added according to the issue template.

Lorys89, 2024-01-30:
> Close until complete information is added according to the issue template.
[Kernel-2024-01-30-094249.panic.zip](https://github.com/OpenIntelWireless/itlwm/files/14095036/Kernel-2024-01-30-094249.panic.zip)

zxystd, 2024-01-30:
@Lorys89 Please add `keep_syms=1` to `boot-args` and trigger again, the panic info lack of symbols.

Lorys89, 2024-01-30:
> @Lorys89 Please add `keep_syms=1` to `boot-args` and trigger again, the panic info lack of symbols.
[Kernel-2024-01-30-100854.panic.zip](https://github.com/OpenIntelWireless/itlwm/files/14095377/Kernel-2024-01-30-100854.panic.zip)

zxystd, 2024-01-30:
sorry I misspell it, it is `keepsyms=1`.

Lorys89, 2024-01-30:
> sorry I misspell it, it is `keepsyms=1`.

[Kernel-2024-01-30-102933.panic.zip](https://github.com/OpenIntelWireless/itlwm/files/14095639/Kernel-2024-01-30-102933.panic.zip)

Lorys89, 2024-01-30:
[Kext From KDK 14.4 beta.zip](https://github.com/OpenIntelWireless/itlwm/files/14095984/Kext.From.KDK.14.4.beta.zip)

zxystd, 2024-01-30:
@Lorys89 Got it, will resolve tomorrow. Sonoma is definitely shit.

Lorys89, 2024-01-30:
> @Lorys89 Got it, will resolve tomorrow. Sonoma is definitely shit.

Ok, Thank you for your commitment, sonoma is really shit, I sent you kext taken from kdk if you need anything else let me know.

Lorys89, 2024-01-31:
@zxystd you have news?

zxystd, 2024-02-01:
@Lorys89 I had a fever yesterday. will see the kexts difference later.

zxystd, 2024-02-01:
[Sonoma_14.4Beta.zip](https://github.com/OpenIntelWireless/itlwm/files/14119732/Sonoma_14.4Beta.zip)

Lorys89, 2024-02-01:
> [Sonoma_14.4Beta.zip](https://github.com/OpenIntelWireless/itlwm/files/14119732/Sonoma_14.4Beta.zip)

Thanks, the kext works on 14.4 😉

zxystd, 2024-02-01:
@Lorys89 
- ax210 is gen3 device which controls rate scaling inside the firmware, different with gen1 devices(the issue #934 author has), so I think you are facing not 'the same' issue.
- From Sonoma, Apple drops `IO80211FamilyLegacy`, I build AirportItlwmV2 on the top of `IO80211Family`, but using some hacks, you can simply interpret it as me implementing a set of IO80211FamilyLegacy myself. This implementation may have side effects such as the iService not working etc. Since `IO80211Family` uses skywalk API instead of original Ethernet API(Also we can foresee that the Ethernet API will also be dropped in macOS 15), without these hacks we should follow the Apple's API and rewrite the whole driver, that's what I would never do.

alighafourian, 2024-02-01:
@zxystd hi and tnx for 14,4 fix 
i have one probelm Airplay work lagy and i repoert this problem before:
https://github.com/OpenIntelWireless/itlwm/issues/923#issuecomment-1770584322

 you say disable MIMO and its make this problem can you make boot-arg for force enable MIMO?

5T33Z0, 2024-02-03:
FYI: this updated 14.4 kext is not backward compatible. It causes a KP in 14.3!

zxystd, 2024-02-03:
Apple drops the compatible, not me. that is why I call them shit.

5T33Z0, 2024-02-03:
Ah, ok. I feel the same way. 14.4 beta 1 killed my whole system. I had to reinstall 14.3 on a new volume from USB flash drive, move files over and then kill the old volume.

alighafourian, 2024-02-05:
https://github.com/OpenIntelWireless/itlwm/actions/runs/7782670974 
last build dosnt work with my 7265 wifi or its release for 14.3 not for 14.4b1?

CloverLeafBG, 2024-02-08:
> https://github.com/OpenIntelWireless/itlwm/actions/runs/7782670974 last build dosnt work with my 7265 wifi or its release for 14.3 not for 14.4b1?

Correct. Last build is released for 14.3. The beta version issued above will not work on anything but 14.4

DmitriyyyyS, 2024-02-22:
@zxystd Hi, could you please update the repository with the latest Airportitlwm changes you made for macOS 14.4 (beta) or post the source code here, thanks!

zxystd, 2024-02-22:
> @zxystd Hi, could you please update the repository with the latest Airportitlwm changes you made for macOS 14.4 (beta) or post the source code here, thanks!

14.4 is still in beta, we don't know if it will change something on later beta versions, when I upload the changes, 14.3 and below version will never support.

ankit0183, 2024-02-24:
can you please give me kext for 14.2.1  wifi i tried but not working anything

alighafourian, 2024-03-07:
@zxystd  14.4 out

zxystd, 2024-03-08:
Already merge ef2e0762d1548b07bcab95570ae44e9bb6470933.

closed 2024-03-08 by zxystd

Beefcat, 2024-05-13:
With iMac20,1 (Gigabyte Z490 Aorus Master) I have AXE300RGB pci card. My system have 2 networks. Onboard works fine (en1). 

I want to have pci work (wifi 6e) but it is not allowing me to toggle en2 on. Do I need to have Securebootmodel enabled ? My device-id is
[image]

Tx.