OpenIntelWireless/itlwm #955: Hope to increase driver support for be200

sunhongzhan, 2024-02-03:
The newly purchased motherboard comes with an Intel Be200 Wi Fi 7 network card.
Hope to receive support soon! Thank you.

zxystd, 2024-02-07:
I don't have this device. 
[9518e29b11b6d93654527c2ce3c838dc6ca441f7](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/9518e29b11b6d93654527c2ce3c838dc6ca441f7)
just added bluetooth support, can you check that?

mishurov, 2024-02-26:
@zxystd on Linux the standard driver can load manually downladed `iwlwifi-gl-c0-fm-c0-83.ucode` and `iwlwifi-gl-c0-fm-c0.pnvm` firmware and it works probably without some new fancy features. For bluetooth it uses `ibt-0291-0291.ddc` and `ibt-0291-0291.sfi`

https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/linux-firmware.git/tree/?h=for-upstream

HQuest, 2024-03-09:
Another BE200 owner here. BT works like a champ, tyvm for adding support. Happy to test and report on any code.

sunhongzhan, 2024-03-10:
> tyvm

“tyvm” is xiami？

Daasin, 2024-03-13:
I'm adding in the Intel BE200 now and will be happy to test anything

jokerknight, 2024-04-16:
Bluetooth is confirmed support but is wifi can also support ?

zxystd, 2024-04-25:
Hi all, progress will update here. https://github.com/OpenIntelWireless/itlwm/issues/955#issuecomment-1931113636

jokerknight, 2024-05-18:
> Bluetooth is confirmed support but is wifi can also support ?

Update: Sometime bluetooth will crash and can't reconnect, util next reboot computer.
description: 
           BlueTooth disconnect automatic(maybe bluetooth crashed), And you can't connect 
it again from system or status bar. util you press your computer reboot button.

Lorys89, 2024-05-18:
[iwlwifi-gl-c0-fm-c0-86.zip](https://github.com/OpenIntelWireless/itlwm/files/15366240/iwlwifi-gl-c0-fm-c0-86.zip)
[iwlwifi-gl-c0-fm-c0.zip](https://github.com/OpenIntelWireless/itlwm/files/15366241/iwlwifi-gl-c0-fm-c0.zip)

zxystd, 2024-05-24:
[image]
Bought a piece of BE200 and the Key E adapter card.

zxystd, 2024-05-27:
Seems that BE200 has a hardware/firmware bug, verified by linux iwlwifi that it can not work in MSI interrupt mode, but macOS only support MSI-X mode on Ventura/Sonoma...

HQuest, 2024-05-27:
There are [a few](https://community.intel.com/t5/Wireless/Intel-BE200-card-drivers-in-linux-distro/m-p/1576423) [reports](https://community.intel.com/t5/Wireless/Help-with-BE200-driver-for-Linux/m-p/1588732) of working WiFi on BE200 cards. Unknow if using interrupt or MSI/MSI-X modes, though. There is also this note saying [linux doesn't support 6GHz](https://github.com/cth1008/Linux-Firmware-for-BE200) but it does work with 2.4/5Ghz.
There is also this [more recent -90 ucode](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/iwlwifi-gl-c0-fm-c0-90.ucode?id=da7133bbdfa9019e261e7c94e70eb4a63f18a975) (and its counterpart [pnvm](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/iwlwifi-gl-c0-fm-c0.pnvm?id=da7133bbdfa9019e261e7c94e70eb4a63f18a975)) perhaps with better chances to our cards.

Thank you for doing your best for the community.

zxystd, 2024-05-27:
@HQuest Thanks for the information, but BE200 is already working on 11ac mode, have to do some stuff to resolve background scan issue later.

vandroiy2013, 2024-05-27:
> Thanks for the information, but BE200 is already working on 11ac mode.
Have hope BE202 will work too)) If you need test it i'm ready.

HQuest, 2024-05-27:
> @HQuest Thanks for the information, but BE200 is already working on 11ac mode, have to do some stuff to resolve background scan issue later.

Thank you! Excited to try this upcoming code/driver. Will keep an eye on the repo commits. Even just 11ac will be much better than no wireless 😄

zxystd, 2024-05-27:
tested with Sonoma 14.3 Recovery since I don't have Ventura+ hackintosh machine and BE200 only support on Ventura/Sonoma.
[image]
[image]

zxystd, 2024-05-27:
[Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/15457871/Ventura.zip)
[Sonoma14.0.zip](https://github.com/OpenIntelWireless/itlwm/files/15457872/Sonoma14.0.zip)
[Sonoma14.4.zip](https://github.com/OpenIntelWireless/itlwm/files/15457874/Sonoma14.4.zip)
Tested reports are welcome.

vandroiy2013, 2024-05-27:
BE202 up and running. So far so good)))

[image]
[image]
[image]

mishurov, 2024-05-27:
BE200 complectovation. Serves as expected.

HQuest, 2024-05-27:
Works as 802.11ac with pure/transitional WPA/WPA2 networks, however does not seem to support pure WPA3 network - is this tied to 11ax?

hla63, 2024-05-28:
> Seems that BE200 has a hardware/firmware bug, verified by linux iwlwifi that it can not work in MSI interrupt mode, but macOS only support MSI-X mode on Ventura/Sonoma...

[https://github.com/acidanthera/MacKernelSDK/pull/22](url)

zhuzhenkun, 2024-05-29:
是否支持双向隔空？

jokerknight, 2024-05-30:
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/15457871/Ventura.zip) [Sonoma14.0.zip](https://github.com/OpenIntelWireless/itlwm/files/15457872/Sonoma14.0.zip) [Sonoma14.4.zip](https://github.com/OpenIntelWireless/itlwm/files/15457874/Sonoma14.4.zip) Tested reports are welcome.

I've tested but it's not working, can't open the network switcher in statusbar and system settings.
and perharps kext not loading ?
 Card Be200  
 OS version 14.4.1
  ❯ kextstat | grep -i "itlwm"
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
❯ kextstat | grep -i "AirportItlwm"
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release

what log should i provider for you ?
And can you provider some startup docs for this special kext ?

HQuest, 2024-05-30:
@jokerknight make sure you are following the [installation guide](https://openintelwireless.github.io/itlwm/Installation.html) - you have to ensure a few items before dropping the kext. I too had to clear my OC config from previous original Apple/Broadcom card before I got this kext to work.

fantasy0808, 2024-05-31:
太牛了 谢谢你钟大 i卡之父 中国骄傲

HQuest, 2024-06-10:
Thank you again for all your work to add BE200 and variants into this kext. I noticed there is a fresh v2.4.0 bump for the main extension code, however I haven't seen any code relating to this device in the commit. Is it still too soon for these changes to make to this git repo?

Malumen, 2024-06-10:
Fingers crossed `BE200` support comes soon!!!

kevon2019, 2024-06-11:
15beta有了吗？测试版2.4

Lorys89, 2024-06-23:
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/15457871/Ventura.zip) [Sonoma14.0.zip](https://github.com/OpenIntelWireless/itlwm/files/15457872/Sonoma14.0.zip) [Sonoma14.4.zip](https://github.com/OpenIntelWireless/itlwm/files/15457874/Sonoma14.4.zip) Tested reports are welcome.

Can you also send itlwm to use with heliport?

Malumen, 2024-06-29:
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/15457871/Ventura.zip) [Sonoma14.0.zip](https://github.com/OpenIntelWireless/itlwm/files/15457872/Sonoma14.0.zip) [Sonoma14.4.zip](https://github.com/OpenIntelWireless/itlwm/files/15457874/Sonoma14.4.zip) Tested reports are welcome.

Can I try with Monterey?

zxystd, 2024-06-29:
> Can I try with Monterey?

No, Monterey only support MSI interrupt mode, so BE200 can not run on it.

Malumen, 2024-06-29:
> No, Monterey only support MSI interrupt mode, so BE200 can not run on it.

So should I upgrade to Ventura or Sonoma? Or wait for full release? (will there ever be support for BE200 on Monterey?)

zxystd, 2024-06-29:
@Malumen This is not the driver issue, BE200 hardware now only support MSI-X interrupt mode, Apple support this mode on Ventura and higher macOS versions, so you should upgrade, or you can wait until Intel official resolve this issue(other products both support MSI and MSI-X).

ephlodur2, 2024-07-11:
Confirm the last version working under sonoma 14.5 - thank you very much - great job!

Daasin, 2024-07-19:
First attempt to use prevents boot on 14.6 beta

going to try resetting nvram with securebootmodel as default later (currently disabled)

Daasin, 2024-07-21:
I assume 6ghz bands are out of the question? 😅

xiaoleGun, 2024-08-09:
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/15457871/Ventura.zip) [Sonoma14.0.zip](https://github.com/OpenIntelWireless/itlwm/files/15457872/Sonoma14.0.zip) [Sonoma14.4.zip](https://github.com/OpenIntelWireless/itlwm/files/15457874/Sonoma14.4.zip) Tested reports are welcome.

Thank you for your work, it works on Sonoma 14.6. Can you push about the change of BE200 to the testing branch for others

dogansan, 2024-10-07:
> Thank you for your work, it works on Sonoma 14.6. Can you push about the change of BE200 to the testing branch for others

I am trying to activate BE200 wifi 7  in Sequoia but I failed. Would you please share your EFI so I can understand what I am missing

dogansan, 2024-10-08:
> BE202 up and running. So far so good)))
> 

Sir, would you please help me , I have a ASUS ROG DARK HERO with B200 wifi 7 , I succeded to run BT , but wifi does not work with itlwm 2.4.0 + heliport on Sequoia . It is clear something is missing in EFI or config.plist. Would you please share your EFI folder

FredWst, 2024-10-21:
Hi,

I've tested BE200 with Sonoma14.4 AirportItlwm.kext on Sonoma 14.6
I can connect to Wifi but routeur doesn't give ip,router and dns.
I put one in Manuel every seems ok but no route to reach internet, not able to ping any address.

itlwm driver doesn't work on Sequoia.

Bluetooth is working with beta 2.5 driver on Sonoma and Sequoia.

Any beta to test ?

Fred

Plutous, 2024-11-12:
> Tested on Sonoma 14.5. Works as 802.11ac with pure/transitional WPA/WPA2 networks, however does not seem to support pure WPA3 network - is this tied to 11ax?

I have the same problem in Sonoma 14.4. computer can connect WPA2 network but not connect WPA3 network, have you solved it.

Gouster4, 2025-01-26:
> [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/15457871/Ventura.zip) [Sonoma14.0.zip](https://github.com/OpenIntelWireless/itlwm/files/15457872/Sonoma14.0.zip) [Sonoma14.4.zip](https://github.com/OpenIntelWireless/itlwm/files/15457874/Sonoma14.4.zip) Tested reports are welcome.

Can you please publish rest releases for Sequoia 15.2? Thank you.

ValeZAA, 2025-04-08:
> Seems that BE200 has a hardware/firmware bug, verified by linux iwlwifi that it can not work in MSI interrupt mode, but macOS only support MSI-X mode on Ventura/Sonoma...

I mean it may be a software bug https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/backport-iwlwifi.git/commit/?id=4988b4853626d96a1114e7596df22304dedeb2d4

grizzleeadam, 2025-04-28:
I am running on Ventura. The beta version in your comment works, sort of - the network scanning is intermittent and only shows a few networks at a time randomly. If mine happens to show up, I can connect and everything works; however, once I reboot, it's a gamble if I can connect again. I'm considering upgrading to Sonoma to try it there.

Lorys89, 2025-05-07:
> No, Monterey only support MSI interrupt mode, so BE200 can not run on it.

Hi, can you itlwm compatible with be200? I need it to use macos sequoia.
Thank you 🙏

Daasin, 2025-05-13:
> > > > [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/15457871/Ventura.zip) [Sonoma14.0.zip](https://github.com/OpenIntelWireless/itlwm/files/15457872/Sonoma14.0.zip) [Sonoma14.4.zip](https://github.com/OpenIntelWireless/itlwm/files/15457874/Sonoma14.4.zip) Tested reports are welcome.
> > > 
> > > 
> > > Can I try with Monterey?
> > 
> > 
> > No, Monterey only support MSI interrupt mode, so BE200 can not run on it.
> 
> Hi, can you send itlwm compatible with be200? I need it to use macos sequoia. Thank you 🙏

I just used the ventura one on sequoia, still works with the spoof and patch method

ValeZAA, 2025-06-02:
>Upgrade iwx supported firmware version to -77 which is the the last non-MLD API aware firmware version. 100%

Weird on Linux BE200 first ucode that supports MLD is 97, not 77

dogansan, 2025-06-20:
> > > > > [Ventura.zip](https://github.com/OpenIntelWireless/itlwm/files/15457871/Ventura.zip) [Sonoma14.0.zip](https://github.com/OpenIntelWireless/itlwm/files/15457872/Sonoma14.0.zip) [Sonoma14.4.zip](https://github.com/OpenIntelWireless/itlwm/files/15457874/Sonoma14.4.zip) Tested reports are welcome.
> > > > 
> > > > 
> > > > Can I try with Monterey?
> > > 
> > > 
> > > No, Monterey only support MSI interrupt mode, so BE200 can not run on it.
> > 
> > 
> > Hi, can you send itlwm compatible with be200? I need it to use macos sequoia. Thank you 🙏
> 
> I just used the ventura one here on sequoia with BE200, still works with the spoof and patch method

Can you please g,ve the details on how you have done spoof and patch ?

Brandonmccoub2, 2025-07-13:
@Daasin Are you able to give details on how you were able to make this work on Sequoia?

mishurov, 2025-07-13:
@Daasin  does spoof and patch work in Tahoe?

kgp-macPro, 2026-10-30:
## Intel BE200 working on macOS Tahoe — AirportItlwm-Tahoe 1.2.0 BE200 Alpha 1

I have some news I have been looking forward to sharing: **my Intel BE200 is now providing working Wi-Fi under macOS Tahoe with AppleVTD enabled.**

On the latest tested build, the machine booted directly into a connected Wi-Fi session. DHCP worked, Internet access worked, and after putting the system to sleep, Wi-Fi returned immediately on wake.

**Four consecutive Wi-Fi OFF/ON cycles have now passed as well, each with automatic reconnection.**

After the succession of builds and hardware tests that brought us here, that was a particularly good moment.

The first **BE200 alpha and its matching source code** are now available for testing.

### Required root patching

**This alpha requires Retro-OCLP-CustoMac root patching and the restored Ventura legacy wireless frameworks.**

Root-patching project:
https://github.com/kgp-macPro/OCLP-CustoMac

### Download and source code

- **[Kext download — AirportItlwm-Tahoe-1.2.0-BE200-Alpha-1.zip](https://github.com/kgp-macPro/AirportItlwm-Tahoe/releases/download/v1.2.0-be200-alpha.1/AirportItlwm-Tahoe-1.2.0-BE200-Alpha-1.zip)**
- **[Alpha release notes and installation information](https://github.com/kgp-macPro/AirportItlwm-Tahoe/releases/tag/v1.2.0-be200-alpha.1)**
- **[Matching source code](https://github.com/kgp-macPro/AirportItlwm-Tahoe/tree/v1.2.0-be200-alpha.1)**

### Initial hardware results

Initial hardware testing on my **ASUS X299 system under macOS Tahoe 26.7.2 (25G332)** has confirmed:

- Intel BE200 **8086:272B** detection and AirportItlwm binding.
- Network discovery and association using the macOS Wi-Fi interface.
- Automatic connection immediately after login.
- A valid DHCP configuration and working Internet access.
- **Four successful Wi-Fi OFF/ON cycles with Auto-Join.**
- **Immediate Wi-Fi recovery after a manual system Sleep/Wake cycle.**
- AppleVTD enabled during testing.

The tested connection uses **5 GHz / 80 MHz / 802.11ac**. The attached Internet speed test shows **79.5 Mbps download and 28.5 Mbps upload**.

**My Internet connection is limited to 80 Mbps download and 30 Mbps upload**, so this result comes close to its available capacity. Local Wi-Fi throughput will be measured separately using a wired LAN server.

**Further performance optimization will follow**, alongside controlled local-network measurements and broader recovery testing.

### Bluetooth configuration

**Bluetooth is also working during these tests**, with both Magic Keyboard and Magic Mouse connected. The Bluetooth configuration used is:

- [**BlueToolFixup.kext v2.7.2**](https://github.com/acidanthera/BrcmPatchRAM/releases) by **acidanthera**.
- [**IntelBluetoothFirmware v2.5.1**](https://github.com/lshbluesky/IntelBluetoothFirmware/releases/tag/v2.5.1) from **lshbluesky**.
- Intel USB Bluetooth controller **8087:0036**.

### Alpha scope and ongoing development

**This is an experimental alpha.** The initial results cover BE200 in the tested 802.11ac configuration. Long-term stability, broader hardware coverage and additional wireless features require further testing. Please keep a working EFI backup and follow the release’s installation and rollback instructions.

The **ZC28-based 1.1.0 beta remains available and its testing continues**. This BE200 alpha is an additional development track; **1.0.0 remains the stable release**.

### Credits

Many thanks to **zxystd/OpenIntelWireless, DexterSLamb, HuajiFruit/AirportBZ** and the other upstream contributors. Their work provides the foundation for this development. Thanks also to **acidanthera and lshbluesky** for the Bluetooth components used here.

I have directed and physically tested the changes, working with ChatGPT and Codex throughout the investigation and implementation.

There is more work ahead, especially on recovery and performance. Tonight, though, I’m enjoying having the BE200 connection carry the traffic while I write this update.

### Screenshots

The ten screenshots below document the hardware, AppleVTD, Wi-Fi connection, DHCP configuration, firmware, Internet speed test, Bluetooth peripherals and macOS test environment.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

kgp-macPro, 2026-10-30:
### AirportItlwm-Tahoe 1.2.0 BE200 Alpha 2 — performance hotfix

Alpha 2 substantially improves BE200 throughput. My complete 3×14 local benchmark on macOS Tahoe 26.7.2 with AppleVTD enabled produced the following three-round averages:

| Test | Alpha 1 | Alpha 2 |
|---|---:|---:|
| TCP TX P1 | 28.6 Mbit/s | 481.2 Mbit/s |
| TCP TX P4 | 28.5 Mbit/s | 479.4 Mbit/s |
| TCP RX P1 | 195.4 Mbit/s | 481.7 Mbit/s |
| TCP RX P4 | 213.1 Mbit/s | 472.4 Mbit/s |

UDP RX at 600 Mbit/s achieved approximately the requested rate in all three rounds, with 0.08–0.20% reported loss.

Wi-Fi OFF/ON and Sleep/Wake also passed my manual tests. Six repeated Ookla runs produced almost identical results near my 80/30 Mbit/s Internet line limit.

Retro-OCLP-CustoMac root patching remains required. Remote Find My visibility remains unresolved on my system. Further RX investigation and AX201/AX210/AX211 testing continue.

**[Releases and downloads](https://github.com/kgp-macPro/AirportItlwm-Tahoe/releases)**

kgp-macPro, 2026-11-02:
Some encouraging news for BE200 users!

@gsfront has submitted a contribution adding optional Wi-Fi 6 (802.11ax / HE) support to the BE200, building on our published Alpha 2 / P8R1 foundation.

His PR also includes larger receive-reordering windows, optional beamforming support, improved handling of firmware receive notifications, and additional diagnostics. He reports successful HE operation with link rates up to 2400 Mbps on his 160 MHz setup, together with improved download performance. These are initial results from his BE200/AP combination; broader validation is still pending.

A big thank you to @gsfront for contributing both the implementation and the testing effort! He has also offered to run our shared iperf3 campaign against the unchanged Alpha-2 reference.

There is potentially useful groundwork here for AX201, AX210 and AX211 as well. Several improvements affect shared code, while the BE200-specific firmware changes require separate assessment and adaptation for the AX families. This PR therefore does not yet establish working Wi-Fi 6 support on those cards. We will check their existing functionality for regressions, and the contribution could provide a basis for a subsequent AX-focused HE extension.

Our Codex-assisted development workflow is currently interrupted by an unresolved Codex/Daybreak access issue, which has been escalated to OpenAI Support. Once access is restored, we plan to review the PR and comparative results, discuss any necessary adjustments with gsfront, and prepare a separate experimental candidate.
My current access points support Wi-Fi 5, so HE-specific validation will depend on contributors with suitable equipment.

There is no release date yet, and the published Alpha-2 reference remains unchanged. Wi-Fi 7, MLO and 6 GHz are outside this PR’s scope. Our native AirDrop work continues separately.

[Discussion and updates — Issue #4](https://github.com/kgp-macPro/AirportItlwm-Tahoe/issues/4)
[Implementation and initial results — PR #5](https://github.com/kgp-macPro/AirportItlwm-Tahoe/pull/5)

One correction to my previous Alpha-2 update: Find My visibility is now working on my BE200 setup after enabling “Find My Mac” in iCloud settings. No driver change was required. Offline finding remains untested.

Adpocalyptic, 2026-11-02:
> Some encouraging news for BE200 users!
>
> @gsfront has submitted a contribution adding optional Wi-Fi 6 (802.11ax / HE) support to the BE200, building on our published Alpha 2 / P8R1 foundation.
>
> His [...]

@gsfront @kgp-macPro So... There are forks that work? Hopefully one singular one that everyone can get behind. Fragmentation makes it hard to figure out what's happening. I can try test different equipment. Got three different WiFi 6 routers and can swap WiFi cards in my OpenCore Tahoe setup with a second laptop setup, and the patching route is... Painful 😅 

I know some feature related stuff was turned off with the hope of getting stability but hopefully can be brought back even if borked. Like if people want stability they can go with Ethernet (or Airport) but having the 'native' connection open doors is what drew me to the Intel kext.

kgp-macPro, 2026-11-02:
@Adpocalyptic ! Thanks for offering to help! Testing across three different Wi-Fi 6 routers and several Intel cards would be particularly useful.

I understand the concern about fragmentation. @gsfront has submitted his changes directly to AirportItlwm-Tahoe through a pull request, so we can review, test and integrate the contribution within the same project.

For BE200, our published [Alpha 2 / P8R1](https://github.com/kgp-macPro/AirportItlwm-Tahoe/releases/tag/v1.2.0-be200-alpha.2) already provides working Wi-Fi in the tested 802.11ac configuration, including local throughput measurements and successful manual reconnection and sleep/wake tests. It remains an experimental alpha. The additional HE support is currently in [PR #5](https://github.com/kgp-macPro/AirportItlwm-Tahoe/pull/5), which is still open and awaiting review and broader validation.

Regarding patching: our current Alpha 2 still requires Retro-OCLP-CustoMac and the restored legacy wireless frameworks. Native macOS Wi-Fi integration does not currently mean a patch-free installation.

I also share your interest in the Apple integration features. Optional experimental features are welcome—that is one reason gsfront’s opt-in approach is useful. However, some missing functionality needs substantial implementation work. Native AirDrop, for example, still lacks a complete working radio/data path; it cannot currently be restored simply by re-enabling a switch.

Could you post your router models and firmware versions, available Intel cards, Tahoe versions and current patching setup in [Issue #4](https://github.com/kgp-macPro/AirportItlwm-Tahoe/issues/4)? We can then agree on a focused comparison between the unchanged Alpha-2 reference and a reviewed HE candidate, keeping the framework and patching environment consistent.

One update to my earlier post: I have now ordered an HE160-capable access point with 2.5-Gbit Ethernet, expected on Wednesday, so I will also be able to test HE80 and HE160 locally once the candidate is ready. This is an interim setup while I wait for the FRITZ!Repeater 6700 Pro to become available for my planned network upgrade.

My DSL Internet connection will remain limited to 80 Mbps downstream and 30 Mbps upstream until my planned upgrade to 1-Gbit/s fiber in Q1 2027, using a FRITZ!Box 4690 with an external fiber modem. This limits Internet speed tests, but does not constrain our local iperf3 measurements between the Wi-Fi client and a suitably connected wired server.

Your additional routers would give us valuable coverage beyond our own setups. Thanks again—this is exactly the kind of testing that can help move the contribution forward!

gsfront, 2026-11-02:
Thanks @kgp-macPro for the shout-out, and for Alpha 2, which all of this builds on! 🙏

A short update. I've now run the shared 3×14 iperf3 campaign (full results and ZIPs in [PR #5](https://github.com/kgp-macPro/AirportItlwm-Tahoe/pull/5)). The controlled numbers refine my initial report:

| TCP RX (avg, Mbps) | Alpha 2 · 11ac 80 MHz | HE · 11ax 80 MHz | HE · 11ax 160 MHz |
|---|---|---|---|
| 1 stream | 595 | 596 | **786** |
| 4 streams | 571 | 556 | **786** |

- At **80 MHz**, HE alone is about the same as Alpha 2. The gain comes from **160 MHz**.
- ⚠️ **Tip for Alpha 2 users with a 160 MHz router:** in my setup, Alpha 2 associates as 11ac VHT160, but **RX drops to about 1 Mbps** (TX still works). Setting the router's 5 GHz width to **80 MHz** fixed it immediately. The HE build at 160 MHz is not affected.

These results are from one BE200 and one AP only. Further discussion is in [Issue #4](https://github.com/kgp-macPro/AirportItlwm-Tahoe/issues/4) / PR #5.

gsfront, 2026-11-02:
⚠️ **Correction to my previous comment.** The "set your router to 80 MHz" tip for Alpha 2 users was **wrong**, so please ignore it.

I've since run Alpha 2 at **160 MHz on a fresh boot**, and the full iperf3 campaign completed normally (TCP RX ~780–850 Mbps). What I had seen was a **one-time RX stall** (RX ~1 Mbps, with most received frames dropped as duplicates by the driver). A **reassociation** cleared it: changing the router's width only fixed it because it forced the Wi-Fi to reconnect. If you ever see Wi-Fi RX suddenly drop to ~1 Mbps while upload still works, toggling Wi-Fi off/on should clear it. Please also report it in [Issue #4](https://github.com/kgp-macPro/AirportItlwm-Tahoe/issues/4).

The complete comparison (5 campaigns) also shows that **Alpha 2 and the HE build perform the same** on my AP at 80 and at 160 MHz. The speed gain I first reported came from 160 MHz, not from HE. Full details and result ZIPs are in [PR #5](https://github.com/kgp-macPro/AirportItlwm-Tahoe/pull/5). Sorry for the confusion! 🙏

kgp-macPro, 2026-11-06:
[image]

## AirportItlwm-Tahoe 2.0.0 — BE200 VHT, HE and optional HE+BF with AppleVTD/IOMMU support

Some substantial progress since the earlier BE200 Alpha updates: **AirportItlwm-Tahoe 2.0.0 is now available.**

The release incorporates **gsfront’s BE200 HE/HE+BF contribution**, alongside our subsequent integration, fixes and hardware validation.

Complete BE200 campaigns are available for **VHT, HE and HE with BF opt-in**. In the extended three-round campaigns, receiver-side TCP RX averages were:

- **HE:** approximately **1.68 Gbit/s P1 / 1.86 Gbit/s P4**.
- **HE with BF opt-in:** approximately **1.70 Gbit/s P1 / 1.88 Gbit/s P4**.

Each campaign completed all **60 local tests** and the separate online test. High-rate UDP loss and TCP retransmissions remain documented; completion does not imply loss-free operation.

The validation setup used **AppleVTD enabled with the mapped DMA route verified**, a 160-MHz connection, and a wired server limited to **2500Base-T for HE/HE+BF**.

**OCLP-CustoMac 3.0.0 or newer with the applicable Ventura Legacy wireless root patches is required.**

- **[[Release and downloads](https://github.com/kgp-macPro/AirportItlwm-Tahoe/releases/tag/v2.0.0)](https://github.com/kgp-macPro/AirportItlwm-Tahoe/releases/tag/v2.0.0)**
- **[[Results, screenshots, device coverage and source](https://github.com/kgp-macPro/AirportItlwm-Tahoe)](https://github.com/kgp-macPro/AirportItlwm-Tahoe)**
- **[[Benchmark guide and complete tools](https://github.com/kgp-macPro/AirportItlwm-Tahoe/blob/main/docs/benchmark/README.md)](https://github.com/kgp-macPro/AirportItlwm-Tahoe/blob/main/docs/benchmark/README.md)**
- **[[OCLP-CustoMac](https://github.com/kgp-macPro/OCLP-CustoMac/releases)](https://github.com/kgp-macPro/OCLP-CustoMac/releases)**

BF refers to the optional capability being enabled; actual over-the-air beamforming requires separate evidence. **Wi-Fi 7/EHT remains future work and is not included in 2.0.0.**

Many thanks to **OpenIntelWireless/zxystd**, **gsfront**, **laobamac**, and everyone who contributed code, testing and feedback. Full attribution and build provenance are included in the project documentation.

kgp-macPro, 2026-11-12:
Another BE200 progress update: **successful bidirectional AirDrop transfers with a MacBook Pro M1 and an iPhone 18 Pro are now confirmed in my experimental AirportItlwm-Tahoe setup!** 🎉

Files have arrived in **all four directions**: Hackintosh → MacBook Pro, MacBook Pro → Hackintosh, Hackintosh → iPhone, and iPhone → Hackintosh.

The attached screenshots show the Intel BE200 hardware identification, native **“Sent”** confirmations on both Macs, and **“Mac Pro — Gesendet”** on the iPhone.

The tested combination is **AirportItlwm AWDL R25 + experimental BLE-R25-01**, using the **IO80211FamilyLegacy / OCLP** integration. The devices are running **macOS Tahoe 26.7.2 (25G332)** on the Hackintosh, **macOS 27.2.1 (26B131)** on the MacBook Pro M1, and **iOS 27** on the iPhone.

**This is still experimental.** Initial attempts can fail or be delayed, so this is a working development milestone rather than a stable AirDrop release.

Thank you to the OpenIntelWireless developers for the BE200 foundations, and to acidanthera, Bananz0 and ForbiddenItlwm for the additional work and references.

Project: https://github.com/kgp-macPro/AirportItlwm-Tahoe

[image]

[image]

[image]

kgp-macPro, 2026-11-12:
Another **Intel BE200** milestone: **Personal Hotspot connection to my iPhone 18 Pro now works, including Internet access!** 🎉

Successfully tested with **AirportItlwm-Tahoe experimental R27 + BLE-R25-01** on **macOS Tahoe 26.7.2 (25G332)**, using the **IO80211FamilyLegacy / OCLP** integration.

**This build adapts mingww64’s WPA compatibility changes, merged by laobamac through PR #36:**

https://github.com/laobamac/itlwm/commit/7204c69d1bfa7f5cc0f7864e08fb03e55d0ec29e

The reference fix improves WPA2/WPA3 transition-mode authentication selection, including PSK fallback and management-frame-protection-aware SHA256-PSK preference. We adapted the relevant behavior to our Legacy path.

The screenshots show the native iPhone hotspot connection and working web access, with both Ethernet interfaces disconnected or inactive.

Many thanks to **mingww64, laobamac and pico joe**, and to the **OpenIntelWireless developers and acidanthera** for the foundations.

Following the successful bidirectional AirDrop transfers reported earlier, it is great to see another BE200 feature working! 🙂

Project: https://github.com/kgp-macPro/AirportItlwm-Tahoe

[image]

[image]

[image]

[image]