OpenIntelWireless/itlwm #957: Kernel Extensions in backtrace

starlifezzz, 2024-02-25:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: 2.3
- WiFi Card Model: Ax210
- PCI Product ID: pci8086,2725
- macOS Version: 13.6.4

**Description**

  开机进不了系统， 最初是用2.2reasle本发现，后更为2.3还是会发生
**Bug Report Archive**

panic(cpu 2 caller 0xffffff80151e0625): zfree_drop: double free of 0xffffff8b796d4b80 to zone kalloc.type.var3.32 @zalloc.c:815
Panicked task 0xffffff90456c29d8: 212 threads: pid 0: kernel_task
Backtrace (CPU 2), panicked thread: 0xffffff9511850b30, Frame : Return Address
0xffffffc8d5dc7790 : 0xffffff8014a6fc7d 
0xffffffc8d5dc77e0 : 0xffffff8014bc3f74 
0xffffffc8d5dc7820 : 0xffffff8014bb3a87 
0xffffffc8d5dc7870 : 0xffffff8014a10971 
0xffffffc8d5dc7890 : 0xffffff8014a6ff5d 
0xffffffc8d5dc7980 : 0xffffff8014a6f607 
0xffffffc8d5dc79e0 : 0xffffff80151daa9b 
0xffffffc8d5dc7ad0 : 0xffffff80151e0625 
0xffffffc8d5dc7b00 : 0xffffff8014ae0423 
0xffffffc8d5dc7b30 : 0xffffff80150cfff0 
0xffffffc8d5dc7b50 : 0xffffff801936fe64 
0xffffffc8d5dc7b80 : 0xffffff80193788fd 
0xffffffc8d5dc7c90 : 0xffffff801936fe01 
0xffffffc8d5dc7d40 : 0xffffff801938319d 
0xffffffc8d5dc7d80 : 0xffffff8019388f6b 
0xffffffc8d5dc7dd0 : 0xffffff801938a68a 
0xffffffc8d5dc7e40 : 0xffffff801931ebf7 
0xffffffc8d5dc7fa0 : 0xffffff8014a1019e 
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(2.3)[CA4117E5-A162-3A8C-9FE3-AE92B531862F]@0xffffff8019312000->0xffffff801a2a5fff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[380F5347-094D-36D8-A11C-E6AE92F49815]@0xffffff80166f3000->0xffffff8016839fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[9464DB11-7BAB-372E-BE42-312E2C708040]@0xffffff8017253000->0xffffff8017269fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[654A8CD8-6853-3ED2-B7F1-E00515127C3D]@0xffffff80174b4000->0xffffff80174e5fff

Process name corresponding to current thread (0xffffff9511850b30): kernel_task

Mac OS version:
22G513

Kernel version:
Darwin Kernel Version 22.6.0: Sun Dec 17 22:18:09 PST 2023; root:xnu-8796.141.3.703.2~2/RELEASE_X86_64
Kernel UUID: E842293E-6652-3980-AA66-4C1D156AB7AB
roots installed: 0
KernelCache slide: 0x0000000014600000
KernelCache base:  0xffffff8014800000
Kernel slide:      0x00000000146dc000
Kernel text base:  0xffffff80148dc000
__HIB  text base: 0xffffff8014700000
System model name: iMacPro1,1 
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 139450237194
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000002077e1c3f8
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000ea2c9a108 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff804371d000 - 0xffffffa04371d000
  . PGZ   : 0xffffff804371d000 - 0xffffff804771e000
  . VM    : 0xffffff804771e000 - 0xffffff8513a51000
  . RO    : 0xffffff8513a51000 - 0xffffff86ad0b7000
  . GEN0  : 0xffffff86ad0b7000 - 0xffffff8b793ea000
  . GEN1  : 0xffffff8b793ea000 - 0xffffff904571d000
  . GEN2  : 0xffffff904571d000 - 0xffffff9511a50000
  . GEN3  : 0xffffff9511a50000 - 0xffffff99ddd83000
  . DATA  : 0xffffff99ddd83000 - 0xffffffa04371d000
  Metadata: 0xffffffffb9fee000 - 0xffffffffd9fee000
  Bitmaps : 0xffffffffd9fee000 - 0xffffffffdffee000
  Extra   : 0 - 0

**Kext Download Source**

closed 2024-02-25 by starlifezzz

woodlai, 2024-10-13:
怎么解决的

starlifezzz, 2024-10-14:
> 怎么解决的

没有解决，反正更新了最新版