OpenIntelWireless/itlwm #958: Airportitlwm not works on macOS 14.2.1

blairBai, 2024-02-29:
**Have You Read Our Docs**
yes

**Are You Reporting A Bug**
yes

**Environment**

- Kext Version: AirportItlwm.kext 2.3.0
- WiFi Card Model: Intel AX210 NGW
- PCI Product ID: 
- macOS Version: Sonoma 14.2.1

**Description**

安装Sonoma14.2.1或14.3.1，使用网卡AX210以及最新的AirportItlwm版本无法驱动wifi,蓝牙可以使用。已经按照文档进行配置，reset nvram重启后也不行。

**Bug Report Archive**

[bugreport_37034.zip](https://github.com/OpenIntelWireless/itlwm/files/14445649/bugreport_37034.zip)

**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.3.0-alpha

zxystd, 2024-02-29:
```
[   13.971579]: itlwm: iwx_enable_fwload_interrupt
[   14.176770]: itlwm: : acquiring device failed
[   14.176774]: itlwm: : could not init context info
[   14.176775]: itlwm: : failed to load init firmware
```
PCI状态不对，不是itlwm驱动的问题，你换什么网卡都没用的，你的EFI有问题。
自行研究解决办法。

closed 2024-02-29 by zxystd

blairBai, 2024-02-29:
感谢

zxystd, 2024-02-29:
https://bbs.pcbeta.com/viewthread-1993298-1-1.html
https://bbs.pcbeta.com/viewthread-1987238-1-7.html
这里有几个人跟你一样的问题，都是AMD的机器，貌似都用的别人的EFI，我的建议是尽可能少地用补丁跟驱动，看看能不能把Wi-Fi驱动起来，确定是哪个东西导致的。我也是有些好奇的。

blairBai, 2024-02-29:
> https://bbs.pcbeta.com/viewthread-1993298-1-1.html https://bbs.pcbeta.com/viewthread-1987238-1-7.html 这里有几个人跟你一样的问题，都是AMD的机器，貌似都用的别人的EFI，我的建议是尽可能少地用补丁跟驱动，看看能不能把Wi-Fi驱动起来，确定是哪个东西导致的。我也是有些好奇的。

如你所说，我用的也是这个设备和EFI，当时图方便找了这个，看着价格不错又适配好了。唉，实在不行就装回13随便用用得了。感谢感谢

zxystd, 2024-02-29:
> 
> 如你所说，我用的也是这个设备和EFI，当时图方便找了这个，看着价格不错又适配好了。唉，实在不行就装回13随便用用得了。感谢感谢

EFI有问题，装13应该也是一样的。什么机器？用的哪个EFI？

zxystd, 2024-02-29:
1. 把注入的DeviceProperties全部删掉
2. CaseySJ | probeBusGated | Disable 10 bit tags | 12.0+ 这个补丁先禁用
3. 暂时禁用NootedRed、SMCRadeonGPU、RadeonSensor这些驱动
4. ACPI能不注入的就不注入，能开机就行
只要做到能开机进系统看看能不能驱动Wi-Fi就可以。

还不行的话，Q群 1132224334 沟通

blairBai, 2024-02-29:
好的，我去试试，十分感谢

blairBai, 2024-02-29:
https://github.com/daliansky/Beelink-SER5-Hackintosh

零刻 SER 5 PRO  
cpu: ADM R5 5600H

[EFI.zip](https://github.com/OpenIntelWireless/itlwm/files/14446552/EFI.zip)

zxystd, 2024-03-01:
[image]

@blairBai 问过兵哥本人了，这是这款主机的通病，多系统切换会导致某些设备初始化失败。你可以按照图上的进行操作，应该是可以用了，以后用单系统就好了。

blairBai, 2024-03-01:
太感谢了，昨天我把那些注入驱动补丁去掉，还是没驱动，原来如此。

zxystd, 2024-03-01:
更深层次的原因

[image]

[image]

[image]
[image]
宪武大大yyds

blairBai, 2024-03-04:
Emmm... 还是没有驱动，清空了cmos，硬盘中只有一个mac系统。
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/14477361/itlwm_logs.log)
换了AX200也试了一下，错误信息没变。
终端：
```
j-mcgill@J-McGilldeiMac ~ % kextstat | grep "Itlwm"
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   99    0 0xffffff8004950000 0xfab000   0xfab000   com.zxystd.AirportItlwm (2.3.0) FFAC42BC-F325-3492-B025-DF773FEB18C2 <98 97 56 54 18 7 6 3 1>
j-mcgill@J-McGilldeiMac ~ % 
  [恢复于2021年1月1日 08:02:00]
Last login: Fri Jan  1 08:01:58 on console
Restored session: 2021年 1月 1日 星期五 08时54分42秒 CST
j-mcgill@J-McGilldeiMac ~ % kextstat | grep "Itlwm"
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   97    0 0xffffff8004950000 0xfab000   0xfab000   com.zxystd.AirportItlwm (2.3.0) FFAC42BC-F325-3492-B025-DF773FEB18C2 <96 95 56 54 18 7 6 3 1>
```

blairBai, 2024-03-04:
看了一些相近的配置的小主机，amd r5 5600h, intel AX200\210 的网卡，大部分都能驱动wifi,包括零刻 ser5 MAX也没问题。能搜索到的基本都是ser5 PRO这个机型的有这个问题。

htmambo, 2024-03-05:
我是morefine s500，换网卡、换系统都折腾过了，目前基本上是放弃状态了，反正台式机有线一样用。

zxystd, 2024-03-12:
@blairBai 跟你楼下的群友在群里研究了两天了，基本确认了问题，确实是PCIE设备的问题，设备硬件本身就报错了，不知道是BIOS的原因还是什么

blairBai, 2024-03-12:
> @blairBai 跟你楼下的群友在群里研究了两天了，基本确认了问题，确实是PCIE设备的问题，设备硬件本身就报错了，不知道是BIOS的原因还是什么

哇，辛苦了，那看来这个问题是有些棘手了，之前也想着调整一下bios，但是可能没抓住重点，没有效果。

zxystd, 2024-03-12:
@blairBai 看看BIOS里有没有SERR Generation相关的选项

zxystd, 2024-03-12:
应该是PCI设置，一个个都点进去看看，类似下面这样的
[image]

zxystd, 2024-03-12:
进Q群沟通吧，在这里会影响别人。

blairBai, 2024-03-12:
[image]

好的，我已经发送申请了