OpenIntelWireless/itlwm #959: location now work in 14.4 Beta (23E5211a)

alighafourian, 2024-02-29:
in 14.4 Beta (23E5211a) loacation not work i try tern off and on it but nothing change wont work :

**Have You Read Our Docs**
Yes 

**Are You Reporting A Bug**
Yes

Loges
without DebugEnhancer.kext
[Log_2024-02-29_19-48-09.log](https://github.com/OpenIntelWireless/itlwm/files/14450729/Log_2024-02-29_19-48-09.log)

**Environment**

- Kext Version: https://github.com/OpenIntelWireless/itlwm/issues/953#issuecomment-1920413379
- WiFi Card Model: 7265n
- PCI Product ID: 0x8086, 0x5002
- macOS Version: 14.4 Beta (23E5211a)

**Description**
[image]

[image]

alighafourian, 2024-02-29:
with DebugEnhancer.kext

[Log_2024-02-29_19-52-09.log](https://github.com/OpenIntelWireless/itlwm/files/14450791/Log_2024-02-29_19-52-09.log)

zxystd, 2024-03-01:
Location service is confirmed working.

closed 2024-03-01 by zxystd