OpenIntelWireless/itlwm #961: Ventura Killer ax1650i can click on the icon select wifi，but the link will never be successful

gzliudianxin, 2024-03-01:
Have You Read Our Docs
Yes

Are You Reporting A Bug
Yes

Environment

Kext Version: AirportItlwm.kext.v2.2.0
WiFi Card Model: Killer ax1650i
PCI Product ID: DELL xps 9500
macOS Version: Ventura 13.6 
Description
I have Dell xps 9500 Running Macos Ventura  with Intel  ax1650i
I'm using AirportItlwm.kext - IntelBluetoothFirmware.kext - BlueToolFixup.kext with the wifi is s unstable Keeps disconnecting.
The wifi icon shows on the top bar, input right password but wifi connection always  keep loading . Then back to the login page，never success.
[image]

Log
[bugreport_37562.zip](https://github.com/OpenIntelWireless/itlwm/files/14463850/bugreport_37562.zip)

zxystd, 2024-03-01:
Can you please try Heliport+itlwm.kext to see if it will work?
Anyway, can you tell me what's the wifi encrypt algorithm? WPA2 only or WPA2/WPA3 mixed?

gzliudianxin, 2024-03-02:
Thanks for you reply.  
My WiFi encrypt model is WPA2 only and I try to connect to my iPhone hotspot , is the same problem can select the hotspot but can't connect。
that's my WiFi settings
[image]

gzliudianxin, 2024-03-02:
I try to use Heliport+itlwm.kext and it works ,connect to the wifi quickly.
Anyone else face the same problem?
[image]

gzliudianxin, 2024-03-03:
But I try another version AirportItlwm-Ventura-v2.3.0-DEBUG-alpha-618eb94.zip，And it doesn`t work...  
So I make the kext back to Heliport+itlwm.kext 
And after that,Even by the Heliport way  I can`t connect to the WIFI!!! ，say the wrong password again. 
Orz，    I'm gonna break down! 
[image]

and that`s the log
[bugreport_5471.zip](https://github.com/OpenIntelWireless/itlwm/files/14473481/bugreport_5471.zip)

zxystd, 2024-03-03:
Q群 1132224334 沟通