OpenIntelWireless/itlwm #962: Intel wireless card 7265NGW and 8265NGW very slow after running macOS for a short period

RIPDucky24, 2024-03-04:
**Have You Read Our Docs**
Yes
**Are You Reporting A Bug**
Most likely

**Environment**

- Kext Version: same problem on 2.2.0-stable and 2.3.0-alpha
- WiFi Card Model: 8265NGW and same problem on 7265NGW
- PCI Product ID: 0x8086 (Let know if this isnt it)
- macOS Version: Same issue on latest Mac OS 11,12,13 and 14

**Description**
On Mac OS both cards listed are unusable on itlwm and airportitlwm they will work for a short period of time and then after a few minutes the download speed lowers to average 85KBS which most websites or files will not  download/render with that conditioning I have tried all 3 boot args to slight avail

**Bug Report Archive**
Startup Mac OS, wait 5-10 minutes, issue starts, restart to temporarily fix

**Kext Download Source**
This Github Page aka Github releases

zxystd, 2024-03-07:
It's quite inexplicable, but it looks like a compatibility issue with a router. I suggest trying a different router. Leave it open for days temporary to see if anyone have the same problem.

Scharkenberg, 2024-03-07:
> It's quite inexplicable, but it looks like a compatibility issue with a router. I suggest trying a different router. Leave it open for days temporary to see if anyone have the same problem.

I experience this with airportitlwm (always bad on 5 GHz ac, not as bad on 2.4 GHz n), but not with itlwm + heliport.

- routers: several models from different manufacturers
- card: Intel 8265
- OS: 11, 12, 13, 14 (currently on 14.4)

zxystd, 2024-03-09:
@Scharkenberg For your case, was it normal on v2.2.0 stable version?

zxystd, 2024-03-09:
@RIPDucky24 
[image]
Tested with 7260ac about 26 minutes，very smooth speed.

RIPDucky24, 2024-03-09:
Ok yeah then it's probably my router then, I'll go ahead and speak to my isp to get me a replacement. Thanks!

closed 2024-03-09 by RIPDucky24

Scharkenberg, 2024-03-11:
> @Scharkenberg For your case, was it normal on v2.2.0 stable version?

I have been using 2.3.0-alpha for the past year. AirportItlwm 2.2.0 doesn't have a Sonoma release, so I could not test that combination.

I ended up reinstalling Ventura, where 2.2.0-stable works more or less as expected (connecting to known networks after reboot/sleep is slow, otherwise speeds are mostly stable and there are no sudden disconnects). Should I plan on staying on Ventura for the time being? Or can 2.2.0-stable also be compiled for Sonoma perhaps?