OpenIntelWireless/itlwm #963: Airport itwlm issue

karizr45, 2024-03-05:
Bug report

**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: --Latest airportitlwm for ventura
- WiFi Card Model: Intel AX200
- PCI Product ID: 0x2723
- macOS Version: MacOS 14.3.1 Sonoma

**Description**

I've got a problem with airportitlwm not seeing any networks but heliPort does
i've tried to use beta sonoma version but macOS wont boot

**Bug Report Archive**

[Log_2024-03-05_21-53-17.log.zip](https://github.com/OpenIntelWireless/itlwm/files/14498805/Log_2024-03-05_21-53-17.log.zip)

**Kext Download Source**

Downloaded from github latest ventura version

karizr45, 2024-03-07:
Tried 2.3.0 alpha and got kernel panic
Any solutions? @zxystd

zxystd, 2024-03-07:
>
> Kext Version: --Latest airportitlwm for **ventura**

Are you kidding me?

>
> Tried 2.3.0 alpha and got kernel panic

2.3.0-alpha is working fine on 14.4 below versions, for above, use this one https://github.com/OpenIntelWireless/itlwm/issues/953#issuecomment-1920413379.

It is a user mistake, I am going to close this issue.

closed 2024-03-07 by zxystd