OpenIntelWireless/itlwm #965: AirportItlwm-Sonoma14.4-v2.3.0-DEBUG-alpha-ef2e076.zip连接到任何网络都会失败（2.4GHz 和 5GHz 都一样）

xjh169, 2024-03-09:
https://github.com/daliansky/XiaoMi-Pro-Hackintosh/issues/752#issue-2024159258

这个是[AirportItlwm-Sonoma14.4-v2.3.0-DEBUG-alpha-ef2e076.zip](https://github.com/OpenIntelWireless/itlwm/releases/download/v2.3.0-alpha/AirportItlwm-Sonoma14.4-v2.3.0-DEBUG-alpha-ef2e076.zip)连接失败的kext日志。
[Log_2024-03-09_19-02-11.log](https://github.com/OpenIntelWireless/itlwm/files/14546063/Log_2024-03-09_19-02-11.log)

换另外一个Sonoma_14.4Beta.kext， Wi-Fi恢复正常连接
可以正常使用下的日志
[Log_2024-03-09_19-08-18.log](https://github.com/OpenIntelWireless/itlwm/files/14546065/Log_2024-03-09_19-08-18.log)
[Sonoma_14.4Beta.zip](https://github.com/OpenIntelWireless/itlwm/files/14546067/Sonoma_14.4Beta.zip)

zxystd, 2024-03-09:
这两个应该是一样的，你再试试。

xjh169, 2024-03-09:
> 这应该是一样的，你再尝试两次。

是https://github.com/daliansky/XiaoMi-Pro-Hackintosh/actions/runs/8213285046的.kext无法正常使用，尴尬了，不好意思😬
[AirportItlwm_Sonoma144.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/14546114/AirportItlwm_Sonoma144.kext.zip)

xjh169, 2024-03-09:
[image]
估计是为小米笔记本Pro定制出现了问题 @stevezhengshiqi

stevezhengshiqi, 2024-03-09:
@xjh169 好的，那就回到小米仓库讨论这个问题的后续，不打扰z大了。
可以用小米定制的 https://github.com/daliansky/XiaoMi-Pro-Hackintosh/actions/runs/8213285046 再生成一次日志吗？
小米仓库的AirportItlwm删除了无关的固件然后进行重新编译，目的就是减小体积。

xjh169, 2024-03-10:
https://github.com/daliansky/XiaoMi-Pro-Hackintosh/issues/752#issuecomment-1987219324

closed 2024-03-10 by xjh169

stevezhengshiqi, 2024-03-10:
参考：https://github.com/daliansky/XiaoMi-Pro-Hackintosh/issues/752#issuecomment-1987179479