OpenIntelWireless/itlwm #966: Intel Wireless 7260 " iMessage & FaceTime " Sonoma 14.4 ?

Odroid13, 2024-03-12:
First of all I wanted to thank you for the work you do ^^

Thanks to your kext, I have Wifi & Bluetooth working. perfectly, on Mac OS Monterey, I can use iMessage & FaceTime, everything is perfect.

I installed Sonoma 14.4, I took the Kext for Sonoma, everything works perfectly but I can't use iMessage & FaceTime.
iCloud works

When I use the Hackintool software, I noticed that the Intel Wirless 7260 Wifi card is checked with Mac OS Sonoma 14.4 and that on Mac OS Monterey, the Wirless 7260 Wifi card is unchecked.

[image]

I tell myself that maybe this is the problem that is preventing me from having the iMessage & FaceTime services?

When I use the RJ45 cable, the iMessage & FaceTime services work perfectly.

I want to do a test but I don't know how to uncheck the WiFi card like on the Monterey screeshot.

[image]

Thanks

Odroid13, 2024-03-12:
I just tried itlwm.kext with Heliport and everything works ^^

In Hackintool, the Wifi card is unchecked, I think that to solve the problem with AirportItlwm.kext it must not be checked in Hackintool.

For the moment I use itlwm.kext with Heliport.
Thanks

anonymous-writer, 2024-03-14:
built-in can be changed with this parameter. Right device property you will get from the Hackintool.

[image]

zxystd, 2024-03-14:
It should be build-in, the hackintool reads the IOBuildIn property to desire whether it should be checked.

closed 2024-03-14 by zxystd

kamikaze07, 2024-04-18:
there is any fix for this?, iServices doesn´t work on Sonoma 14.4.1 with AirportItlwm.kext 2.3.0 kext

Rainringgoaway, 2024-05-19:
> I just tried itlwm.kext with Heliport and everything works ^^
> 
> In Hackintool, the Wifi card is unchecked, I think that to solve the problem with AirportItlwm.kext it must not be checked in Hackintool.
> 
> For the moment I use itlwm.kext with Heliport. Thanks

heliport can't read location.