OpenIntelWireless/itlwm #968: AC-9560 refusing any Wi-Fi connections

VouEuSou, 2024-03-19:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: itlwm-v2.3.0-DEBUG-alpha-e48e7cb.zip
- WiFi Card Model: Intel AC-9560
- PCI Product ID:  0x9DF0
- macOS Version: 14.4 

**Description**
Wi-Fi does not connect at all (on itlwm and AirportItlwm). I tried on 2 different routers, phone hotspot (Mi 9, Galaxy S23) and it still did not work. The computer keeps asking for the password over and over (WPA 2 Personal on all cases) . When the Wi-Fi has no password, it just straight up refuses to connect

**Bug Report Archive**

[bugreport_29841.zip](https://github.com/OpenIntelWireless/itlwm/files/14652356/bugreport_29841.zip)
Attaching both just because it could be helpful, the above is my first try with the logs and for that reason has quite a lot of turning it off and on to see if it works

[bugreport_58191.zip](https://github.com/OpenIntelWireless/itlwm/files/14652353/bugreport_58191.zip)
This one has one the "trying to connecting to a network" part specifically

**Kext Download Source**
[itlwm-v2.3.0-DEBUG-alpha-e48e7cb.zip](https://github.com/OpenIntelWireless/itlwm/releases/download/v2.3.0-alpha/itlwm-v2.3.0-DEBUG-alpha-e48e7cb.zip)

zxystd, 2024-03-19:
```
[   79.100588]: itlwm: itlwm: device timeout
[   79.100593]: itlwm: itlwm: dumping device error log
[   79.336900]: itlwm: itlwm: acquiring device failed
[   79.336909]: itlwm: itlwm: reading errlog failed
```
seems that something error on your card, does it works fine on previous any itlwm version? or does it works fine on Linux?

VouEuSou, 2024-03-19:
> ```
> [   79.100588]: itlwm: itlwm: device timeout
> [   79.100593]: itlwm: itlwm: dumping device error log
> [   79.336900]: itlwm: itlwm: acquiring device failed
> [   79.336909]: itlwm: itlwm: reading errlog failed
> ```
> 
> seems that something error on your card, did it worked on previous any itlwm version? or does it works fine on Linux?

I was previously using windows on this computer and now i booted the Ubuntu Live install,  its working fine

zxystd, 2024-03-19:
https://github.com/OpenIntelWireless/itlwm/issues/953#issuecomment-1920413379
also this version? and v2.2.0 stable version of itlwm?
the log says that the card stop response after connected and start Tx aggregation, don't know why.

andreszerocross, 2024-03-19:
> **Have You Read Our Docs** Yes
>
> **Are You Reporting A Bug** Yes
>
> **Environment**
>
> * Kext Version: itlwm-v2.3.0-DEBUG-alpha-e48e7cb.zip
> * WiFi Card Model: Intel AC-9560
> * PCI Product ID:  0x9DF0
> * [...]

Your notebook is Lenovo Thinkpad?

VouEuSou, 2024-03-19:
> Your notebook is Lenovo Thinkpad?

Yes! It's a Thinkpad T490

VouEuSou, 2024-03-19:
> [#953 (comment)](https://github.com/OpenIntelWireless/itlwm/issues/953#issuecomment-1920413379) also this version? and v2.2.0 stable version of itlwm? the log says that the card stop response after connected and start Tx aggregation, don't know why.

I already tested 2.2.0 stable version prior to opening the issue, and that one that you sent me appears to have the same bug :/ 
[bugreport_14418.zip](https://github.com/OpenIntelWireless/itlwm/files/14655381/bugreport_14418.zip)
The logs above are from the kext sent (#953)

andreszerocross, 2024-06-05:
> Yes! It's a Thinkpad T490

I have same notebook Lenovo Thinkpad T490 too with Intel AC 9560 and the problem is exactly same.

This wifi modul is soldered (can't be replaced with other wifi modul :( )

Very sad

VouEuSou, 2024-06-05:
Yeah i got nowhere with it, @andreszerocross. 
Eventually i just settled on linux for the time being, but when a workaround/fix is discovered i'll tag you.

VouEuSou, 2024-06-16:
I'm happy to say that this bug has been fixed on the latest version! I dont know when exactly it was fixed, but today i gave it another go and its perfect. 
Tagging @andreszerocross just for an fyi

closed 2024-06-16 by VouEuSou

andreszerocross, 2024-06-17:
> I'm happy to say that this bug has been fixed on the latest version! I dont know when exactly it was fixed, but today i gave it another go and its perfect. Tagging @andreszerocross just for an fyi

It's still same in my T490. Maybe you did another step. Please remember what did you do?