OpenIntelWireless/itlwm #969: AirportItlwm-Sonoma-v2.3.0 Panic on Sonoma 14.4

denizckai, 2024-03-19:
Have You Read Our Docs
Yes

Are You Reporting A Bug
Yes

Environment
Kext Version: AirportItlwm-Sonoma-v2.3.0-DEBUG-alpha-ff1138b
WiFi Card Model: Intel Wi-Fi 6E AX211 (MSI PRO Z790-A WIFI DDR4 - M.2 WiFi 6E Card)
PCI Product ID: 0x8086 (Vendor)
macOS Version: Sonoma 14.4 (23E214)

Description
First the wifi connection disconnects, then it connects again, but then randomly (or I don't know) it panics and suddenly restarts. The following data is included in the problem report when starting up Sonoma. Attached you can find the error reports after panic and restart from the previous days. And this disconnection and panic problem is happening more and more frequently.

> panic(cpu 0 caller 0xffffff8013be0b9f): zfree_drop: double free of 0xffffff9450d24fa0 to zone kalloc.type.var5.32 @zalloc.c:814
> Panicked task 0xffffff9450fa7dd8: 324 threads: pid 0: kernel_task
> Backtrace (CPU 0), panicked thread: 0xffffff991ce62b30, Frame : Return Address
> 0xffffff814ad0f6a0 : 0xffffff80134253b1 
> 0xffffff814ad0f6f0 : 0xffffff801358ac80 
> 0xffffff814ad0f730 : 0xffffff801357a34c 
> 0xffffff814ad0f7b0 : 0xffffff80133be971 
> 0xffffff814ad0f7d0 : 0xffffff801342569d 
> 0xffffff814ad0f8c0 : 0xffffff8013424d43 
> 0xffffff814ad0f920 : 0xffffff8013bdb03b 
> 0xffffff814ad0fa10 : 0xffffff8013be0b9f 
> 0xffffff814ad0fa40 : 0xffffff80134982ad 
> 0xffffff814ad0fa70 : 0xffffff8013acc500 
> 0xffffff814ad0fa90 : 0xffffff801703b494 
> 0xffffff814ad0fac0 : 0xffffff8017043f2d 
> 0xffffff814ad0fbd0 : 0xffffff801704c774 
> 0xffffff814ad0fc30 : 0xffffff8017046625 
> 0xffffff814ad0fcb0 : 0xffffff801704e356 
> 0xffffff814ad0fd30 : 0xffffff801704e1eb 
> 0xffffff814ad0fd70 : 0xffffff8017054b0e 
> 0xffffff814ad0fdd0 : 0xffffff8017055bae 
> 0xffffff814ad0fe40 : 0xffffff8016fde4b7 
> 0xffffff814ad0ffa0 : 0xffffff80133be19e 
>       Kernel Extensions in backtrace:
>          com.zxystd.AirportItlwm(2.3)[A06E8DA5-8C53-323E-B8FE-17B1CD99CE32]@0xffffff8016fd2000->0xffffff8017f71fff
>             dependency: com.apple.driver.corecapture(1.0.4)[5F8DB873-C992-3063-8589-2BC8EF56F96E]@0xffffff8015fb6000->0xffffff8015fd8fff
>             dependency: com.apple.iokit.IO80211Family(1200.13.0)[4DFEA9FE-B39D-3678-84FC-F9D4BADE290E]@0xffffff8015077000->0xffffff80152bdfff
>             dependency: com.apple.iokit.IONetworkingFamily(3.4)[689F9B98-3302-3AAB-B4AB-C6DB7121FE27]@0xffffff8015504000->0xffffff801551bfff
>             dependency: com.apple.iokit.IOPCIFamily(2.9)[F567F410-AC71-357F-A91F-C1A149F67BD8]@0xffffff8015766000->0xffffff8015797fff
>             dependency: com.apple.iokit.IOSkywalkFamily(1.0)[A5F4F226-90AD-3EB6-83D6-3DC8182630A4]@0xffffff8015852000->0xffffff801587ffff
> 
> Process name corresponding to current thread (0xffffff991ce62b30): kernel_task
> Boot args: agdpmod=pikera 
> 
> Mac OS version:
> 23E214
> 
> Kernel version:
> Darwin Kernel Version 23.4.0: Wed Feb 21 21:44:31 PST 2024; root:xnu-10063.101.15~2/RELEASE_X86_64
> Kernel UUID: A66D3683-FBC7-3319-A37F-12C83D53946F
> roots installed: 0
> KernelCache slide: 0x0000000013000000
> KernelCache base:  0xffffff8013200000
> Kernel slide:      0x00000000130e4000
> Kernel text base:  0xffffff80132e4000
> __HIB  text base: 0xffffff8013100000
> System model name: MacPro7,1 (Mac-27AD2F918AE68F61)
> System shutdown begun: NO
> Panic diags file available: YES (0x0)
> Hibernation exit count: 0
> 
> System uptime in nanoseconds: 3930160818211
> Last Sleep:           absolute           base_tsc          base_nano
>   Uptime  : 0x000003930fd6f469
>   Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
>   Wake    : 0x0000000000000000 0x0000000e9ad90c2e 0x0000000000000000
> Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
> Zone info:
>   Zone map: 0xffffff89188ee000 - 0xffffffa9188ee000
>   . PGZ   : 0xffffff89188ee000 - 0xffffff89208ef000
>   . VM    : 0xffffff89208ef000 - 0xffffff8dec288000
>   . RO    : 0xffffff8dec288000 - 0xffffff8f855bb000
>   . GEN0  : 0xffffff8f855bb000 - 0xffffff9450f54000
>   . GEN1  : 0xffffff9450f54000 - 0xffffff991c8ed000
>   . GEN2  : 0xffffff991c8ed000 - 0xffffff9de8286000
>   . GEN3  : 0xffffff9de8286000 - 0xffffffa2b3c20000
>   . DATA  : 0xffffffa2b3c20000 - 0xffffffa9188ee000
>   Metadata: 0xffffff8058cd1000 - 0xffffff8078cd1000
>   Bitmaps : 0xffffff8078cd1000 - 0xffffff8084cd1000
>   Extra   : 0 - 0

Bug Report Archive
[problem_reports.zip](https://github.com/OpenIntelWireless/itlwm/files/14652639/problem_reports.zip)

Kext Download Source
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.3.0-alpha/AirportItlwm-Sonoma14.4-v2.3.0-DEBUG-alpha-e48e7cb.zip

zxystd, 2024-03-19:
Latest alpha version has fixed this issue. BTW, please add `keepsyms=1` to `bootargs` so that we can get a readable backtrace later.

denizckai, 2024-03-19:
I would like to share my observation, there are problems with the enterprise network (eduroam), but there does not appear to be any problem with the WPA2 standard connection (at least for now).  I've been using it for a while and I haven't had any disconnect or panic problems. However, there are frequent disconnections and reconnections in the Eduroam network.

zxystd, 2024-03-20:
You can open another issue if the issue you describe is persist, close this issue since it is resolved.

closed 2024-03-20 by zxystd